awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1: cmake error :

anandpanchal, 2017-12-01:
hi everyone,

im building the environment using ./install-script-mac.
this is giving error like makefile recipe for target all failed.
hardware used : raspberry pi3
sdk used : c++

please help me.

Thanks

MushMal, 2017-12-01:
Dear Anand,

The script is currently supporting Mac OS v 10.12.4 and above. The main purpose of the script is to pull down and build the open source dependencies for the C++ SDK and the samples it includes. The core or the SDK build is based on cmake with the main build file (CMakeList.txt) located in amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/ directory.

We will be updating the script for Linux based systems shortly.

Kindest regards!

anandpanchal, 2017-12-01:
thank you so much sir. any alternative ?

MushMal, 2017-12-01:
Update the install script for Ubuntu based systems. As I mentioned, the core SDK build is cmake based. If you have all the dependencies in your system you can modify the CMakeList.txt in the kinesis-video-native-build directory to remove the locally built references to build against your system versions.

This said, please give the instal script a try.

anandpanchal, 2017-12-01:
Thank you so much sir.
sir CMakeList.txt file seems to be very complex. please could you suggest
me the solution in steps ?
i would be very thankful to you.

Thanks and Regards
Anand

MushMal, 2017-12-01:
The latest commit contains install-script which can run on Ubuntu based Linux.

anandpanchal, 2017-12-01:
I'm running it on raspberry pi 3. And based on my understanding,error is in
pynt createstack step.because of Rollback. Could you please tell me steps
to avoid this error ?

MushMal, 2017-12-01:
Hi Anand, I am not sure I understand as I am not familiar with pynt or Rollback. Can you please clarify if you are running install-script? This script pulls down and installs open source components which are needed for the build. The last two lines invoke the cmake for the core SDK build. If you are running the install-script then can you tell which open source component it is failing to build? Could you run the same on an ubuntu box?

gauravgh, 2017-12-01:
Additionally, here's a link to a Getting Started example for Raspberry Pi: https://s3-us-west-2.amazonaws.com/kinesis-video-streaming/kinesis-video-raspberry-pi-getting-started.pdf

anandpanchal, 2017-12-02:
thank you so much sir. i will follow this pdf. and if any issue aries, i
will let you know

anandpanchal, 2017-12-02:
hello sir,

im not able to find ./install_kinesis_video_demo.sh

please would you tell me the location of the file

thanks

arunned, 2017-12-05:
cmake issue in macOS

if test "emacs" != "no"; then \
	  am__dir=. am__subdir_includes=''; \
	  case po-compat.elc in */*) \
	    am__dir=`echo 'po-compat.elc' | sed 's,/[^/]*$,,'`; \
	    am__subdir_includes="-L $am__dir -L ./$am__dir"; \
	  esac; \
	  test -d "$am__dir" || ../../build-aux/install-sh -c -d "$am__dir" || exit 1; \
	  emacs --batch \
	      \
	    $am__subdir_includes -L . -L . \
	    --eval "(defun byte-compile-dest-file (f) \"po-compat.elc\")" \
	    --eval "(unless (byte-compile-file \"po-compat.el\") (kill-emacs 1))"; \
	else :; fi
Wrote /Users/XXXXXXXXXX/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/downloads/gettext-0.19.8/gettext-tools/misc/po-compat.elc
if test -f archive.dir.tar; then \
	  inputfile=archive.dir.tar; \
	else \
	  inputfile='.'/archive.dir.tar; \
	fi; \
	./convert-archive dir git "$inputfile" archive.git.tar.gz
fatal: Failed to resolve 'HEAD' as a valid ref.
convert-archive: *** git init failed
convert-archive: *** Stop.
make[4]: *** [archive.git.tar.gz] Error 1
make[3]: *** [all-recursive] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

arunned, 2017-12-05:
will be helpful to get this script ./install_kinesis_video_demo.sh

MushMal, 2017-12-07:
@arunned could you please try the latest commit? We have made some improvements and updated the README.md with instructions on the build-time dependencies. Could you check your build tools versions?

Also, the install-script will handle the Raspberry PI

closed 2017-12-09 by MushMal