awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #5: CMake Error "does not appear to contain CMakeLists.txt"

gcoop, 2017-12-03:
FYI:

Whilst executing `./kinesis-video-native-build/install-script` I got the following error `CMake Error: The source directory "/Users/Gavin/repos/thirdparty/amazon-kinesis-video-streams-producer-sdk-cpp" does not appear to contain CMakeLists.txt.`. To resolve I moved CMakeLists.txt from kinesis-video-native-build to root of repo. Updated the src paths in CMakeLists.txt to match new locations and reran the command to resolve the issue.

MushMal, 2017-12-03:
I am wondering whether you executed ./install-script from within the kinesis-video-native-build directory?

gcoop, 2017-12-04:
No I executed it from the repo root. That's probably why it couldn't locate `CMakeLists.txt`.

closed 2017-12-04 by MushMal

enusbaum, 2018-03-04:
I'm having this issues on my rpi while following the tutorial at

https://s3-us-west-2.amazonaws.com/kinesis-video-streaming/kinesis-video-raspberry-pi-getting-started.pdf

zhiyua-git, 2018-03-04:
Hi enusbaum,

Please change directory to kinesis-video-native-build and run install-script from there.

**cd kinesis-video-native-build; ./install-script**

We will also fix that on installation guide soon.

Thanks,
-ZH

MushMal, 2018-03-22:
Apologies again. Please feel free to reach out for any question no matter how small or obvious those are.

triggerx, 2018-06-04:
Guide is still not updated... what's the deal?

hussamsayeed, 2019-09-25:
$ cd kinesis-video-native-build; ./min-install-script
bash: cd: kinesis-video-native-build: No such file or directory
./min-install-script: line 15: /home/Hussam: No such file or directory

this is the error raising while Building the Kinesis Video Streams Producer SDK using min-install-script.. Anyone please help me out!

MushMal, 2019-09-25:
@hussamsayeed you need to navigate to the SDK directory in the terminal and then navigate down to the native build directory by calling `cd kinesis-video-native-build`

hussamsayeed, 2019-09-25:
[image]
I already did that.[can see in Screenshot]

zhiyua-git, 2019-09-25:
Your user name contains space. The script was not able to handle the path, can you replace **$KINESIS_VIDEO_ROOT** with **"$KINESIS_VIDEO_ROOT"** in these two places and retry? 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/min-install-script#L15
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/min-install-script#L206

hussamsayeed, 2019-09-25:
Should I replace that code with the code which is there in those links?

hussamsayeed, 2019-09-25:
Hussam Sayeed@DESKTOP-6TJVREH MINGW64 ~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
$ ./min-install-script
Setting up for KVS Producer SDK installation
Current working path cannot have space in it !

zhiyua-git, 2019-09-26:
Hi @hussamsayeed ,

Seems mingw doesn't support space in the path. Can you either try to install your mingw into directory which contains **no space** in the path or use the **short (8.3) name** as mentioned [here](https://stackoverflow.com/questions/5999507/mingw-make-cant-handle-spaces-in-path)?

You can do **export KINESIS_VIDEO_ROOT="/Your/Updated/Path"** and rerun the min-install-script.

hussamsayeed, 2019-09-26:
I changed the path and it works. thank you for the support @zhiyua-git 
[image]
[image]

but the problem is segmentation failed[attached].
I followed the this documentation [https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msys2.md] 

Can anyone please help me out!

zhiyua-git, 2019-09-26:
@hussamsayeed , you need to put in your **actual** AccessKey/SecretKey in 
export AWS_ACCESS_KEY_ID=YOUR_ACCESS_KEY
export AWS_SECRET_ACCESS_KEY=YOUR_SECRET_ACCESS_KEY
to run it. 

Also make sure you have configured proper Kinesis Video Streams permission for the role of AccessKey/SecretKey you are going to use. It can be as simple as adding AmazonKinesisVideoStreamsFullAccess policy to your role. https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iam.html

hussamsayeed, 2019-09-30:
@zhiyua-git thank you so much for the suppport!
yeah i gave the actual access keys and also configured the permisiions.. still the same error!!!!!

zhiyua-git, 2019-09-30:
@hussamsayeed  Can you provide the log for the failure? Thanks.

hussamsayeed, 2019-10-01:
@zhiyua-git Yeah.. attached the log for the failure.
[image]

zhiyua-git, 2019-10-01:
@hussamsayeed , what command did you run? Sample app or gstreamer plugin kvs-sink? Can you provide log with export GST_DEBUG=4? Thanks.

hussamsayeed, 2019-10-04:
@zhiyua-git I'm following this document [https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msys2.md]

So when running the sample app im getting the error.[gst-launch-1.0 ksvideosrc ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name=stream-name access-key=YOUR_ACCESS_KEY secret-key=YOUR_SECRET_ACCESS_KEY_ID
] even after i've changed with my aws keys.

zhiyua-git, 2019-10-04:
@hussamsayeed,  Can you provide log with **export GST_DEBUG=4**? Thanks.

hussamsayeed, 2019-10-04:
@zhiyua-git [attached
[image]
[image]

]

chehefen, 2019-10-04:
can you copy libcproducer.dll and libproducer.dll to C:\msys64\mingw64\bin and copy libgstkvssink.dll to C:\msys64\mingw64\lib\gstreamer-1.0 and make sure you are not exporting GST_PLUGIN_PATH. Then try your pipeline again.

hussamsayeed, 2019-10-14:
Thank you so much @zhiyua-git  @chehefen  for the support. I got the output in the kinesis streams!!