awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #6: autoreconf is not installed in Ubuntu 17.01 using the default installation script

luiscosio, 2017-12-05:
OS: Ubuntu 17.01

Issue:

./install-script: line 211: autoreconf: command not found

How to fix it:

sudo apt-get install autoconf autogen

MushMal, 2017-12-05:
Thanks for reporting Luis. We will include required build tools in the Readme.md shortly

luiscosio, 2017-12-06:
Thank you @MushMal, I'm having an issue I cant' seem to fix:

/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/../kinesis-video-producer-jni/src/include/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.h:14:10: fatal error: jni.h: No such file or directory

I already installed open jdk and Oracle's distribution.

Could you help me out with it?

oily126, 2017-12-06:
Setting up your JAVA_HOME env variable to local java directory would help find jni.h.

For example, export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/

zhiyua-git, 2017-12-06:
Setting up your JAVA_HOME env variable to local java directory would help find jni.h.

For example, export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/

luiscosio, 2017-12-06:
Thank you @zhiyua-git, I will close this issue.

closed 2017-12-06 by luiscosio