awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #7: Incorrect AWS REGION

bvasan, 2017-12-06:
Hi,
The GStreamer sample C++ application creates the Kinesis Video stream in US-West-2, even if I set the AWS_DEFAULT_REGION to us-east-1. The region parameter in credentials file is also set to us-east-1. 
$echo $AWS_DEFAULT_REGION
us-east-1
$ ./kinesis_video_gstreamer_sample_app home-driveway 
creates the stream in us-west-2. 
How do I override this. I would like to create the stream in us-east-1

Thanks
Babu Srinivasan

bvasan, 2017-12-06:
Looks like the values are hard coded in 
./kinesis-video-producer/src/DefaultCallbackProvider.h
./kinesis-video-producer/src/DefaultCallbackProvider.cpp
I think we should read the value from the environment variable $AWS_DEFAULT_REGION

Thanks
Babu Srinivasan

MushMal, 2017-12-06:
Sorry for the delayed reply. Indeed, the regions are hard-coded for both the integration unit tests and the sample gstreamer application. This is a great suggestion though to use env variables - just like the credentials. We will implement this ASAP

MushMal, 2017-12-06:
The latest commit contains this fix

closed 2017-12-06 by MushMal

bvasan, 2017-12-07:
Thanks for the update!!

bvasan, 2017-12-07:
Thanks!! I will pull the latest and test it out.