awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #10: Unable to stream

bvasan, 2017-12-09:
Hi,
I am unable to stream from RaspberryPi+Camera to Kinesis Video using the demo app. I am getting the following error (after fixing the region related error - Issue#9). 

DEBUG - Stream is ready
**Unable to set the pipeline to the playing state.**

Any inputs on how to troubleshoot this error? 

Thanks,
Babu Srinivasan

MushMal, 2017-12-09:
Babu, there are number of reasons why the GStreamer pipeline won't transition to "playing" state. In order to debug you could set the default verbosity level of the GStreamer by specifying GST_DEBUG=x environment variable. For example, you could get more information why the pipeline fails to get to "ready" state launching the demo app like this..

GST_DEBUG=4 AWS_ACCESS_KEY_ID=xx AWS_SECRET_ACCESS_KEY=xx ./kinesis_video_gstreamer_sample_app <stream>

For more verbosity you can specify GST_DEBUG=5

We also encountered the following which is in README.md
* Raspberry PI failure to load the camera device. To check this is the case run ls /dev/video* - it should be file not found. The remedy is to run the following:

>ls /dev/video*
>vcgencmd get_camera 
-- supported=1 detected=1

>sudo modprobe bcm2835-v4l2 
>ls /dev/video*

bvasan, 2017-12-09:
Thanks for the update. sudo modprobe bcm2835-v4l2 resolved the issue. I am running into other library related issues now - I should be able to resolve those and hopefully, have the video streaming from the RaspberryPi this weekend!! Thanks

MushMal, 2017-12-09:
Sounds great! Please keep us posted. Please resolve the issue once you are able to stream

bvasan, 2017-12-10:
Hi 
The demo app failed first failed with the following error. 
./kinesis_video_gstreamer_sample_app: error while loading shared libraries: libgstreamer-1.0.so.0: cannot open shared object file: No such file or directory
I updated LD_LIBRARY_PATH environment variable to include native build lib path to resolve this issue.
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib

However, I am now stuck the following issue.  Any suggestions? Thanks!

0:00:01.735129915  3247  0x1869260 DEBUG         v4l2bufferpool gstv4l2bufferpool.c:122:gst_v4l2_buffer_pool_copy_buffer:<source:pool:src> copy video frame
0:00:01.735271113  3247  0x1869260 DEBUG        GST_PERFORMANCE video-frame.c:372:gst_video_frame_copy_plane: copy plane 0, w:1280 h:720
0:00:01.737919808  3247  0x1869260 DEBUG        GST_PERFORMANCE video-frame.c:372:gst_video_frame_copy_plane: copy plane 1, w:640 h:360
0:00:01.738876994  3247  0x1869260 DEBUG        GST_PERFORMANCE video-frame.c:372:gst_video_frame_copy_plane: copy plane 2, w:640 h:360
0:00:01.739847774  3247  0x1869260 DEBUG         v4l2bufferpool gstv4l2bufferpool.c:1397:gst_v4l2_buffer_pool_release_buffer:<source:pool:src> release buffer 0x544318b8
**
**ERROR:gstv4l2bufferpool.c:1119:gst_v4l2_buffer_pool_qbuf: assertion failed: ("Value of time " "timestamp" " is out of timeval's range" && ((timestamp) / GST_SECOND) < G_MAXLONG)**

Thanks,
Babu Srinivasan

bvasan, 2017-12-10:
Issue is resolved!!!! Updating RaspberryPi Kernel resolved the issue with w4l2 driver

$sudo rpi-update 
and reboot

I am now able to stream video from my Raspberry Pi to Kinesis Video!!!!! Thank you for your support. 

Thanks,
Babu Srinivasan

**Note**:  I was unable to stream to US-East-1. I am getting bad gateway error in Kinesis Video preview console. Seems to be a service issue in east region. I will re-try us-east region tomorrow.

MushMal, 2017-12-11:
Apologies for your experience with us-east-1 - must have fallen with the time we had a brief period of reconfiguration. Could you retry please? 

Please update the issue with your findings and resolve the issue if all is OK

bvasan, 2017-12-11:
I retried in Us-east-1 and I am still getting the same "bad gateway error". Screenshot attached
[image]

Thanks

MushMal, 2017-12-11:
@bvasan apologies again. Could you please retry? We have had transient issues with some sessions. Those should be fixed.

bvasan, 2017-12-11:
Its working us-east now. Thanks for the quick follow-up!! 

Thanks
Babu Srinivasan

closed 2017-12-11 by bvasan