awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #11: GStreamer-CRITICAL **: gst_value_set_fraction_range_full: assertion

dogtownmedia, 2017-12-11:
Hi.

Getting the following when I run the compiled  installed on a Pi with the compiled _kinesis_video_gstreamer_sample_app_ with  min-install-script.

(kinesis_video_gstreamer_sample_app:570): GStreamer-CRITICAL **: gst_value_set_fraction_range_full: assertion 'gst_util_fraction_compare (numerator_start, denominator_start, numerator_end, denominator_end) < 0' failed

Thanks

MushMal, 2017-12-11:
@dogtownmedia, have you run full install-script before? If not, it will attempt to link against the system versions of the library which are installed on your Raspberry PI that might have a mismatch with the version of the libx264enc encoder.

Could you either re-install the system libraries with the exact versions as scripted in the install-script or run install-script on a clean repository?

dogtownmedia, 2017-12-12:
HI @MushMal. I wiped the Pi SD card, installed a fresh OS then built using the full install script and getting the same error. Could this be the uv4l config?

MushMal, 2017-12-12:
It's hard to tell - I am not very familiar with it.

Could I ask you to try the following from the readme:

$ vcgencmd get_camera
$ sudo modprobe bcm2835-v4l2
$ ls /dev/video*

If the camera is successfully detected and the driver is loaded then could you please follow with

$ sudo rpi-update 
$ sudo shutdown -r 0

Please let me know if this helped

dogtownmedia, 2017-12-12:
OK we got it. The uv4l service was running in the background on boot. I killed that and it worked. Thanks

closed 2017-12-12 by dogtownmedia

MushMal, 2017-12-12:
Whohoo!! Thanks for the update