awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #13: Internal data stream error

backdoorcodr, 2017-12-15:
Hello,

I am using Ubuntu system at the moment to execute the sample application. Another issue I am facing now is while executing the command AWS_ACCESS_KEY_ID=<my-access-key-id> AWS_SECRET_ACCESS_KEY=<my-secret-access-key> ./kinesis_video_gstreamer_sample_app my-test-stream

As i execute it, I get the error message as the following: 
`INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 123156 Expiration: 0
INFO - New credentials expiration is 125556
DEBUG - createDeviceHandler invoked

createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 123156 Expiration: 0
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.513368609391E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:123123123:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:123123123:stream/my-test-stream/1231231231213","StreamName":"my-test-stream","Version":"BLBpBtdpsQ3NC44AMyEh"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:123123123:stream/my-test-stream/12312312312312314456
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:123123123:stream/my-test-stream/12312312312312314456

describeStreamResultEvent(): Describe stream result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 123157 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-123123134.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-123123134.kinesisvideo.us-west-2.amazonaws.com

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 123158 Expiration: 125556
INFO - New credentials expiration is 125558

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream my-test-stream is Ready.
DEBUG - Stream is ready
**Error received from element source: Internal data stream error.**
**Debugging information: gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:test-pipeline/GstV4l2Src:source:
streaming stopped, reason not-negotiated (-4)**

freeKinesisVideoClient(): Freeing Kinesis Video ClientDEBUG - streamDataAvailableHandler invoked
DEBUG - streamClosedHandler invoked
`

I understood from the previously [closed issue](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/10) that it might has to be because of the activation of driver of my source. However, after the execution of `sudo modprobe gspca_zc3xx`; it does not seem to have worked and still throwing the same error. 

Any hints / guidance what is the issue here and how to resolve ?

MushMal, 2017-12-15:
Can you validate if your camera is detected? Could you please run 
`ls /dev/video0` - the output should list the device.

Not sure entirely but this could be caused by the camera USB driver. Can you try some other applications and see if they detect the camera OK. Here is a link to a page with some debugging info: https://help.ubuntu.com/community/Webcam

backdoorcodr, 2017-12-16:
@MushMal this is what I was suspecting as well; therefore I did that earlier (plus after your response one more time just to be sure). I was able to find `video0` as a response from this command. 

Furthermore, I followed the link you provided and tested if the camera works or not. I chose cheese, vlc  and streamer; and they all seem to be working quite well. `hwinfo` also seems to be accessing the camera and providing the details of it as well. 

Can there be any other possibility or reason? What is -4 code mentioned in the error? 

I went to the line 2939 of gstbaserc.c and understood that once it gets a null_buffer, it throws error that "internal data flow error". May be there what we need is to have some waiting time for the stream or the user kills the process himself? Just a suggestion. 

Looking forward to your reason behind this error message and hints / suggestions to resolve it. 

Many thanks

MushMal, 2017-12-16:
@backdoorcodr, I am not familiar with this particular code - it's in GStreamer. Let me try to reproduce your scenario here locally by setting up a box and I will have a better understanding.

Apologies for this but it seems that the GStreamer pipeline might be hitting this issue on Ubuntu.

backdoorcodr, 2017-12-16:
@MushMal Many thanks. that will be great. Let me know how it goes.

unicornss, 2018-01-06:
Hi backdoorcodr,  

Given you were successfully able to stream using Java SDK :)  are you still facing issues with CPP SDK?  Please use the latest SDK in that case. 

Thanks
-SS

MushMal, 2018-01-24:
@backdoorcodr any luck in trying this out? There is a couple of updates to the GStreamer sample application which applies ranges to the resolutions which might resolve the negotiation issue you were seeing.

Please give it a try.

backdoorcodr, 2018-02-08:
@unicornss @MushMal sincere apologies for replying back late. I have been on vacations. Let me try it out this weekend and let you guys know. Although, yes I have been successful to stream using the Java SDK.

unicornss, 2018-02-08:
Glad to hear that @backdoorcodr

closed 2018-02-08 by unicornss

backdoorcodr, 2018-02-13:
@unicornss @MushMal Latest SDK resolves this issue :) Many thanks for this great support.