awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #15: How to Build Kinesis Producer cpp on windows

ibr123, 2017-12-18:
Hi,

In the [document](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/kinesisvideo-dg.pdf#gs-console-whatnext)  at the prerequisite section there is a paragraph saying:

> > Your NativeLibraryPath must contain your KinesisVideoProducerJNI file, available at https://
> github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp. The file extension for this file
> will depend on your operating system:
> • KinesisVideoProducerJNI.so for Linux
> • KinesisVideoProducerJNI.dylib for macOS
> • KinesisVideoProducerJNI.dll for Windows (not currently available)
> Note
> Pre-built libraries are not currently available. You must build the libraries yourself for your
> platform.

as stated that there is no ready scripts to build on windows, but how to do that?

Thanks

sambithota, 2017-12-21:
Is there an answer to this. I am facing the same issue. I am getting this. 
EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError
Because libKinesisVideoProducerJNI.dll is not present. How to generate this in windows?

zhiyua-git, 2017-12-21:
Thanks for reaching out, we are still looking into the WIndows build and hope to get the Windows solution out soon.

sambithota, 2018-01-02:
Any update on this one?

unicornss, 2018-01-06:
We will publish once we get the Windows Build for Kinesis Video Streams in our roadmap.
Closing this for now.

Thanks
-SS

closed 2018-01-06 by unicornss

sandeepatgetronics, 2018-05-11:
Am I right in understanding that as the  KinesisVideoProducerJNI.dll for windows is missing, we cannot use the Producer library when developing in Windows. If we have to use the Producer libary we should either be developing in Linux or macOs. Is that correct?

MushMal, 2018-05-11:
@sandeepatgetronics you are correct in your assumption. Unfortunately, we don't have any build support for the Windows platform yet. We are planning on closing that gap and supporting Windows as a first class platform in a very near future.

LilaRios, 2019-11-12:
Hello,

Is there any update for the build support for the Windows platform?

zhiyua-git, 2019-11-12:
Hi @LilaRios , it is available now. You can use https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/lib/windows or you can build it following https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md