awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #16: Streaming from PI gives error

smitshilu, 2017-12-19:
Hello,
I have installed producer SDK on pi successfully but I am receiving following error while running sample app.

```
AWS_ACCESS_KEY_ID=############
AWS_SECRET_ACCESS_KEY=################################### ./kinesis_video_gstreamer_sample_app smit_test_stream
```

zhiyua-git, 2017-12-20:
Could I ask you to try the following from the readme:

$ vcgencmd get_camera
$ sudo modprobe bcm2835-v4l2
$ ls /dev/video*

If the camera is successfully detected and the driver is loaded then could you please follow with

$ sudo rpi-update
$ sudo shutdown -r 0

Please let me know if this helped

zhiyua-git, 2017-12-27:
Hi,
We just released SDK Version 1.1.1 which contains an USB camera gstreamer configuration fix. If you were testing on your USB camera, can you try the latest release and see if it solves your issue? Thanks.

shiopon01, 2017-12-28:
Hi,
I was in trouble too with this problem. (using Raspberry Pi)
However, It worked by correcting line 336 and 356 in `kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp` 😃
(change the number from 1280 and 720 to 640 and 480)

I think it would be better to write the file name in README.md along with the line number.
Thanks.

unicornss, 2018-01-06:
Thanks, Shiopon01. Glad you are able to resolve and also posting your notes to help others :)
-SS

unicornss, 2018-01-06:
Hi  smitshilu ,  Are you still having issues with the install steps with the latest instructions and SDK version?

Thanks
-SS

smitshilu, 2018-01-08:
I am sorry for late reply. @unicornss I have changed my platform to TX2 so not sure about pi.

unicornss, 2018-01-08:
Glad to hear that.

Thanks
-SS

closed 2018-01-08 by unicornss