awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #17: liblog4cplus-1.2.so.5 does not exist error

shiopon01, 2017-12-25:
Hello.
I have builded producer SDK on Ubuntu 17.10 successfully but I am receiving following error appears when starting `kinesis-video-native-build/start`.

```
$ ./kinesis-video-native-build/start
./kinesis-video-native-build/start: error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory
```

I have not `liblog4cplus 1.2.so.5` and I can not install it by apt-get, so I am in trouble.

liblog4cplus exists if it is `1.1-9`.

```
$ apt-cache search liblog4cplus
liblog4cplus-1.1-9 - C++ logging API modeled after the Java log4j API - shared library
liblog4cplus-dbg - C++ logging API modeled after the Java log4j API - debug library
liblog4cplus-dev - C++ logging API modeled after the Java log4j API - development library
```

zhiyua-git, 2017-12-25:
Hi, this is a known issue we are trying to fix. Now there is a temporary workaround in in README.md:
Troubleshooting.
Ubuntu builds link against the system versions of the open source component libraries or missing .so files (./kinesis-video-native-build/start shows linkage against system versions of the open source libraries). We are working on providing fix but the immediate steps to remedy is to run rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles and run ./kinesis-video-native-build/install-script to rebuild and re-link the project only.

Can you try this out and let me know if it works for you? Thanks.

shiopon01, 2017-12-25:
I did it and `./start` worked properly!
(I got a `libssl.so.1.1` error, but this was my `$LD_LIBRARY_PATH` problem.)

Close the issue. Thanks!!😄

closed 2017-12-25 by shiopon01