awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #18: Not able to start Streaming

rajdarbar, 2017-12-27:
Hello,
I am new in kinesis video streaming service,when i try to run   AWS_ACCESS_KEY_ID=################# AWS_SECRET_ACCESS_KEY=################# ./kinesis_video_gstreamer_sample_app my_stream

it gives me following error and not able to start streaming

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream my_stream is Ready.
DEBUG - Stream is ready
Not all elements could be created.
INFO - Freeing Kinesis Video Stream my_stream

freeKinesisVideoStream(): Freeing Kinesis Video stream.DEBUG - streamDataAvailableHandler invoked
DEBUG - streamClosedHandler invoked

freeKinesisVideoClient(): Freeing Kinesis Video Client
commonHeapDebugCheckAllocator(): Heap is initialized
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
commonHeapDebugCheckAllocator(): Heap size: 				0
commonHeapDebugCheckAllocator(): Number of allocations: 				0
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7f106928d010
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:	0x7f106928d010		size:	536870879
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Curl shutdown.

unicornss, 2017-12-27:
Thanks rajdarbar for checking the Kinesis Video Stream SDK. 

Could you provide us details on your setup?
- the environment you are running the demo application from
 (e.g. ubuntu linux, mac or raspberry pi )  
- which camera device are you using for streaming (e.g. usb camera or built in)..

-ss

rajdarbar, 2017-12-28:
Hi unicornss,

i am using **ubuntu 17.04** and **usb webcam camera.**

Thanks 
Raj

unicornss, 2017-12-28:
Hi Raj,

To learn more about the  Kinesis Video Streams  please refer the documentation here:

- http://docs.aws.amazon.com/kinesisvideostreams/latest/dg/what-is-kinesis-video.html

Additional Readings:

- https://aws.amazon.com/kinesis/video-streams/
- https://aws.amazon.com/blogs/aws/amazon-kinesis-video-streams-serverless-video-ingestion-and-storage-for-vision-enabled-apps/

The error you are seeing "Not all elements could be created. " suggests that your GStreamer pipeline does not have all elements which could be that camera is not detected/loaded (drivers).

Can you run the gst-launch-1.0 to see if the USB video camera is able to capture the video through GStreamer pipeline?
Example:
`gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,width=1280,height=720,framerate=30/1 ! xvimagesink`

If your camera is supported and drivers are installed correctly then you should be able to see the camera output displayed on your computer screen.

You can also run the command `gst-inspect-1.0` to check which GStreamer plugins are available in your system.
Hope this helps you to debug the setup and required drivers for your system.
You can also refer to the https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md for troubleshooting.

-ss

rajdarbar, 2018-01-02:
Hi unicornss,

i need your one help ,
i want to achive following steps

1)Pass live stream to kinesis video stream. 
2)Store this stream in s3 bucket.
3)Using kinesis video stream api show that live stream in mobile.

first step is done using offline video , but how can i show that same video in mobile using kinesis video stream api. 
Thanks
Raj

unicornss, 2018-01-03:
Hi Raj,

_1) Pass live stream to kinesis video stream._

Glad you are able to store offline video to kinesis video stream.  
The demo sample should be able to get the video stream from a video camera or from laptop built-in webcam sent to Kinesis Video Streams service.

Sample output (from running the demo app in Ubuntu):

INFO - Initializing curl.
createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 608787 Expiration: 0
INFO - New credentials expiration is 611187
DEBUG - createDeviceHandler invoked
createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream my-stream-jan3

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 608787 Expiration: 0
DEBUG - Awaiting for the stream to become ready...
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.515006185591E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:828808393645:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:828808393645:stream/my-stream-jan3/1515006185591","StreamName":"my-stream-jan3","Version":"eFggwrXsBUEI1KqWN8Zy"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:828808393645:stream/my-stream-jan3/1515006185591
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:828808393645:stream/my-stream-jan3/1515006185591

describeStreamResultEvent(): Describe stream result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 608787 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 608787 Expiration: 611187
INFO - New credentials expiration is 611187

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream my-stream-jan3 is Ready.
DEBUG - Stream is ready
DEBUG - putStreamHandler invoked
DEBUG - No active streams states for my-stream-jan3

putStreamResultEvent(): Put stream result event.DEBUG - streamDataAvailableHandler invoked
DEBUG - Note data received: duration(100ns): 200000 bytes: 9858
INFO - Creating new connection for Kinesis Video stream: my-stream-jan3

getKinesisVideoMetrics(): Get the memory metrics size.
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00000000019ca226.INFO - Refreshing credentials. Force refreshing: 0 Now time is: 608793 Expiration: 0
DEBUG - Kinesis Video client and stream metrics
	>> Overall storage size: 536870912
	>> Available storage size: 536861021
	>> Allocated storage size: 9891
	>> Total view allocation size: 144056
	>> Total streams frame rate: 30
	>> Total streams transfer rate: 4194304
	>> Current view duration: 200000
	>> Overall view duration: 200000
	>> Current view size: 9858
	>> Overall view size: 9858
	>> Current frame rate: 30
	>> Current transfer rate: 4194304
DEBUG - streamDataAvailableHandler invoked
DEBUG - Note data received: duration(100ns): 4600000 bytes: 10669
DEBUG - postHeaderReadFunc (curl callback) invoked
DEBUG - Curl post header write function returned:HTTP/1.1 100 Continue

DEBUG - postHeaderReadFunc (curl callback) invoked
DEBUG - Curl post header write function returned:

DEBUG - postBodyStreamingReadFunc (curl callback) invoked
[... 27 lines cut ...]
DEBUG - Curl post header write function returned:Transfer-Encoding: chunked

DEBUG - postHeaderReadFunc (curl callback) invoked
DEBUG - Curl post header write function returned:Date: Wed, 03 Jan 2018 19:08:41 GMT

DEBUG - postHeaderReadFunc (curl callback) invoked
DEBUG - Curl post header write function returned:

DEBUG - postBodyStreamingWriteFunc (curl callback) invoked
INFO - Curl post body write function returned:{"EventType":"BUFFERING","FragmentTimecode":0,"FragmentNumber":"91343852333181640069503053123985593269300060085"}

DEBUG - Processed ACK OK.

_2)Store this stream in s3 bucket._
Kinesis Video Streams stores the media in S3 and you can refer the API for GetMedia in https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_Operations_Amazon_Kinesis_Video_Streams_Media.html 

_3)Using kinesis video stream api show that live stream in mobile_
There is no support for get media to get live streaming yet.  Kinesis Video Streams console provides a sample of using the get media api for rending the video in the browser.  

4)  Yes, the live stream analysis in amazon-rekognition uses Kinesis Video Streams.

Hope this helps.

Thanks
-SS

rajdarbar, 2018-01-04:
Thank you so much unicornss. i am able to start live stream now with your help .
i have one more question is that what is that **service name of kinesis video stream**
and also is there any lambda function available using that i can put my kinesis video stream videos in s3 bucket . 

thanks
Raj

zhiyua-git, 2018-01-06:
Hi Raj,

If you are specifying service name you should use in IAM policy for our service, then it is **kinesisvideo**. https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iam.html

Now we don't have lambda function for this purpose. But the video data is already store in our S3 bucket if your stream's data retention is not 0. You can use [Parser Library](https://github.com/aws/amazon-kinesis-video-streams-parser-library) to get your video data.

Thanks,
ZH

unicornss, 2018-02-19:
Closing this issue as is able to stream successfully.  If needed pl. an open a separate issue for parsing the video. https://github.com/aws/amazon-kinesis-video-streams-parser-library

closed 2018-02-19 by unicornss

Shumakriss, 2018-02-24:
Hi @unicornss, in your snippet:

`gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,width=1280,height=720,framerate=30/1 ! xvimagesink`

Should it matter if I am using xvimagesink or autovideosink? Using autovideosink I do get the camera input displayed on-screen. I found [this thread](http://gstreamer-devel.966125.n4.nabble.com/xvimagesink-Could-not-initialise-Xv-output-and-No-port-available-td3092246.html) and ran xvinfo and tells me I have no adapters present. However, I assume that shouldn't matter for the gstreamer demo since it's CLI-based, it shouldn't need X configuration. Is there another test you could recommend?

unicornss, 2018-02-26:
GStreamer demo app now can take command line arguments for height, width and framerate.
```
 "Usage: AWS_ACCESS_KEY_ID=SAMPLEKEY AWS_SECRET_ACCESS_KEY=SAMPLESECRET ./kinesis_video_gstreamer_sample_app my-stream-name -w width -h height -f framerate -b bitrateInKBPS");
```

> A.  If the resolution is provided then the sample will try to check if the camera supports that resolution. If it does detect that the camera can support the resolution supplied in the command line, then streaming starts; else, it will fail with an error msg Resolution not supported.

> B. If no resolution is specified, the demo will try to use these three resolutions 
      1920x1080,    
      1280x720 and
      640x480 in that order (highest resolution first) and will start streaming once the camera supported resolution is detected.

You can also use *GST_DEBUG=4* at the start of the command line to see the capabilties of the camera, or if any Element/Pipeline error you notice in your environment.

Please let us know if this helps.

Shumakriss, 2018-02-26:
It looks like none of my GST plugins are loading. I am not super familiar with GST but I suppose that might impact the element factories that are throwing warnings too. I think I need to verify that the system GST plugin .so's are all linked properly but it's been awhile since I did that with a Linux/C++ program. Here's the logs.

[gst-output.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1756915/gst-output.txt)
[sample-app.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1756916/sample-app.txt)

Also, if it's easier I can work toward just this ticket or just #33 to avoid confusion. I have two RPi's but only one picamera so I can only test one at a time. For this ticket, I am using a slightly older board RPi Model B+ v1.2 but still using a recently installed and upgraded Raspbian.

Shumakriss, 2018-02-26:
I just noticed the log in [gst-output.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1756915/gst-output.txt): 

`0:00:00.001101454   800   0xb27e00 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib`

My download directory is in kinesis-video-native-build since I changed directories there to run the install-script. Copied the files from /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib to /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib and GST seems to find the plugins now. This issue might be due to not following installation steps properly but I thought running install-script in kinesis-video-native-build was OK (unless I did something else wrong too!). I don't think that applies to this issue though. If I do another install later and have the same directory problem, I will try to get it reproducible and file an issue.

Unfortunately, this install and RPi board is giving me issue #33 too. I don't think I have any reason to keep posting on this issue (18) at this point. 

[issue-18-output-2.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1756965/issue-18-output-2.txt)
[gst-output-2.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1756966/gst-output-2.txt)

RushabhK23, 2018-03-07:
Can someone tell me about working get media api for rending the video in the browser ??

MushMal, 2018-03-07:
@RushabhK23 please open a separate issue for this in the consume library. I will provide some answers here though for your convenience.

* If you want to use the standard sandboxed browsers environment then you need to rely on MSE, otherwise, you need to install flash or similar product that will allow some surface/hardware acceleration and most browsers/environments will not allow it.
* Most browsers support HLS/MPEG-DASH playback but all of these are "Live" or "VOD" playbacks and can not be real-time. You will always have at least the fragment duration latency.
* We are working on providing HLS/MPEG-DASH solution - no ETA yet.

teamone-saml, 2019-10-14:
Hi Unicornss , I am working on saving Kinesis video stream into my own S3 bucket using lambda function and I have written the lambda function for the same but facing issue. I could see my video in S3 but that can not be played in the browser. I think its the issue with video parsing,need your help here. Please let me know what all details you need

unicornss, 2019-10-15:
@teamone-saml 

Could you elaborate on  

> could see my video in S3 but that can not be played in the browser

- How are you saving the ingested KVS stream into a file in S3? 
- Are you using [Kinesis video streams parser library](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExampleTest.java)? 
- While ingesting the file are able to see in live stream in the console? Are you able to use HLS [viewer](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/hls-playback.html) [sample](https://github.com/aws-samples/amazon-kinesis-video-streams-hls-viewer/blob/master/index.html ) for playback? 

As posted by @zhiyua-git in [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/273#issuecomment-542389235) you can refer to the [parser library ](https://github.com/aws/amazon-kinesis-video-streams-parser-library)[samples](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExampleTest.java) to save the ingested video stream into local [files](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/main/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExample.java#L199) and then upload to S3.

hameedriswan7, 2020-01-02:
is there any possible way restart to Kinesis video stream every 1 hour . i can able to stream video in aws console i need to restart every one hour of period.

zhiyua-git, 2020-01-02:
Hi @hameedriswan7 , 

Can you clarify more on why do you want to restart KVS stream every 1 hour? Internally due to connection expiration for security, it is already doing connection restart for every 40 minutes.

Thanks,
-ZH

hameedriswan7, 2020-01-03:
Hi @zhiyua-git,
Using getmedia i can retrieve stream in local as Single stream.With single stream i cannot able to upload in S3 it cause Storage issue. so i am running live stream using putmedia in jar file.In that if i stop and again start it can break the stream i can upload to S3 1 hour  breakage of live stream video stored as normal in video in local.Again restart the stream get another 1 hour video like the way i asked.

These things are need to complete the process link given below please refer.For that still i am not getting any feature. so i decide to breal stream every hour to upload in s3 for that i need to restart stream for every one hour. Rekognation.

https://github.com/aws/amazon-kinesis-video-streams-parser-library/issues/77

zhiyua-git, 2020-01-03:
Hi @hameedriswan7 ,

Rather than stop PutMedia every hour, sound like you would want to restart on GetMedia side every hour, in that way, you video can be continuously sending. 

You can do the disconnect on GetMedia using https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/57063da12cbfbfa95abbf04bb8d472fe559e7497/src/main/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExample.java#L126-L134.