awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #19: Not able to compile on Ubuntu

rnzsgh, 2017-12-28:
Greetings,

I am not able to compile. Please let me know if I have done something silly.

Thanks,

Ryan

Host: Linux ip-172-31-67-232 4.4.0-1043-aws #52-Ubuntu SMP Tue Dec 5 10:49:06 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4macros'.
libtoolize: copying file 'm4macros/libtool.m4'
libtoolize: copying file 'm4macros/ltoptions.m4'
libtoolize: copying file 'm4macros/ltsugar.m4'
libtoolize: copying file 'm4macros/ltversion.m4'
libtoolize: copying file 'm4macros/lt~obsolete.m4'
configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
configure.ac:2457: warning: PKG_PROG_PKG_CONFIG is m4_require'd but not m4_defun'd
m4macros/gtk-doc.m4:7: GTK_DOC_CHECK is expanded from...
configure.ac:2457: the top level
configure.ac:88: error: possibly undefined macro: AC_DEFINE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1

zhiyua-git, 2017-12-28:
Thanks for reaching out!

All the prerequisites need to be installed before running install script. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#install-steps-for-ubuntu-17x

For your case, can you verify that you have pkg-config installed by running "pkg-config --version 0.29.1"? If you don't have it installed, you can fix it by running "sudo apt-get install pkg-config".

Let me know if this solves your issue. Thanks.

rnzsgh, 2017-12-28:
Those steps worked and it compiled, thanks for the quick response.

I TL;DRd the docs and just tried to install what was listed under the "Build Dependencies" section.

There is a small typo in cmake install command - it says: "isnstall"

Originally, I ran it by typing: ./kinesis-video-native-build/install-script

This resulted in an error about the directory not being a part of the CMakeLists.txt, but it ran fine when I changed directory to kinesis-video-native-build and ran ./install-script

zhiyua-git, 2017-12-28:
Glad to hear it worked for you!

We will fix the typo and keep updating our instructions. Thanks.

closed 2017-12-28 by zhiyua-git