awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #20: Failed to stream on OSX

jonlindeheim, 2018-01-03:
Hi,

I'm trying to run kinesis_video_gstreamer_sample_app on my MacBook Pro with High Sierra (10.13.1). But I get this error:
 
```
getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream my_test_stream is Ready.
DEBUG - Stream is ready
Error received from element source-actual-src-avfvide: Internal data stream error.
Debugging information: gstbasesrc.c(2939): void gst_base_src_loop(GstPad *) (): /GstPipeline:test-pipeline/GstAutoVideoSrc:source/GstAVFVideoSrc:source-actual-src-avfvide:
streaming stopped, reason not-negotiated (-4)
```

Any ideas where I need to start troubleshooting?

/Jon

unicornss, 2018-01-03:
Hi Jon,

Could you please check if the GStreamer pipeline is able to stream from the built-in webcam?

For example, the following pipeline works in MacBook Pro with Sierra 10.12.6.

`gst-launch-1.0 avfvideosrc  ! video/x-raw,width=1280,height=720,framerate=30/1  ! osxvideosink` 

Thanks
-SS

jonlindeheim, 2018-01-04:
Thanks for the answer @unicornss!
I get the following output when running the gst-launch command:

```
Setting the pipeline to PAUSED... (translated to english)
Illegal instruction: 4
```

I think I saw some warnings when running the install-script, however i got the "Success" at the end. I should maybe investigate those warnings...

Have you guys tried it with a 2017 MacBook Pro?

jonlindeheim, 2018-01-04:
I got this to work

`gst-launch-1.0 videotestsrc ! "video/x-raw,width=1280,height=720,framerate=30/1" ! autovideosink`

So I will try to tweak the avfvideosrc element..

closed 2018-01-04 by zhiyua-git