awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #22: security token included in the request is invalid

efflorez, 2018-01-11:
Hello , i'm working in ubuntu 16.04  , trying to execute ./start but i have the following error
```
36d65efdd50075e09
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    X-Amz-Date: 20180111T235218Z
2018-01-11 18:52:23 [139842008155904] DEBUG - describeStream response: 
2018-01-11 18:52:23 [139842008155904] INFO - Describe stream did not find the stream ebenezer in Kinesis Video (stream will be created)
2018-01-11 18:52:23 [139842008155904] 
describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
2018-01-11 18:52:23 [139842008155904] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 25192 Expiration: 0
2018-01-11 18:52:24 [139842016548608] WARN - HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AccessKey/20180111/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=2cd756ea587308cb47c3a18aaadbe064fd6dbfd50b2c97a4fef411a6ebdd5ca0
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    X-Amz-Date: 20180111T235224Z
2018-01-11 18:52:24 [139842016548608] DEBUG - describeStream response: {"message":"The security token included in the request is invalid."}
2018-01-11 18:52:24 [139842016548608] INFO - Describe stream did not find the stream ebenezer in Kinesis Video (stream will be created)
2018-01-11 18:52:24 [139842016548608] 
describeStreamResultEvent(): Describe stream result event.ERROR - Submitting event result for stream: 17557258 failed with: 1375731728
2018-01-11 18:52:24 [139842016548608] ERROR - Reporting stream error. Errored timecode 0 with status code 1375731728
2018-01-11 18:52:24 [139842016548608] ERROR - Failed to create Kinesis Video Stream - timed out.
2018-01-11 18:52:48 [139842209044288] INFO - Freeing Kinesis Video Stream ebenezer
2018-01-11 18:52:48 [139842209044288] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.DEBUG - streamDataAvailableHandler invoked
2018-01-11 18:52:48 [139842209044288] DEBUG - streamClosedHandler invoked
2018-01-11 18:52:48 [139842209044288] unknown file: Failure
C++ exception with description "Failed to create Kinesis Video Stream - timed out." thrown in the test body.
INFO - Tearing down test: ProducerApiTest
2018-01-11 18:52:48 [139842209044288] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
commonHeapDebugCheckAllocator(): Heap is initialized
commonHeapDebugCheckAllocator(): Heap limit: 				134217728
commonHeapDebugCheckAllocator(): Heap size: 				0
commonHeapDebugCheckAllocator(): Number of allocations: 				0
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7f2f82b9b014
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:	0x7f2f82b9b014		size:	134217695
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.[  FAILED  ] ProducerApiTest.create_produce_stream (30003 ms)
[----------] 1 test from ProducerApiTest (30003 ms total)

[----------] Global test environment tear-down
[==========] 308 tests from 30 test cases ran. (32829 ms total)
[  PASSED  ] 307 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ProducerApiTest.create_produce_stream

 1 FAILED TEST
```

MushMal, 2018-01-12:
@efflorez your security credentials might be missing or invalid. The test application gets the credentials through the environment variables. Here is the relevant info from the Readme.md:

Define AWS_ACCESS_KEY_ID and AWS_SECRET_ACCESS_KEY environment variables with the AWS access key id and secret key: export AWS_ACCESS_KEY_ID={AccessKeyId} export AWS_SECRET_ACCESS_KEY={SecretAccessKey}

optionally, set AWS_SESSION_TOKEN if integrating with temporary token and AWS_DEFAULT_REGION for the region other than us-west-2

Please let me know whether this works.

However, if you have specified your credentials properly and still have auth issues then please take a look at the credential rights in the IAM policy console and make sure you have the right settings there.

efflorez, 2018-01-12:
thanks for your answer,now  i get this issue , i think that there are a request problem 
```createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
2018-01-12 10:23:07 [139687916152640] DEBUG - Awaiting for the stream to become ready...
2018-01-12 10:23:07 [139687916152640] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 4984 Expiration: 0
2018-01-12 10:23:07 [139687715264256] ERROR - curl perform failed for url https://kinesisvideo.us-east-1.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5516 milliseconds
2018-01-12 10:23:12 [139687715264256] WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJWJGEQVYE2XKA7BA/20180112/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=798aba43ff5845ec7acb7eb1f6e94dc00d0ac3528bd9d770a3a74b2c3f85fbe9
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    X-Amz-Date: 20180112T152307Z
2018-01-12 10:23:12 [139687715264256] DEBUG - describeStream response: 
2018-01-12 10:23:12 [139687715264256] INFO - Describe stream did not find the stream ebenezer in Kinesis Video (stream will be created)
2018-01-12 10:23:12 [139687715264256] 
describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
2018-01-12 10:23:12 [139687715264256] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 4990 Expiration: 0
```

MushMal, 2018-01-12:
This still looks like an issue with your credentials. 
* Are you using temporary credentials? If so, you will need to specify the session token too
* Are your temp credentials expired?
* Do you have the appropriate rights setup?

efflorez, 2018-01-12:
thanks , wich polices should be active ? i have the following  ----AmazonKinesisVideoStreamsFullAccess-----AmazonEC2FullAccess

MushMal, 2018-01-15:
@efflorez, from what I see you have a timeout:
2018-01-12 10:23:07 [139687715264256] ERROR - curl perform failed for url https://kinesisvideo.us-east-1.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5516 milliseconds
2018-01-12 10:23:12 [139687715264256] WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJWJGEQVYE2XKA7BA/20180112/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=798aba43ff5845ec7acb7eb1f6e94dc00d0ac3528bd9d770a3a74b2c3f85fbe9
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    X-Amz-Date: 20180112T152307Z

The SDK is configured to timeout after ~5 seconds of no response. Can you check you have external gateway configured properly? Do you have internet access on the device? Are the ports enabled (443 port for standard https)?

efflorez, 2018-01-16:
i test it on raspberry pi and it works good (big latency and the cores was on the top ) but in my laptop i couldnt connect

closed 2018-01-16 by efflorez

iaashish, 2019-01-10:
@MushMal - I'm using a temporary token us explanation has helped me