awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #25: can sample program send streaming from file?

hryktrd, 2018-01-17:
I use kinesis_video_gstreamer_sample_app and I succeed to send video stream from raspberyyPi3 + camera v2.
I want to send video stream from named pipe forexample,
```sh
mkfifo /tmp/input
myprogram  > /tmp/input
kinesis_video_gstreamer_sample_app streamName /tmp/input
```

Can I do this by modifiying kinesis_video_gstreamer_sample_app.cpp?

unicornss, 2018-01-17:
Glad you are able to send to Kinesis Video Stream from Raspberry Pi. 

Are you planning to create the video stream from `myprogram` and store it in `/tmp/input` then send to Kinesis Video Stream? If so, then please take a look at unit tests in particular ProducerAPITest.cpp. 

Please note that you can view the video playable in the console then the stream has to be h.264 encoded before sending through `putFrame` in Kinesis Video Streams Producer SDK. 

Hope this helps.

Thanks
-SS

MushMal, 2018-01-17:
@hryktrd, as @unicornss mentioned, you can take a look at the samples and see how a media pipeline produced frames can be pushed to the KVS. If you have a random pipe with the video data (or any other data) then in the application you have to handle two things:

* Have a way to determine the frame start/size so you can operate at a frame granularity instead of a byte stream
* Have a way to extract the frame timestamps. If it's just a stream of video, you can configure your KVS stream to timestamp the frames as they arrive.
* Have a way to determine whether the frame is a key-frame or not. Perhaps, examining the NAL units to see whether it's an I-frame or have another way of extracting this information.

Best regards!

unicornss, 2018-02-19:
Are you planning to use PI to run the producer SDK? Are you trying to get the videos saved into a folder before streaming to KVS?  What kind of video analytics are you trying to build from those video streams?

unicornss, 2018-03-02:
Closing this for now. Please reopen if the support for this use case/request/issue is required.

closed 2018-03-02 by unicornss