awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #26: Failed to allocate heap

dpmcgarry, 2018-01-18:
After compiling on a pi 2 B+ i was able to run all unit tests.
When i run the streamer (I exported variables for AWS Access / Secret keys and region) it gives the following error:
INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
aivHeapInit(): Failed to allocate heap with limit size 536870912
heapInitialize(): Failed to initialize native heap.
heapRelease(): Freeing native heap.
Segmentation fault

MushMal, 2018-01-18:
@dpmcgarry, the sample application tries to allocate 512MB of storage which apparently is not available on your system. The code throws an exception which is not caught as it's a sample app and we haven't yet hardened it for these types of failures. Generally, in a real-life application, one would try to determine how much storage would be needed and will handle the failures gracefully.

In this case I would recommend changing the allocation from 512MB to a much smaller (single stream PI with limited bandwidth and good networking would require far less storage).

instead of 
device_info.storageInfo.storageSize = 512 * 1024 * 1024;

please specify
device_info.storageInfo.storageSize = 128 * 1024 * 1024;

We will, however, take a note and update the sample application to have more dynamic allocation.
We will also document the best practice on determining how much storage one would need and how to configure the StreamDefinition better for your needs.

dpmcgarry, 2018-01-18:
Hmmm that doesn't make sense...I have 24G free for my root partition:
pi@pi-dev:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ df -h

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  4.3G   24G  16% /
devtmpfs        182M     0  182M   0% /dev
tmpfs           186M     0  186M   0% /dev/shm
tmpfs           186M  2.7M  183M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           186M     0  186M   0% /sys/fs/cgroup
/dev/mmcblk0p1   41M   21M   21M  52% /boot
tmpfs            38M     0   38M   0% /run/user/1000

MushMal, 2018-01-18:
The available storage is on eMMC - read file system. The current implementations of the heap are all RAM based. We will be adding a hybrid storage which would utilize the file-backed storage as a spill-over later.

This said, for your PI you will hardly need that large of the storage anyway.

dpmcgarry, 2018-01-19:
Ok now i am getting an error stating unable to set the pipeline to the playing state:

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 28023 Expiration: 30422
INFO - New credentials expiration is 30423

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream demo-stream is Ready.
DEBUG - Stream is ready
Unable to set the pipeline to the playing state.
INFO - Freeing Kinesis Video Stream demo-stream

freeKinesisVideoStream(): Freeing Kinesis Video stream.DEBUG - streamDataAvailableHandler invoked
DEBUG - streamClosedHandler invoked

dpmcgarry, 2018-01-19:
Disregard.  Was able to get it working by running modprobe

closed 2018-01-19 by dpmcgarry