awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #30: Error std::make_unique

evi8899, 2018-02-01:
Hi everyone,

I'm building the environment using ./install-script

And get this error:
[ 76%] Building CXX object CMakeFiles/producer.dir/home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-vid                             eo-pic/src/view/tst/ViewTestFixture.cpp.o
[ 77%] Building CXX object CMakeFiles/producer.dir/home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-vid                             eo-producer/src/KinesisVideoProducer.cpp.o
In file included from /home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideo                             Producer.h:9:0,
                 from /home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideo                             Producer.cpp:1:
/home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:43:17: e                             rror: ‘make_unique’ is not a member of ‘std’
                 std::make_unique<EmptyCredentialProvider>(),
                 ^
/home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:43:57: e                             rror: expected primary-expression before ‘>’ token
                 std::make_unique<EmptyCredentialProvider>(),
                                                         ^
/home/evi/works/Cloud_AI/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:43:59: e                             rror: expected primary-expression before ‘)’ token
                 std::make_unique<EmptyCredentialProvider>(),

Please help me.

Thanks

zhiyua-git, 2018-02-01:
Hi evi8899,

Can you check your GCC version by running g++ --version? This is a C++14 feature which is supported after GCC 5. ([C++14 Language Features](https://gcc.gnu.org/projects/cxx-status.html))

You will need to update your GCC and rebuild to fix this issue.

Thanks,
-ZH

evi8899, 2018-02-01:
I have updated to gcc 5.
Close the issue. Thanks!!

closed 2018-02-01 by evi8899

ACGAshwin, 2018-06-12:
I am currently on GCC/C++ 7.2 yet i still get this error

chehefen, 2018-06-12:
do you have multiple compilers installed? can you try export CC=path/to/gcc-7.2.0 and CXX=path/to/g++-7.2.0? Also remember to delete CMakeCache.txt before running install-script.

ACGAshwin, 2018-06-12:
Thanks Chehefen. That didnt resolve the issue.

Did the following
export CXX=/opt/rh/devtoolset-7/root/usr/bin/g++
export CC=/opt/rh/devtoolset-7/root/usr/bin/gcc

and then removed the CMakeCache.txt from the native build directory and still got the same error

ACGAshwin, 2018-06-12:
configure is able to detect as well 

```
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... /opt/rh/devtoolset-7/root/usr/bin/gcc
checking whether the C compiler works... yes 
```

ACGAshwin, 2018-06-12:
Alternative : I tried compiling by forcing the following definition of make_unique into the DefaultCallbackProvider.h . The file compiled but now i am stuck at another error  error: invalid use of auto’   PServiceCallContext service_call_ctx) -> auto {

`namespace std {
    template<class T> struct _Unique_if {
        typedef unique_ptr<T> _Single_object;
    };

    template<class T> struct _Unique_if<T[]> {
        typedef unique_ptr<T[]> _Unknown_bound;
    };

    template<class T, size_t N> struct _Unique_if<T[N]> {
        typedef void _Known_bound;
    };

    template<class T, class... Args>
        typename _Unique_if<T>::_Single_object
        make_unique(Args&&... args) {
            return unique_ptr<T>(new T(std::forward<Args>(args)...));
        }

    template<class T>
        typename _Unique_if<T>::_Unknown_bound
        make_unique(size_t n) {
            typedef typename remove_extent<T>::type U;
            return unique_ptr<T>(new U[n]());
        }

    template<class T, class... Args>
        typename _Unique_if<T>::_Known_bound
        make_unique(Args&&...) = delete;
}`

unicornss, 2018-06-13:
Hi @ACGAshwin , 

Could you please try

`export PATH=/opt/rh/devtoolset-7/root/usr/bin:$PATH` 

 `rm -rf yourpathtosdk/kinesis-video-native-build/CMakeCache.txt yourpathtosdk/kinesis-video-native-build/CMakeFiles`

before running the `./install-script` ?   

Thanks