awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #32: Illegal Instruction during install-script

nicklee1990, 2018-02-04:
**OS**: OSX High Sierra - 10.13.3
Whilst running the `install-script`, I encountered the following error:

`21171 Illegal instruction: 4  src/bison --version > doc/bison.help.tmp` as well as several messages saying XXXX  has no symbols

It looks as though this [might be a known issue](http://lists.gnu.org/archive/html/bug-bison/2017-12/msg00006.html) with bison on OSX 10.13. I've tried searching for workarounds but haven't been able to find any so far.

Are there any suggestions for a way round this? 

**Detailed output:**
```
...similar statements
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: lib/libbison.a(unistd.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: lib/libbison.a(wctype-h.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: lib/libbison.a(xsize.o) has no symbols
  CCLD     src/bison
  GEN      doc/bison.help
/bin/sh: line 1: 21171 Illegal instruction: 4  src/bison --version > doc/bison.help.tmp
make[2]: *** [doc/bison.help] Error 132
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

nicklee1990, 2018-02-04:
I got past this by just commenting out the bison installation in the install-script after the initial fail since it did actually download bison, but that's not a great workaround!

unicornss, 2018-02-07:
Glad you were able to resolve it. 
Thanks for the note on install-script.. and we will check and resolve if  it still exists  in the High Sierra. Thanks -SS

andreabedini, 2018-02-14:
I have the same issue but I got away using brew and the minimal installer script `min-install-script`

1. `$ brew install cmake gstreamer gst-plugins-good # something else might be needed`
2. `$ ./min-install-script`
3. profit

unicornss, 2018-02-19:
We will add `brew install` instructions separately (apart from the full `install-script`).

closed 2018-02-19 by unicornss