awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #34: Install Script Issue - Ubuntu 17.10

francoisdutoit, 2018-02-16:
The below error occurs up during the install script execution

Makefile:1696: recipe for target 'stage1scan.c' failed
make[2]: *** [stage1scan.c] Segmentation fault (core dumped)
make[2]: Leaving directory '/home/francois/downloads/flex-2.6.4/src'
Makefile:546: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/francois/downloads/flex-2.6.4/src'
Makefile:533: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
francois@francois-VirtualBox:~$

unicornss, 2018-02-16:
Hi @francoisdutoit  We will check this in the Ubuntu version 17.10. In the meantime could you tell us the g++, gcc version in your system? Also, could pl. try upgrading gcc, g++ and try install-script again.

Thanks

francoisdutoit, 2018-02-16:
Hi @unicornss ,

Here are the versions as requested.
g++ (Ubuntu 7.2.0-8ubuntu3) 7.2.0
gcc (Ubuntu 7.2.0-8ubuntu3) 7.2.0

I will try your suggestion and provide some feedback.

Thanks! :)

francoisdutoit, 2018-02-16:
Hi @unicornss ,

The latest version is 7.3 - It's very new - 7.2 should be fine?

unicornss, 2018-02-16:
Hi @francoisdutoit  I am seeing the issue flex 2.6.4 and gcc 7.2.0 https://github.com/westes/flex/issues/219 

Let me try flex 2.6.3 or 2.6.1 and let update you soon.

francoisdutoit, 2018-02-16:
Hi @unicornss,
When I check the version of flex it shows 2.6.1 - however...when running the install script it does not find flex and downloads 2.6.4.... Could this be the issue?

unicornss, 2018-02-16:
It seems the GCC version 7.2 is not able to compile the 2.6.4. In you case since you already have flex 2.6.1 you could comment out the flex section in the install-script and tell us how it goes. 

 To make it self-contained install-script creates a separate folder to download and compile the open source dependencies (amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local)

francoisdutoit, 2018-02-17:
Ok - so getting a bit further now - I commented out the flex section as well as any other section that points to already installed packages ( the ones I installed as per Install steps for Ubuntu 17.x ) But now getting an error as below -

[ 73%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/view/tst/ViewApiTest.cpp.o
[ 74%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/view/tst/ViewTestFixture.cpp.o
[ 75%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o
In file included from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Auth.h:3:0,
                 from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/AwsV4Signer.h:3,
                 from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:6,
                 from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.h:9,
                 from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp:1:
**/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Request.h:8:10: fatal error: curl/curl.h: No such file or directory
 #include <curl/curl.h>**
          ^~~~~~~~~~~~~
compilation terminated.
CMakeFiles/producer.dir/build.make:1886: recipe for target 'CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o' failed
make[2]: *** [CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/producer.dir/all' failed
francois@francois-VirtualBox:~/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ curl --version
curl 7.55.1 (x86_64-pc-linux-gnu) libcurl/7.55.1 OpenSSL/1.0.2g zlib/1.2.11 libidn2/2.0.2 libpsl/0.18.0 (+libidn2/2.0.2) librtmp/2.3
Release-Date: 2017-08-14
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp smb smbs smtp smtps telneffrafrancois@frafrfrancfrfrancoisfranfrancois@francois@franfrancois@francfrancois@francois-VirtualBox:~/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$

francoisdutoit, 2018-02-17:
Hi @unicornss,

regarding above - also checked my curl version...

francois@francois-VirtualBox:~$ curl --version
curl 7.55.1 (x86_64-pc-linux-gnu) libcurl/7.55.1 OpenSSL/1.0.2g zlib/1.2.11 libidn2/2.0.2 libpsl/0.18.0 (+libidn2/2.0.2) librtmp/2.3
Release-Date: 2017-08-14
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP UnixSockets HTTPS-proxy PSL

unicornss, 2018-02-19:
Hi @francoisdutoit, you don't need to comment any sections in the install-script. The only change in ubuntu17.04 requires the Flex to be 2.6.3 (instead of 2.6.4). Pl. check out the latest install-script and should work with Flex 2.6.3.   Please see the earlier comment on how to set Flex for 2.6.3 version.  Please let us know how it goes. Thanks.

francoisdutoit, 2018-02-19:
Hi @unicornss,

Thanks - as mentioned above - the script is now running fine up to the point where I get the error mentioned in my last comment -  Please see below...

"Ok - so getting a bit further now - I commented out the flex section as well as any other section that points to already installed packages ( the ones I installed as per Install steps for Ubuntu 17.x ) But now getting an error as below -

[ 73%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/view/tst/ViewApiTest.cpp.o
[ 74%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/view/tst/ViewTestFixture.cpp.o
[ 75%] Building CXX object CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o
In file included from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Auth.h:3:0,
from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/AwsV4Signer.h:3,
from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:6,
from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.h:9,
from /home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp:1:
**/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Request.h:8:10: fatal error: curl/curl.h: No such file or directory
#include <curl/curl.h>**
^~~~~~~~~~~~~
compilation terminated.
CMakeFiles/producer.dir/build.make:1886: recipe for target 'CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o' failed
make[2]: *** [CMakeFiles/producer.dir/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/producer.dir/all' failed
francois@francois-VirtualBox:/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ curl --version
curl 7.55.1 (x86_64-pc-linux-gnu) libcurl/7.55.1 OpenSSL/1.0.2g zlib/1.2.11 libidn2/2.0.2 libpsl/0.18.0 (+libidn2/2.0.2) librtmp/2.3
Release-Date: 2017-08-14
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp smb smbs smtp smtps telneffrafrancois@frafrfrancfrfrancoisfranfrancois@francois@franfrancois@francfrancois@francois-VirtualBox:/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$"

francoisdutoit, 2018-02-19:
Hi @unicornss,
 If I only change that bit in the install script ( as per what you suggested ) I get the following 

Checking util-linux at /home/francois/downloads/local/lib/libmount.dylib/.so
m4 not found. Installing
curl: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory

So if I comment out all the sections that refer to libraries that I already have installed - everything runs fine up to the point where I get the below error 

**/home/francois/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Request.h:8:10: fatal error: curl/curl.h: No such file or directory
#include <curl/curl.h>**

unicornss, 2018-02-19:
The error you are seeing is because it does not have sources for `curl` . This is used (along with libssl) by Producer SDK to send streams to Kinesis. If these are commented out in the `install-script` then it will not download the curl and ssl dependencies.  

[image]

 I would suggest pull the latest sdk (install-script) and run again. To initiate a clean download and build you can use the troubleshooting instructions ...
```
rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles
```
and run 
```
./kinesis-video-native-build/install-script
``` 
to rebuild and re-link the project. Let us know if you have questions.

francoisdutoit, 2018-02-19:
Ok...So here is my plan now.....
1. Clone repo
2. Update install script to flex 2.6.3
3. Run script
Also  ....I am not going to manually install any of prerequisites......
Ps :  I Also have a fresh ubuntu installtion

francoisdutoit, 2018-02-19:
Thanks again for all the help....lets hope it works this time ☺

unicornss, 2018-02-19:
Perfect. 
Since you have latest Ubuntu installation, you do not need the build cache clean up. 
Thanks for notifying us the Flex version conflict (with GCC)  in the latest Ubuntu 17.10.

unicornss, 2018-02-19:
Great to see it progressed further. Could you try using` automake 1.15.1` as below?

```
#--------aclocal, aclocal-1.15 (hard linked with aclocal), automake, and automake-1.15 (hard linked with automake)--
  if [ ! -f $DOWNLOADS/local/bin/automake ]; then
    echo "automake not found. Installing"
    if [ ! -f $DOWNLOADS/automake-1.15.1.tar.gz ]; then
      cd $DOWNLOADS
      curl -L "https://ftp.gnu.org/gnu/automake/automake-1.15.1.tar.gz" -o "automake-1.15.1.tar.xz"
    fi
    cd $DOWNLOADS
    tar -xvf automake-1.15.1.tar.xz
    cd $DOWNLOADS/automake-1.15.1
    ./configure --prefix=$DOWNLOADS/local  CFLAGS="-I$DOWNLOADS/local/include" LDFLAGS="-L$DOWNLOADS/local/lib"
    make
    make install
fi
```

francoisdutoit, 2018-02-19:
Ok - so I have updated the script - but cant seem to get a clean install going...

cmake version 3.9.1 CMake suite maintained and supported by Kitware (kitware.com/cmake).
Checking openssl at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libssl.a
Checking curl at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libcurl.a
Checking log4cplus at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/liblog4cplus.dylib/.so
Checking gettext at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libgettextlib.dylib/.so
Checking libffi at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libffi.a
Checking pcre at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libpcre.dylib/.so
Checking zlib at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libz.dylib/.so
Checking iconv at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libiconv.dylib/.so
Checking util-linux at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libmount.dylib/.so
Checking glib at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libglib-2.0.dylib/.so
Checking googletest at /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/libgtest.a
automake not found. Installing
curl: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
francois@francois-VirtualBox:~/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp$

francoisdutoit, 2018-02-19:
PS : I did run the below 

rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles

and run

./kinesis-video-native-build/install-script

unicornss, 2018-02-19:
Strange .. can you confirm if the package `pkg-config` is installed on your system? if not, you can install by using the following:
`apt-get install -y pkg-config `

To resolve the shared library not found - you could try building after setting the LD_LIBRARY_PATH environment variable:
`export LD_LIBRARY_PATH=<your_sdk_kinesis_video_native_build_folder_path>/downloads/local/lib:$LD_LIBRARY_PATH`

On a different note, 

- What sort of device(s) are you planning to run the producer SDK?
- What sort of video processing or analytics do you intend to perform?

Thanks

francoisdutoit, 2018-02-20:
For now I'll be running on Ubuntu accessing Onvif IP Camera - so yeah just want to be able to stream from an IP camera....

I did mange to get a new build going - and again the build went further - but failed at the below...
[ 89%] Building CXX object CMakeFiles/KinesisVideoProducerJNI.dir/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp.o
In file included from /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp:5:0:
/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/../kinesis-video-producer-jni/src/include/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.h:14:10: fatal error: jni.h: No such file or directory
 #include <jni.h>                  // Basic native API
          ^~~~~~~
compilation terminated.
CMakeFiles/KinesisVideoProducerJNI.dir/build.make:62: recipe for target 'CMakeFiles/KinesisVideoProducerJNI.dir/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp.o' failed
make[2]: *** [CMakeFiles/KinesisVideoProducerJNI.dir/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/KinesisVideoProducerJNI.dir/all' failed
make[1]: *** [CMakeFiles/KinesisVideoProducerJNI.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

francoisdutoit, 2018-02-20:
PS: francois@francois-VirtualBox:~$ pkg-config --version
0.29.1

unicornss, 2018-02-20:
Hi @francoisdutoit 

Glad you are able to get the open source dependencies compiled for **Ubuntu 17.10.**

As referred in the https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md 
   
you have to **Install Open JDK**: 
`sudo apt-get install openjdk-8-jdk`

**Set JAVA_HOME environment variable:**
` export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/`

**Run the build script:** 
`./install-script`

francoisdutoit, 2018-02-20:
It Built!!! :)

[100%] Linking CXX executable kinesis_video_gstreamer_sample_app
[100%] Built target kinesis_video_gstreamer_sample_app
**********************************************************
Success!!!
**********************************************************
Thanks for all the help and quick response :)

closed 2018-02-20 by francoisdutoit