awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #38: Link Error libssl.so.1.1

francoisdutoit, 2018-02-22:
Hi ,

When building the demo project - at first I got the error : liblog4cplus-1.2.so.5: cannot open shared object file....after which - as suggested in installation instruction was to remove CMakeCache.txt  and the CMake folder and run the install script again - which seem to have worked except now I'm getting the same error but for libssl.so.1.1. I did try repeating the steps but I still get the below error....

ERROR / KinesisVideo: 2018-02-22T20:09Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory: /home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libKinesisVideoProducerJNI.so: libssl.so.1.1: cannot open shared object file: No such file or directory

How can I resolve this issue?

Thanks!

unicornss, 2018-02-22:
Looks like you are trying the Java SDK - Pl. check https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/18 . 
can you also try `export LD_LIBRARY_PATH=/home/francois/Documents/Amazon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH`

francoisdutoit, 2018-02-22:
Sorry - I will open this issue there - been hopping between the two repos.

closed 2018-02-22 by francoisdutoit