awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #39: Describe stream result event.ERROR

antoniolobefaro, 2018-02-27:
describeStreamResultEvent(): Describe stream result event.ERROR - Submitting event result for stream: 94040122179894 failed with: 1375731729
ERROR - Reporting stream error. Errored timecode: 0 Status: 1375731729
ERROR - Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.

What is this Error?
Thanks

unicornss, 2018-02-27:
Please see if the steps to resolution outlined here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/22 help - in terms of connectivity and/or credentials issues. 

A detailed log (after removing credentials) would be helpful to debug the issue.

MushMal, 2018-02-27:
@antoniolobefaro, could you please mention whether this is a persistent error? Could you include more logs to see if the SDK retry results in the same error (which in this case a generic timeout). Also, please make sure your device has a network connection and there are no blocking proxies in between.

antoniolobefaro, 2018-02-28:
Hi,
I checked the rtsp url connection and I had problem with that. after I fixed that problem everything is ok. Now my rtsp streaming is on kinesis video stream.
Thanks

unicornss, 2018-02-28:
Glad to hear that you are able to successfully stream from RTSP url to Kinesis Video Stream. 

-SS

unicornss, 2018-02-28:
Hi @antoniolobefaro 
It would be great if you could provide us the **camera type** and the **OS version** where you are running the streaming from?  Thanks -SS

antoniolobefaro, 2018-03-01:
I deployed the environment in ubuntu 17.04 and I'm able to use the integrated camera and a DLink DCS942L. 
The same environment has been deployed on an ARM system (not rpi) and also it is able to stream the DLink camera. 
the main difficult is to set the environment configuration, the different libraries (and the correct version), dependencies and path. 
@VincTric is working to optimize the video streaming.

unicornss, 2018-03-01:
Thanks @antoniolobefaro  
We are revising the install documentation to be easily followed for different OS - MacOS, Ubuntu and in both cases, we want to provide the flexibility to use either  _build from sources_ or  _package managers_. There is also a separate document we are working on just for Raspberry PI which will be published soon. If you have any specific notes on  ARM installation  please let us know. 

-SS

closed 2018-03-01 by unicornss

himadrii27, 2024-06-25:
can 

> Hi, I checked the rtsp url connection and I had problem with that. after I fixed that problem everything is ok. Now my rtsp streaming is on kinesis video stream. Thanks

i also established the rtsp connection but still its not working. Can you lemme know how did you establish your rtsp connection and which os you are using?