awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #40: Curl SSL certificate problem on OS X

FOXNEOAdvancedTechnology, 2018-02-27:
I built amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build on OS X.
 
However now when I run “./kinesis_video_gstreamer_sample_app VidTransDemo” I get this:
 
`ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Peer certificate cannot be authenticated with given CA certificates: SSL certificate problem: unable to get local issuer certificate`
 
I realize that Kinesis Video Streams Produicer SDK for C++ needs to establish trust with the backend service through TLS and that there is a .pem file for this (SFSRootCAG2.pem), but although there are online instructions for installing the .pem file in Linux, I can’t find any info on how to install this properly on OS X.

Or maybe the curl could be configured to optionally use --insecure ?

Thanks!

MushMal, 2018-02-28:
@FOXNEOAdvancedTechnology great to hear you compiled the demo app with no issues. Indeed, we do have TLS security configured in the CURL client. As you noticed correctly, this works by adding the cert to the appropriate store in the linux. Mac OS x is very similar to it. For example, a lot of the systems do come preconfigured with a set of well-known certificate fingerprints in `/etc/ssl/cert.pem` file. You can add the content of the SFSRootCAG2.pem to the end of this file.

FOXNEOAdvancedTechnology, 2018-02-28:
My OS X (El Capitan V. 10.11.6) had no /etc/ssl directory.  Even when I add one with SFSRootCAG2.pem as /etc/ssl/cert.pem, I get the same authentication problem with curl.

MushMal, 2018-02-28:
@FOXNEOAdvancedTechnology could you please reboot the computer and then ensure the `/etc/ssl/cert.pem` file has a read access for the account you are using? `ls -l`.  

A couple of other things to try to shed more light into the issue:
`curl https://www.amazon.com` 
`curl https://kinesisvideo.us-west-2.amazonaws.com`
and see if curl is able to get at least a response.

Please respond so we can help you debug the networking issue better.

FOXNEOAdvancedTechnology, 2018-02-28:
OK, rebooted MacBook with OS X El Capitan 10.11.6.

$pwd
/etc/ssl
$ ls -l
total 8
-rw-r--r--@ 1 root  wheel  1424 Feb 27 12:03 cert.pem

$ curl https://www.amazon.com [returns a ton of HTML as expected]

$ curl https://kinesisvideo.us-west-2.amazonaws.com
<MissingAuthenticationTokenException>
  <Message>Missing Authentication Token</Message>
</MissingAuthenticationTokenException>

bdhandap, 2018-03-01:
@FOXNEOAdvancedTechnology I apologize for the inconvenience. Looks like the curl command is unable to refer the local cert path. Can you please navigate to `<producer_sdk_path>/kinesis-video-native-build/downloads/local/bin` and run the following commands and paste the output here?
 
`./curl-config –ca`

`./curl https://kinesisvideo.us-west-2.amazonaws.com --verbose`

NOTE: Please use the curl binary from the `downloads/local/bin` instead of the system default curl.

Thanks
Babu

bdhandap, 2018-03-01:
Also can you run `otool -L kinesis_video_gstreamer_sample_app` inside kinesis-video-native-build folder and paste the output? We had a similar issue recently where the `kinesis_video_gstreamer_sample_app` was referring to the system default libcurl instead of the one which we download and build as part of the `install-script`.

FOXNEOAdvancedTechnology, 2018-03-01:
I'll note that I did get kinesis_video_gstreamer_sample_app running on Raspberry Pi on Raspian with the Pi cam, but still no luck on OS X.

```
$ otool -L kinesis_video_gstreamer_sample_app 
kinesis_video_gstreamer_sample_app:
	@rpath/libproducer.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstreamer-1.0.0.dylib (compatibility version 1204.0.0, current version 1204.0.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstapp-1.0.0.dylib (compatibility version 1204.0.0, current version 1204.0.0)
	/[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgobject-2.0.0.dylib (compatibility version 5401.0.0, current version 5401.2.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libglib-2.0.0.dylib (compatibility version 5401.0.0, current version 5401.2.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus-1.2.5.dylib (compatibility version 7.0.0, current version 7.4.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local//lib/libcrypto.1.1.dylib (compatibility version 1.1.0, current version 1.1.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local//lib/libssl.1.1.dylib (compatibility version 1.1.0, current version 1.1.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/googletest-release-1.8.0/googletest/libgtest.dylib (compatibility version 0.0.0, current version 0.0.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/googletest-release-1.8.0/googletest/libgtest_main.dylib (compatibility version 0.0.0, current version 0.0.0)
	[redacted]/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl.4.dylib (compatibility version 10.0.0, current version 10.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
```

bdhandap, 2018-03-02:
That's exactly the problem `./curl-config --ca` should return `/etc/ssl/cert.pem`.  Basically curl is not able to refer the cert ca bundle and thus fails validation. The `curl` which was downloaded as part of the install-script uses libssl and other dependencies from the local folder (<producer_sdk_path/kinesis-video-native-build/downloads/local/lib). Looks like your curl was not build properly. Can you please reinstall curl by following below steps? This should fix the issue. 

```
  rm <producer_sdk_path/kinesis-video-native-build/downloads/local/lib/libcurl*
  rm <producer_sdk_path/kinesis-video-native-build/downloads/local/bin/curl*
  cd <producer_sdk_path/kinesis-video-native-build/downloads/curl-7.57.0
  export DOWNLOADS=<producer_sdk_path>/kinesis-video-native-build/downloads
  make clean
  ./configure --prefix=$DOWNLOADS/local/ --enable-dynamic --disable-rtsp --disable-ldap --without-zlib --with-ssl=$DOWNLOADS/local/ --with-ca-bundle=/etc/ssl/cert.pem
  make
  make install
```

By default the curl library should pick up the default path /etc/ssl/cert.pem but looks like its not doing that it in all cases. I'll add this configure option in install-script so that it doesn't happen again.

FOXNEOAdvancedTechnology, 2018-03-03:
That seems to have solved the problem, thanks!

closed 2018-03-03 by FOXNEOAdvancedTechnology

MushMal, 2018-04-09:
@dragosnicolae5555 can I ask you what you are trying to accomplish? The public APIs use AWS Sig V4 authentication aside from using TLS 1.2

https://docs.aws.amazon.com/AmazonS3/latest/API/sig-v4-authenticating-requests.html