awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #41: black screen on kinesis video stream

VincTric, 2018-02-28:
Hi all,
i'm using a Sricam SP012 ipcam (720px, rtsp, h.264) to view its live stream on a kinesis video stream, but even if the application provided in the sdk seems to run correctly i don't see the video in the kinesis video stream preview, only a "live black screen".
I run this:
./kinesis_video_gstreamer_sample_rtsp_app rtsp://<user>:>password>@<ip>:554/onvif1 <stream_name>

Any ideas about this issue? Maybe i must change some parameter in the code inside kinesis_video_gstreamer_sample_rtsp_app.cpp file? For example in this stream definition:
    auto stream_definition = make_unique<StreamDefinition>(stream_name,
                                                           hours(2),
                                                           &tags,
                                                           "",
                                                           STREAMING_TYPE_REALTIME,
                                                           "video/h264",
                                                           milliseconds::zero(),
                                                           seconds(2),
                                                           milliseconds(1),
                                                           true,
                                                           true,
                                                           false,
                                                           true,
                                                           true,
                                                           true,
                                                           0,
                                                           30,
                                                           4 * 1024 * 1024,
                                                           seconds(120),
                                                           seconds(40),
                                                           seconds(30),
                                                           "V_MPEG4/ISO/AVC",
                                                           "kinesis_video",
                                                           nullptr,
                                                           0);
Thanks.

MushMal, 2018-02-28:
@VincTric could you please attach the logs so we can see what's going on. It's quite possible that there are other issues preventing the streaming - for example auth/gstreamer pipeline creation/etc.. It's hard to say what's going on without the logs. 

In order to help you with the parameters, it would be great if you could describe your scenario - what are you trying to accomplish after you've ingested the stream - this will help with fine tuning the parameters for your application. The sample application uses default values which might not be the most optimal for your scenario.

VincTric, 2018-03-01:
@MushMal attached you can find the logs you requested.

The SRICAM log is referred to the black screen issue, the DLINK log (model DCS-942L, 640px, rtsp, h.264) is referred to another ipcam by which I'm able to view the video on aws but sometimes gives me error.
 
At the moment my intention is only to send a video in real time on aws, and then be able to watch that video on remote devices.

[log kinesis video stream SRICAM SP012.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1770740/log.kinesis.video.stream.SRICAM.SP012.txt)
[log kinesis video stream DLINK DCS-942L.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1770741/log.kinesis.video.stream.DLINK.DCS-942L.txt)

unicornss, 2018-03-01:
@ VincTric can you please provide the model for SRICAM camera and its supported RTSP stream formats?  Thanks.

MushMal, 2018-03-01:
@VincTric the first log referring to SRICAM has a very short section and in that section I see that the SDK is able to establish the connection and the GStreamer is initialized properly and moved to "PLAYING" state. I see the frames are being produced and are being streamed. We get BUFFERING/RECEIVED/PERSISTED ACKs from the service - all good. I am not sure whether there are errors encountered later - could you please include longer snippet of log? If there are no errors in the log then it's possible that the camera doesn't produce proper H264 elementary stream or it's in a different format (separate NAL adaptation flags should be used) - same for Codec Private Data bits from the elementary stream. Please include the make/model of the device and attach longer logs.

The seconds log for DLINK indicates a proper initialization of the SDK and GStreamer, streaming for a while and then we encounter error 0x30000005. This error indicates that the frames timestamp (decoding timestamp) is less than the previous frames timestamp + frame duration.

The sample application uses 20ms frame durations. In your case, your camera pipeline is not stable and produces jitter. As a result, the case where it fails the dts of the previous frame is 31091134154 and the dts of the frame that fails is 31109001781 which is around 17ms later. 

Here are suggestions to try:

1) Please include detailed description of what you mean the first camera playback is not working. What do you see in the console? Do you see a spinner which later results in "No fragments found"? Or you see a black screen being played? Attach a snapshot so we can understand whether it's just a dark background that's being streamed.
2) Attach full logs for the first camera so we can see if there are any errors on the producer side like with the second camera.
3) Investigate as to why your camera produces such a jitter. This might be due to high resolution selection or poor network connection to the camera.
3) Set the frame size to be less than 20 ms. This might a temporary solution that can work OK for your case.  Modify kinesis_video_gstreamer_sample_rtsp_app.cp::create_kinesis_video_frame function. LN 155. Specify frame->duration = 10 * HUNDREDS_OF_NANOS_IN_A_MILLISECOND;
See of this helps.

VincTric, 2018-03-03:
@MushMal I changed the duration in the code as you suggested, but nothing, the result is always the same: a black screen; as you can see, from the log files and screenshot it seems that the stream arrive to kinesis video stream object, but the result is however a black view (see attached screenshot).
While I'm sending the stream i see the screenshot1 and when i stop the stream i see (correctly) the screenshot2.

Thanks for your help.

[kinesis SRICAM.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1777267/kinesis.SRICAM.txt)
[image]
[image]

VincTric, 2018-03-03:
@unicornss the cam model is SRICAM SP012, it supports rtsp stream, its parameters are:
Resolution: 720P(1280 x 720), VGA(640 x 360), QVGA(320 x 180). 
Compression: H.264 
frame rate: 25fps max

Thanks.

alopezbnt, 2018-03-10:
I have the same problem

unicornss, 2018-03-12:
@alopezbnt   Are you also using the SRICAM SP012 camera?  Could you provide the logs and screen shot of the console ... in the scenario posted by @VincTric ,  the logs show the stream is sent to KVS but not viewable in the console.  Just wanted to see if you are also seeing the same issue with this camera.

unicornss, 2018-04-06:
@VincTric @alopezbnt  
We tried the SRICAM SP012 and did confirm that the RTSP stream is _not consistently viewable from this model_.  Sometimes we do the stream is viewable fine but at other times we are seeing color plane issues from the stream. It did not work when we tried to view the stream in both VLC and streamed to KVS at the same time. 
If we get additional details on what settings change is required for consistent streams from this camera model, we will update you both.

A partial list of cameras where we have seen working consistently in our tests (successfully streaming continuously for months) are:

- HIKVision 3MP IP Camera DS-2CD2035FWD-I
- Vivotek FD9371 – HTV/EHTV
- Vivotek IB9371 – HT
- VStarcam 720P WiFi IP Camera (TCP)
- Panasonic WV model camera
- Sony SNC model camera
- Ipccam Security Surveillance IP Camera 1080P

Thanks
SS

MushMal, 2018-04-11:
Folks, is there any update on this issue? I am eager to learn and help. If we are OK then could I ask you to close the issue please?

unicornss, 2018-04-11:
Closing this issue for _now_ as the RTSP stream from DLink DCS and other list of cameras are consistently available for streaming. If specific support is required for SRICAM model please reopen. Thanks

closed 2018-04-11 by unicornss

GBoucher42, 2018-05-19:
I have the same issue using the android sdk with a LG V20, android 7.0.

BackFacing camera, codec: h264, and 320x240 resolution.

The stream will work 1 time out of ten.

This is what my Android studio loogs look like

V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@93cec5b
V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@93cec5b
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_createKinesisVideoStream(): Enter
I/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_createKinesisVideoStream(): Creating Kinesis Video stream.
                  createKinesisVideoStream(): Creating Kinesis Video Stream.
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_createKinesisVideoStream(): Leave
D/KinesisVideoAndroidClient: calling describe stream: {StreamName: digigrave,}
D/KinesisVideoAndroidClient: describe stream result: {StreamInfo: {DeviceName: android-client-library,StreamName: digigrave,StreamARN: arn:aws:kinesisvideo:us-west-2:452560998706:stream/digigrave/1526518946571,MediaType: video/h264,KmsKeyId: arn:aws:kms:us-west-2:452560998706:alias/aws/kinesisvideo,Version: xyCIsyMMIVtsu8G2YzOt,Status: ACTIVE,CreationTime: Wed May 16 21:02:26 EDT 2018,DataRetentionInHours: 48}}
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_describeStreamResultEvent(): Enter
I/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_describeStreamResultEvent(): Describe stream event for handle 0x0000007b6d927c40.
                  describeStreamResultEvent(): Describe stream result event.
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_describeStreamResultEvent(): Leave
D/KinesisVideoAndroidClient: calling tag resource: {StreamARN: arn:aws:kinesisvideo:us-west-2:452560998706:stream/digigrave/1526518946571,Tags: {device=Test Device, stream=Test Stream}}
D/KinesisVideoAndroidClient: tag resource result: {}
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_tagResourceResultEvent(): Enter
I/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_tagResourceResultEvent(): tag resource event for handle 0x0000007b6d927c40.
                  tagResourceResultEvent(): Tag resource result event.
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_tagResourceResultEvent(): Leave
D/KinesisVideoAndroidClient: calling get data endpoint: {StreamName: digigrave,APIName: PUT_MEDIA}
D/KinesisVideoAndroidClient: get data endpoint result: {DataEndpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com}
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingEndpointResultEvent(): Enter
I/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingEndpointResultEvent(): get streaming endpoint event for handle 0x0000007b6d927c40.
                  getStreamingEndpointResultEvent(): Get streaming endpoint result event.
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingEndpointResultEvent(): Leave
D/KinesisVideoAndroidClient: Refreshing credentials
D/AWSRefreshingCognitoIdentityProvider: Storing the Refresh token in the loginsMap.
D/CognitoCachingCredentialsProvider: Saving credentials to SharedPreferences
D/KinesisVideoAndroidClient: Returning session credentials with expiration Sun Aug 17 02:12:55 EST 292278994
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingTokenResultEvent(): Enter
I/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingTokenResultEvent(): get streaming token event for handle 0x0000007b6d927c40.
                  getStreamingTokenResultEvent(): Get streaming token result event.
D/KinesisVideoAndroidClient: Stream digigrave is ready
V/platform-utils: Java_com_amazonaws_kinesisvideo_producer_jni_NativeKinesisVideoProducerJni_getStreamingTokenResultEvent(): Leave
I/AndroidCameraMediaSource: encoding starting
D/MediaCodec: Mediacodec create pid=23568
V/LGCodecAdapter: LG Codec Adapter start
V/LGCodecAdapter: load libLGCodecOSAL library
I/Choreographer: Skipped 81 frames!  The application may be doing too much work on its main thread.
V/LGCodecOSAL: Just Validatation check function
I/ACodec: Now uninitialized
I/ACodec: onAllocateComponent
E/libc: Access denied finding property "ro.hwui.texture_cache_size"
I/OMXClient: MuxOMX ctor
I/ACodec: [OMX.qcom.video.encoder.avc] Now Loaded
D/EncoderFactory: video/avc output 320x240 @46080
                  format: {width=320, height=240, bitrate=46080, mime=video/avc, frame-rate=20, i-frame-interval=2, color-format=21}
V/LGCodecAdapter: called getLGCodecSpecificData
V/LGCodecOSAL: Called LGgetCodecSpecificDataMSG
W/IInputConnectionWrapper: finishComposingText on inactive InputConnection
E/ACodec: [OMX.qcom.video.encoder.avc] storeMetaDataInBuffers (output) failed w/ err -1010
I/ExtendedACodec: setupVideoEncoder()
E/libc: Access denied finding property "ro.hwui.texture_cache_size"
W/ACodec: do not know color format 0x7fa30c06 = 2141391878
W/IInputConnectionWrapper: finishComposingText on inactive InputConnection
[... cut ...]
D/EncoderWrapper: frame encoded | threadId=1
D/EncoderWrapper: encoding frame | threadId=1
D/EncoderWrapper: frame sent to encoder | threadId=1
D/EncoderWrapper: time between frames: 109ms
D/FrameUtility: frame timestamp: 1526768236749, index: 7, duration: 2, keyFrame: false, flags: 0
I/AndroidCameraMediaSource: updating sink with frame
D/KinesisVideoAndroidClient: PutFrame index: 7, pts: 15267682367490000, dts: 15267682367490000, duration: 20000, keyFrame: false, flags: 0

And this is a screenshot of my console
[image]