awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #43: 'jni.h' file not found - osx sierra

mkumar20xx, 2018-03-04:
Hi,

I get jni.h file not found while trying to build the native JNI lib for osx. 

sudo ./install-script
...
`-- Build files have been written to: /Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build
[  0%] Building CXX object CMakeFiles/KinesisVideoProducerJNI.dir/Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp.o
In file included from /Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp:5:
/Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/../kinesis-video-producer-jni/src/include/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.h:14:10: fatal error: 
      'jni.h' file not found
#include <jni.h>                  // Basic native API
         ^
1 error generated.
make[3]: *** [CMakeFiles/KinesisVideoProducerJNI.dir/Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp.o] Error 1
make[2]: *** [CMakeFiles/KinesisVideoProducerJNI.dir/all] Error 2
make[1]: *** [CMakeFiles/KinesisVideoProducerJNI.dir/rule] Error 2
make: *** [KinesisVideoProducerJNI] Error 2
Homes-Mac-mini:kinesis-video-native-build home$ `

**Java version**
Homes-Mac-mini:kinesis-video-native-build home$ java -version
java version "1.8.0_162"
Java(TM) SE Runtime Environment (build 1.8.0_162-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.162-b12, mixed mode)
Homes-Mac-mini:kinesis-video-native-build home$ 

It looks like headers file are mapped in ./install-script. 
How do it manually specify the location of jni.h header file ? Or should I look for some other work around ?
I have been into this for weeks, unable to map the headers to install_script. any help is greatly appreciated.

unicornss, 2018-03-04:
Can you try the following from README,

> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md

**Install Open JDK:**

```
$ sudo apt-get install openjdk-8-jdk
$ java -showversion
openjdk version "1.8.0_151"
OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.17.10.2-b12)
OpenJDK 64-Bit Server VM (build 25.151-b12, mixed mode)
```

Set J**AVA_HOME environment variable**:

`$ export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/`

unicornss, 2018-03-04:
Could you please try export JAVA_HOME=\<your jdk home directory\>
To check if the environment variable is set:
` $ echo $JAVA_HOME`
To set the environment variable (e.g in Sierra):
```
$ export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)`
$ echo $JAVA_HOME
/Library/Java/JavaVirtualMachines/jdk1.8.0_161.jdk/Contents/Home
```
If its set then from the directory 

> /Users/home/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build

run
`./install-script `

Could you also provide how are you planning to use the Kinesis Video Streams ?

enusbaum, 2018-03-05:
Received a similar error while compiling latest clone on my Raspberry Pi while running ./install-script:

```
In file included from /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer-jni/src/source/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.cpp:5:0:
/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/../kinesis-video-producer-jni/src/include/com/amazonaws/kinesis/video/producer/jni/KinesisVideoClientWrapper.h:14:54: fatal error: jni.h: No such file or directory
 #include <jni.h>                  // Basic native API
                                                      ^
compilation terminated.
```

enusbaum, 2018-03-05:
Yes, I have Java JDK installed:

```
pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ java -version
java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)
```

```
pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ dpkg --list | grep -i jdk
ii  oracle-java8-jdk                8u65                         armhf        Java™ Platform, Standard Edition 8 Development Kit
```

```
pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ echo $JAVA_HOME
/usr/lib/jvm/jdk-8-oracle-arm-vfp-hflt
```

zhiyua-git, 2018-03-05:
Hi enusbaum,

Can you check if you need to set JAVA_HOME to /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt?

This [post](https://raspberrypi.stackexchange.com/questions/65770/java-home-not-working) looks similar to your case.

Could you also provide how are you planning to use the Kinesis Video Streams?

Thanks,
-ZH

mkumar20xx, 2018-03-05:
@unicornss below is Java_home directory

Homes-Mac-mini:~ home$ echo $JAVA_HOME
/Library/Java/JavaVirtualMachines/jdk1.8.0_162.jdk/Contents/Home

planned usage: I am building a prototype for construction sites workforce remote monitoring and data analysis from video-streams

mkumar20xx, 2018-03-05:
Updated xcode to latest version and tried re running the install script and it worked.So closing it for now.

closed 2018-03-05 by mkumar20xx