awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #44: Doesn't start dummy streaming after unit test passed - OSX High sierra

mkumar20xx, 2018-03-06:
Hi,

Updated access key, secret key and stream name. (stream created earlier) 
Build success
Executed ./start from kinesis-video-stream-native-build directory
Below is last few lines of console output

```
[----------] 1 test from ProducerApiTest (835 ms total)
[----------] Global test environment tear-down
[==========] 309 tests from 30 test cases ran. (8579 ms total)
[  PASSED  ] 309 tests.
  YOU HAVE 1 DISABLED TEST
INFO - Curl shutdown.
```

In metrics I can see only Listfragments.latency not putmedia.incomingbytes

zhiyua-git, 2018-03-07:
Hi mkumar20xx,

Can you provide more log lines for that last test case you run? The log will provide useful information for us to better understand what is happening.

It will take much longer time to finish the test in normal case. This is the test time on my laptop.
```
[----------] 1 test from ProducerApiTest (120415 ms total)

[----------] Global test environment tear-down
[==========] 309 tests from 30 test cases ran. (123643 ms total)
[  PASSED  ] 309 tests.

  YOU HAVE 1 DISABLED TEST
```

Thanks,
-ZH

MushMal, 2018-03-07:
I believe the streaming test is being skipped if the credentials are not specified through the environment variable

unicornss, 2018-03-12:
Hi @mkumar20xx , were you able to resolve the unit test stream as suggested by @zhiyua-git and @MushMal ?  

Also, could you tell us how you are planning to send video streams to Kinesis?

unicornss, 2018-03-19:
Closing this unit-test clarification for now.  Pl. reopen if support is needed.

closed 2018-03-19 by unicornss