awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #45: Cannot stream with kinesis sample app

romangrewal, 2018-03-06:
When running the c++ kinesis sample app with AWS_DEFAULT_REGION="us-east-1a"
it gives me the error
'Could not resolve host: kinesisvideo.us-east-1a.amazonaws.com'

When AWS_DEFAULT_REGION="us-east-1"
it gives me the error
'The request signature we calculated does not match the signature you provided. Check your AWS secret access key and signing method'

What is the correct way to set up the kinesis sample app so that I can stream a sample video?

zhiyua-git, 2018-03-07:
Hi romangrewal,

When you want to change the region to us-east-1, you just need to set AWS_DEFAULT_REGION="us-east-1".

Can you check if you are providing correct AccessKey/SecretKey for running the app, I was getting same error when I input incorrect SecretKey? You can refer [README](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#setting-credentials-in-environment-variables) to set those value.

If your AccessKey/SecretKey are properly set, can you send us the log for further investigation?

Thanks,
-ZH

romangrewal, 2018-03-07:
Log into AWS console
IAM -> Users -> Security Credentials -> Create access key
Copy/paste as env variables AWS_ACCESS_KEY_ID and AWS_SECRET_KEY
run as './kinesis_video_gstreamer_sample_app test_h264_jpg.mkv'

Outputs
[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1790530/log.txt)

romangrewal, 2018-03-07:
Solution:
AWS_SECRET_KEY needs to be AWS_SECRET_ACCESS_KEY

closed 2018-03-07 by romangrewal