awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #47: Raspberry Pi

JulianKling, 2018-03-09:
Hey, tried multiple times to install this on a clean raspbian os downloaded und updated right before the installation and following the official installation guide for raspberry pi. Every time I get errors like

flex not found. Installing
...
curl: (77) error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none

So far I didn't find a solution. Any suggestions? Or can somebody upload a working raspbian image with the video stream producer installed?

unicornss, 2018-03-09:
Please refer the following readme for the certificates for downloading  the open source dependencies from https.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

> Certificate store integration
> 
> Kinesis Video Streams Produicer SDK for C++ needs to establish trust with the backend service through TLS. This is done through validating the CAs in the public certificate store. On Linux-based models, this store is located in /etc/ssl/ directory by default.
> 
> Please download the PEM file from https://www.amazontrust.com/repository/SFSRootCAG2.pem
> 
> to /etc/ssl/cert.pem. Append to the end of the file if it exists.

JulianKling, 2018-03-09:
Still no success:

flex not found. Installing
...
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

Since I wants to eliminate the chance I made a mistake following the provided tutorial (First google link: https://aws.amazon.com/de/kinesis/video-streams/raspberry-pi-tutorial/), did anyone get it to run following exactly these steps? Seems to me like this guide is outdated.

unicornss, 2018-03-09:
As @bdhandap  has posted in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/40 , it could be that the curl was not built with the cacert.
 

> Can you please reinstall curl by following below steps? This should fix the issue.
> 
>   rm <producer_sdk_path>/kinesis-video-native-build/downloads/local/lib/libcurl*
>   rm <producer_sdk_path>/kinesis-video-native-build/downloads/local/bin/curl*
>   cd <producer_sdk_path>/kinesis-video-native-build/downloads/curl-7.57.0
>   export DOWNLOADS=<producer_sdk_path>/kinesis-video-native-build/downloads
>   make clean
>   ./configure --prefix=$DOWNLOADS/local/ --enable-dynamic --disable-rtsp --disable-ldap --without-zlib --with-ssl=$DOWNLOADS/local/ --with-ca-bundle=/etc/ssl/cert.pem
>   make
>   make install
> 
> After reinstalling you can verify it by running `./curl-config --ca`. By default the curl library should pick up the default path /etc/ssl/cert.pem but looks like its not doing that it in all cases. I'll add this configure option in install-script so that it doesn't happen again.

bdhandap, 2018-03-09:
@JulianKling  Curl installation tries to configure the path to `/etc/ssl/cert.pem` and  `curl: (77) error setting certificate verify locations: CAfile: /etc/ssl/cert.pem` error means that the certificate location passed during the installation does not exist. If you did copy the certificate to the mentioned path and tried again, the chances are that configure step did not pick the path as expected. You can verify this by running `./curl-config --ca` inside the `<producer_sdk_path>/kinesis-video-native-build/downloads/local/bin/` folder. Can you please reinstall like mentioned in the [Troubleshooting section](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/40#issuecomment-369782889) ?

JulianKling, 2018-03-09:
Thank you @unicornss and @bdhandap, I think the combination solved it (currently still running). While most likely not recommended, another way would be to add the -k flag to the curl calls inside the installation script (finished successfully).

I think further (potential) customers would appreciate it if you edited to linked guide to include the given help, since these steps seem to be mandatory. (Also, just for the sake of completeness  on how i got it to run, i had to enable the v4l2 drivers on the raspberry pi with sudo rpi-update # to make sure the v4L2 drive is available and sudo modprobe bcm2835-v4l2 # to load it and create /dev/video0)

bdhandap, 2018-03-09:
@JulianKling I'm glad it worked. And thanks for your feedback. We are working on updating the raspberry pi tutorial guide and will be releasing it soon.

HelloEgg, 2018-05-21:
@JulianKling 
How did you add -k flag to curl calls in installation script?
Please help..