awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #48: Unable to set the source to ready state

matszal, 2018-03-11:
Hi Guys,
I followed the AWS Kinesis and Pi tutorial and after comleting all the steps I am running into following error log:

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream Kinesis_user is Ready.
DEBUG - Stream is ready
Unable to set the source to ready state.

freeKinesisVideoClient(): Freeing Kinesis Video ClientDEBUG - streamClosedHandler invoked

commonHeapDebugCheckAllocator(): Heap is initialized
commonHeapDebugCheckAllocator(): Heap limit:                            536870912
commonHeapDebugCheckAllocator(): Heap size:                             0
commonHeapDebugCheckAllocator(): Number of allocations:                                 0
aivHeapDebugCheckAllocator(): Allocated blocks pointer:                                 (nil)
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer:                              0x5614b008
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:    0x5614b008              size:   536870887
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Freeing Kinesis Video Stream Kinesis_user

freeKinesisVideoStream(): Freeing Kinesis Video stream.INFO - Curl shutdown.

Any ideas please?

MushMal, 2018-03-12:
Hi @matszal - this error is indicative that your gstreamer pipeline has an error when trying to move to a "READY" state. Could you please provide more information in the issue by pasting the logs from running with GST_DEBUG=4 environment variable. You can simply set it before running the application:

`export GST_DEBUG=4` 
and then run the application.

This will make the gstreamer be more verbose.

matszal, 2018-03-12:
Hi @MushMal - this is the log after running the test app script with GST_Debug

export GST_DEBUG=4
pi@raspberrypi:~/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ ./kinesis_video_gstreamer_sample_app Kinesis_user
0:00:00.000477445  1823   0x972200 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.000941401  1823   0x972200 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
0:00:00.001036973  1823   0x972200 INFO                GST_INIT gst.c:528:init_pre: Linux raspberrypi 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l
0:00:00.002767276  1823   0x972200 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.006198506  1823   0x972200 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.007531467  1823   0x972200 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.008324900  1823   0x972200 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.008432608  1823   0x972200 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.008623648  1823   0x972200 INFO            GST_REGISTRY gstregistry.c:1721:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin
0:00:00.115061733  1823   0x972200 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin in 0.106230 seconds
0:00:00.115417512  1823   0x972200 INFO            GST_REGISTRY gstregistry.c:1577:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin
0:00:00.120868522  1823   0x972200 INFO            GST_REGISTRY gstregistry.c:1679:scan_and_update_registry: Registry cache has not changed
0:00:00.120927949  1823   0x972200 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.120968939  1823   0x972200 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.121004095  1823   0x972200 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
0:00:00.121036699  1823   0x972200 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 1520889592 Expiration: 0
INFO - New credentials expiration is 1520891992
DEBUG - createDeviceHandler invoked

createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream Kinesis_user

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1520889592 Expiration: 0
DEBUG - Awaiting for the stream to become ready...
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.520806840242E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:831663239333:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:831663239333:stream/Kinesis_user/1520806840242","StreamName":"Kinesis_user","Version":"cPJApA61I8VZS2LcTumX"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:831663239333:stream/Kinesis_user/1520806840242
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:831663239333:stream/Kinesis_user/1520806840242

describeStreamResultEvent(): Describe stream result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1520889593 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1520889594 Expiration: 1520891992
INFO - New credentials expiration is 1520891994

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream Kinesis_user is Ready.
DEBUG - Stream is ready
0:00:02.048032631  1823   0x972200 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstcoreelements.so" loaded
[... cut ...]
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer:                              0x56224008
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:    0x56224008              size:   536870887
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Freeing Kinesis Video Stream Kinesis_user

freeKinesisVideoStream(): Freeing Kinesis Video stream.INFO - Curl shutdown.

MushMal, 2018-03-13:
0:00:02.068808658 1823 0x972200 WARN v4l2 v4l2_calls.c:612:gst_v4l2_open: error: Cannot identify device '/dev/video0'.

Please refer to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#raspberry-pi-failure-to-load-the-camera-device

matszal, 2018-03-13:
@MushMal  - Thanks a lot that solved some of the problems but I am getting more now. After running demo script the terminal is keep printing some DEBUG information with eror below. Wondering if you ever came accross that problem?
getKinesisVideoMetrics(): Get the memory metrics size.
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 0000000001a5cfc2.DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - Kinesis Video client and stream metrics
        >> Overall storage size: 536870912
        >> Available storage size: 534317377
        >> Allocated storage size: 2553535
        >> Total view allocation size: 144064
        >> Total streams frame rate: 29
        >> Total streams transfer rate: 1338112
        >> Current view duration: 0
        >> Overall view duration: 20090000
        >> Current view size: 0
        >> Overall view size: 2552035
        >> Current frame rate: 29.7253
        >> Current transfer rate: 1349724
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postBodyStreamingReadFunc (curl callback) invoked
DEBUG - Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
ERROR - Failed to submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp: 1084933661 presentation timestamp: 1084600075
Dropped frame!

MushMal, 2018-03-13:
@matszal the initial output looks good. You have ~512MB content store and most is available. Your network speed is adequate for your stream so you have 0 latency. You are streaming at ~30 fps with the observed transfer rate of 10Mbps.

Tha last line is an interesting one. 0x32000002 means invalid timestamps have been passed to the SDK - they are not monotonically increasing or your frame references frame from the previous fragment.

Can I get the following info from you:

* What are you trying to accomplish? What's your end-to-end scenario, whats your latency requirements, etc...
* What is your hardware/platform/version
* What is the type of camera
* What application are you running - is this the kinesis video GStreamer sample?
* Have you modified the sample?
* Could you include all the log output so we can try to see what's going on with the GStreamer?
* Could you also set `export GST_DEBUG=4` before running the GStreamer application (assuming this is gstreamer application you are running) - this will produce more GStreamer logs.

matszal, 2018-03-13:
@MushMal 
- I'd like to be able to stream live videos to android app. This is my final year college project and it'll be connected to local network. Best scenario would be to have instant access to stream after pressing the button on android app if possible?
- My platform is raspberry pi v3
- Camera type is genuine pi camera v2 i believe
- Application tested was kinesis_video_gstreamer_sample_app
- example hasn't been modified in any way (followed AWS pi - kinesis tutorial)
I will add log in few minutes

MushMal, 2018-03-13:
Thanks - looking forward to the logs. BTW, there is a producer SDK for Android too that you might be interested in.

matszal, 2018-03-13:
Please see attached log. Sorry, but thers a lot...
pi@raspberrypi:~/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ ./kinesis_video_gstreamer_sample_app Kinesis_user
0:00:00.000471453  1693   0xb7c200 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.000932334  1693   0xb7c200 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
0:00:00.001029000  1693   0xb7c200 INFO                GST_INIT gst.c:528:init_pre: Linux raspberrypi 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l
0:00:00.002732057  1693   0xb7c200 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.006206659  1693   0xb7c200 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.007530344  1693   0xb7c200 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.008301691  1693   0xb7c200 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.008452262  1693   0xb7c200 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.008647885  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1721:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin
0:00:00.126853779  1693   0xb7c200 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin in 0.117994 seconds
0:00:00.127110808  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1577:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv7l.bin
0:00:00.131988417  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1326:gst_registry_scan_path_level:<registry0> cached info for /home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideo4linux2.so is stale
0:00:00.000241821  1694   0x46d400 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.000431298  1694   0x46d400 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
0:00:00.000480568  1694   0x46d400 INFO                GST_INIT gst.c:528:init_pre: Linux raspberrypi 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l
0:00:00.001359362  1694   0x46d400 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.003083981  1694   0x46d400 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.003766735  1694   0x46d400 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.004174179  1694   0x46d400 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.004228397  1694   0x46d400 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.004315063  1694   0x46d400 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.004354854  1694   0x46d400 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.004389489  1694   0x46d400 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
0:00:00.004419958  1694   0x46d400 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
0:00:00.014552100  1694   0x46d400 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/Desktop/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.154772729  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1688:scan_and_update_registry: Registry cache changed. Writing new registry cache
0:00:00.154842833  1693   0xb7c200 INFO            GST_REGISTRY gstregistrybinary.c:369:priv_gst_registry_binary_write_cache: Building binary registry cache image
0:00:00.248173237  1693   0xb7c200 INFO            GST_REGISTRY gstregistrybinary.c:401:priv_gst_registry_binary_write_cache: Writing binary registry cache
0:00:00.677446191  1693   0xb7c200 INFO            GST_REGISTRY gstregistrybinary.c:262:gst_registry_binary_cache_finish: Wrote binary registry cache
0:00:00.677618689  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1697:scan_and_update_registry: Registry cache written successfully
0:00:00.677660355  1693   0xb7c200 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.677700199  1693   0xb7c200 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.677737959  1693   0xb7c200 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
[... cut ...]
        >> Current view duration: 100000
        >> Overall view duration: 100000
        >> Current view size: 962
        >> Overall view size: 962
        >> Current frame rate: 30
        >> Current transfer rate: 4194304
0:00:03.261620543  1693   0xbd7c60 INFO                 v4l2src gstv4l2src.c:825:gst_v4l2src_create:<source> sync to 0:00:00.099999999 out ts 0:00:00.785020535
ERROR - Failed to submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp: 8185625 presentation timestamp: 7850205
Dropped frame!

MushMal, 2018-03-13:
Ok, I see what's going on.. Some encoders act weird and they produce an Idr frame (in our case we treat that as a Key-frame) with different Pts and Dts (presentation, decoding timestamps). This is odd but it seems to be the case. As a result, the Key-frame fails to be pushed to the SDK with 0x32000011

`utStreamResultEvent(): Put stream result event.ERROR - Failed to submit frame to Kinesis Video client. status: 0x32000011 decoding timestamp: 7360151 presentation timestamp: 7280467`

which then leads to a non-key frame to fail.

While we investigate this issue, could I ask you to verify whether you have the latest sample code? We did make a workaround for this particular case in kinesis_video_gstreamer_sample_app.cpp LN 185:

`      if (data->h264_stream_supported) {
            buffer->pts = buffer->dts;
        }`

MushMal, 2018-04-17:
@matszal - is there any progress on this? Can you sync-up the latest codebase? Please update the thread so we can see how best to resolve this issue.