awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #49: Cannot manage logs properly

liuyangedward, 2018-03-12:
Hi team,

First of all, I'd like to thank you for offering this awesome tool. 

Currently, I'm working on integrating kinesis logs into our logging scheme. I've managed to deal with those generated from log4cplus but I found it tricky when dealing with those generated from DLOGX (those macros in PlatformUtils.h). To start with, I'd like to turn the log off (the next step would be to redirect them to a file), is there a way to achieve that without modifying source file?

Below are some of the logs that I hope to get rid of, 
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 0000000003ee1586.
getKinesisVideoMetrics(): Get the memory metrics size.

Thank you!

Regards
Edward

MushMal, 2018-03-12:
@liuyangedward Thanks for your comments. The PIC (Platform Independent Codebase) layer has been designed with full portability in mind. The logging in your case gets defaulted to the system output - the logging is built using the macros which eventually result in a log API call with the right information level. This is the API that we are not exposing through the plumbing yet.

We are working on exposing the logging API through the callbacks so the C++ layer can implement it using log4cpp.
In the interim, you can modify the PlatformUtils.h to map LN:28
#define __LOG(p1, p2, p3, ...) printf(p3, ##__VA_ARGS__) to a logger you need - in your case log4cpp call.
However, if you simply want to disable the metrics logging (which for the diagnostics purposes we print out on each key frame) you can modify the code in KinesisVideoStream.cpp LN 37 and simply remove this if block or put an unconditional false statement - something like this

if (FALSE && (frame.flags & FRAME_FLAG_KEY_FRAME) == FRAME_FLAG_KEY_FRAME) {

zhiyua-git, 2018-04-25:
@liuyangedward We just released the version 1.4.0 with logging API exposed to C++ layer, can you check it out and have a try? Thanks.

unicornss, 2018-05-09:
Closing the issue for now.  If additional support is required please reopen. Thanks

closed 2018-05-09 by unicornss