awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #53: RTSP Quality VS YouTube RTMP

francoisdutoit, 2018-03-19:
Hi,

I am wondering why the quality of my RTSP stream is so poor compared to an RTMP stream sent to youtube live ( see attached video )

I have a screen recording of a simple app - The top video is getting its stream from Kinesis ( pushed up using the gstreamer_sample_rtsp_app ) and the bottom video is locally encoded to RTMP using ffmpeg and then pushed to youtube live and embedded in my page.

The quality is the same in the AWS console.....

Anything I can do to improve this?

Thanks!

[Home Page - AngularCoreAWS - Google Chrome 2018_03_19 19_07_10.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1826082/Home.Page.-.AngularCoreAWS.-.Google.Chrome.2018_03_19.19_07_10.zip)

francoisdutoit, 2018-03-19:
Also - below are the commands I am using and attached my camera settings

//OnUbutntu - RTSP
AWS_ACCESS_KEY_ID=MyKeyID  AWS_SECRET_ACCESS_KEY=MySecret./kinesis_video_gstreamer_sample_rtsp_app  "rtsp://192.168.0.108:554/cam/realmonitor?channel=1&subtype=1&unicast=true&proto=Onvif" MyOnvifStream1

//FromWindows
ffmpeg -f lavfi -i anullsrc -rtsp_transport udp -i "rtsp://192.168.0.108:554/cam/realmonitor?channel=1&subtype=1&unicast=true&proto=Onvif" -tune zerolatency -vcodec libx264 -t 12:00:00 -pix_fmt + -c:v copy -c:a aac -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/youtubekey

[image]

MushMal, 2018-03-19:
@francoisdutoit the KVS end-to-end does not modify the encoded bits so the quality should be what your application/pipeline is producing. 

I looked at the attached videos. The visual artifacts are not related to the quality of the encoder but the fact that the elementary stream is broken - either the timestamps are incorrect or frames are missing.

Could you please ensure that

1) Your application properly submits decoding timestamps and presentation timestamps - those should be properly done. I assume you are using the sample application?
2) There are no dropped frames on the producer side. Please run for the duration of the test and collect all the detailed logs so we can see whether there are any dropped frames, failed frames.
3) Less likely, your browsers support for MSE might have issues with properly decoding (I wouldn't even consider this at this stage)

unicornss, 2018-03-19:
Hi @francoisdutoit 

The demo RTSP app has been tested with streams having video resolutions of 1900x1200 without any visual quality issues.  
- Could you tell us what environment you are running this demo app? 
-- what is the Ubuntu configuration ( CPU / Memory / Network / Wired / Wireless Speed )?  
- Is it possible to run the _ffmpeg_ also in the same environment and send us the screenshots?

Thanks
ss

francoisdutoit, 2018-03-20:
@unicornss @MushMal 

 So running ffmpeg from the ubuntu vm still gives quality results on youtube - the only difference between the two streams is that I am directly pushing a RTSP stream to kinesis and with youtube I am converting the stream to rtmp using ffmpeg and then pushing to youtube.

See attached videos.
[Captures.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1830220/Captures.zip)

francoisdutoit, 2018-03-20:
PS:  My Windows 10 PC is an i7-7700 4.2Hgz with 16GB Ram
The Ubuntu VM that runs on the windows pc has 8GB of ram assigned.
My line speed is 8-10Mbps

francoisdutoit, 2018-03-20:
Here is some of the console output from running the kinesis_video_gstreamer_sample_rtsp_app 

[KinesisConsoleOutput.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1830338/KinesisConsoleOutput.txt)

MushMal, 2018-03-20:
Check if you have errors from KVS Producer SDK. As I mentioned, please attach the full logs for the run so we can help.

francoisdutoit, 2018-03-20:
@MushMal  - Where can I check for errors on the Producer SDK and where are the log files saved? - I am just running the sample app from the command line

unicornss, 2018-03-20:
As @MushMal  has mentioned the sample app sends the incoming data stream directly to KVS as it is received. The quality of the streams you are seeing in the console relates directly to the stream it receives from the source.  
- https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp#L396

You may want to check your camera options (higher bitrate , subtype )  to send the best quality RTSP stream .

MushMal, 2018-03-20:
@francoisdutoit by default the sample applications output the logs on the console. You can capture the entire output from the console.

unicornss, 2018-03-20:
Hi @francoisdutoit 

From your logs, it shows frames are getting dropped due to timestamp issue in the incoming stream.

> ERROR - Failed to submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp: 713726356 presentation timestamp: 713416203

For any errors you see in the logs, You can refer the details in https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html 

Also, make sure you are using the latest sample application from the Github.

Thanks

unicornss, 2018-03-20:
@francoisdutoit  can you also provide details of the camera model you are using ... Thanks

francoisdutoit, 2018-03-20:
@unicornss ,

It a Dahua  IPC-HFW1120S,

Device Type
IPC-HFW1120S
Software Version
2.420.0000.19.R, Build Date: 2016-06-24
WEB Version
3.2.1.355658
ONVIF Version
2.4.2

unicornss, 2018-03-20:
@francoisdutoit  can you try Encode Mode as. H.264B (baseline profile) instead of H.264H ?

francoisdutoit, 2018-03-20:
@unicornss . Ok - Let me try that....

francoisdutoit, 2018-03-20:
@unicornss 
That does not seem to make a difference...... I'm trying a few different settings with the main stream now that are not available in the substream. - Just to see what effect that has...

francoisdutoit, 2018-03-20:
@unicornss ,
I'm going to install the latest version of the app to see if that makes a difference...... Will let you know if that makes any difference.  Thanks!

francoisdutoit, 2018-03-20:
@unicornss  - so I'm running the latest version of the demo app now - but it is just timing out

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575991 Expiration: 0
INFO - New credentials expiration is 1521578391
DEBUG - createDeviceHandler invoked

createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream MyOnvifStream1

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575991 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2WGXLCSWYBFSUFQ/20180320/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=576ae9d8c8bcea527b7058c1b97c2c9e9ae7d0b723d02997caa4471f1bcd3acb
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180320T195951Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575992 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2WGXLCSWYBFSUFQ/20180320/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=90a0279d9378eed4400f0a082dadc5558b5c634ca7ef75e2b764bb3da927594a
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180320T195952Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575993 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2WGXLCSWYBFSUFQ/20180320/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=b14c36600bf87d070e5152f5c45c3ff783996b8f1b2c85056dc3c3f5af64c681
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180320T195953Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575994 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2WGXLCSWYBFSUFQ/20180320/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=e831703be9a57cc6fd7de743f8a556ed7200bd296aaba04a1b1971821d8d9819
    content-type: application/json
[... 16 lines cut ...]
    X-Amz-Date: 20180320T195956Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521575999 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
[... cut ...]
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.ERROR - Submitting event result for stream: 94702723674678 failed with: 1375731729
ERROR - Reporting stream error. Errored timecode: 0 Status: 1375731729
ERROR - Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Aborted (core dumped)

francoisdutoit, 2018-03-20:
I'm getting the following error with the latest version....

describeStreamResultEvent(): Describe stream result event.DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1521577521 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2WGXLCSWYBFSUFQ/20180320/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=101dc453907fe8293648b6ce28be254a11d63f2b30a0553dbe97aa46ca1056fb
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180320T202521Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream MyOnvifStream1 in Kinesis Video (stream will be created)

describeStreamResultEvent(): Describe stream result event.ERROR - Submitting event result for stream: 94696165351990 failed with: 1375731729
ERROR - Reporting stream error. Errored timecode: 0 Status: 1375731729
ERROR - Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Aborted (core dumped)

zhiyua-git, 2018-03-20:
This is caused by certificate issue. Please refer to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#certificate-store-integration to fix this.

unicornss, 2018-03-20:
As @zhiyua-git   mentioned this is due to the certificate issue and you need to copy the certificate as in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#certificate-store-integration . Then follow the steps in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/40 to rebuild the curl.

> Basically curl is not able to refer the cert ca bundle and thus fails validation. curl which was downloaded as part of the install-script uses libssl and other dependencies from the local folder (<producer_sdk_path/kinesis-video-native-build/downloads/local/lib). Looks like your curl was not build properly. Can you please reinstall curl by following below steps? This should fix the issue.
> 
>   rm <producer_sdk_path>/kinesis-video-native-build/downloads/local/lib/libcurl*
>   rm <producer_sdk_path>/kinesis-video-native-build/downloads/local/bin/curl*
>   cd <producer_sdk_path>/kinesis-video-native-build/downloads/curl-7.57.0
>   export DOWNLOADS=<producer_sdk_path>/kinesis-video-native-build/downloads
>   make clean
>   ./configure --prefix=$DOWNLOADS/local/ --enable-dynamic --disable-rtsp --disable-ldap --without-zlib --with-ssl=$DOWNLOADS/local/ --with-ca-bundle=/etc/ssl/cert.pem
>   make
>   make install

francoisdutoit, 2018-03-21:
@unicornss @MushMal 

So running the latest version has resolved the issue :) The quality is fine now :)

Thanks for the help :)

closed 2018-03-21 by francoisdutoit