awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #55: Kinesis Video Metadata

pastawoua, 2018-03-22:
Is there a more efficient way of extracting data from the stream. Attempting to pull JSON out of MKV fragments has yielded the following results:

Profiling for 10 fragments/10 frames (times in ms)
- getting fragment list took 2469 (aus to us-east-1)
- getting media (fragments) took 1927 (aus to us-east-1)
- extracting metadata took: 746 

simply running this loop 
```
while (mkvStreamReader.mightHaveNext) {
          val mkvElement: Optional[MkvElement] = mkvStreamReader.nextIfAvailable
}
```

took the following: 
594 iterations
507 ms

For extracting the data, we've used the visitor pattern, as per the SDK. The issue appears to be that even for a small number of fragments, there is a large number of elements visited (some 50x number of fragments or more).
- Is visiting each MKV element to determine if it contains the required data necessary? 
- Is there a more efficient way of extracting this data? 
- Is this just a limitation of storing data in MKV containers?
- Is the visitor pattern appropriate for this use case?

vijaygj, 2018-03-22:
Can you please report this issue on parser library page - https://github.com/aws/amazon-kinesis-video-streams-parser-library ? We will look into it with more details and it will also help other users.

pastawoua, 2018-03-22:
@vijaygj just did at https://github.com/aws/amazon-kinesis-video-streams-parser-library/issues/17

closed 2018-03-22 by pastawoua

helpingothers4u, 2018-03-22:
Thanks

Angela Risner