awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #56: Issue with listFragments when selector is set too early

pastawoua, 2018-03-22:
When using the "listFragments" method in the Java SDK, if the start timestamp of the SERVER_TIMESTAMP fragment selector is set too early, the response will return a null fragment list. It appears to be that "too early" is "any time before the retention period was changed or the stream was created".

- SDK/API should handle arbitrarily early start timestamps, regardless of stream creation/modification
- The SDK should not return null if no fragments are found, it should be returning an empty list

vijaygj, 2018-03-22:
Hello,

Thank you for reporting the issue. Can you please send us the requestId of the response at kinesis-video-support@amazon.com?

You can use following method to get the same in your code - 

https://docs.aws.amazon.com/AWSJavaSDK/latest/javadoc/com/amazonaws/ResponseMetadata.html#getRequestId--

Thanks,
Vijay.

pastawoua, 2018-03-22:
@vijaygj 

```
a920e9b3-2d75-11e8-96e5-f96abbf797f4 - from epoch til now
bb370a5f-2d75-11e8-a20e-93caf285e492 - from 10 days ago til now
```

vijaygj, 2018-03-22:
Thank you. Can you please also provide us the AWS region and approximate time when this happened?

qqia, 2018-03-26:
@pastawoua

Thanks for the valuable suggestion. We will consider and update you.
Getting Non-empty fragment list from ListFragment API replies on successfully uploaded fragments from PutMedia. But, from our end, we did not find fragments available for the streamName associated with your requestId. Please let us know if you still have ListFragment API issues given PutMedia is running successfully.

seanajar, 2018-03-26:
@qqia 
Thanks for the response. At time of posting, there were valid fragments in that stream. I was able to retrieve those fragments successfully with a start timestamp set to just after the stream had been created, then setting time time stamp just 1 minute earlier yielded a null fragment list.

There was a definitive start time stamp at which before yielded a null result, and i was able (at the time) to isolate this down to the second by binary searching between bounds.

As those results have likely expired (stream retention was only 1 hour), i threw together a quick example, and yields the same result for both server and producer timestamp.:
`{StreamName: 25d30047-6cc0-4cd0-8dac-fc8ca3e4ca0c_live,MaxResults: 100,FragmentSelector: {FragmentSelectorType: SERVER_TIMESTAMP,TimestampRange: {StartTimestamp: Tue Mar 27 07:32:53 AEDT 2018,EndTimestamp: Tue Mar 27 10:32:53 AEDT 2018}}}
requestId - 037db4fb-314e-11e8-b48c-d5341dd3661a
[{FragmentNumber: 91343852333184738618566103961295043955443919975,FragmentSizeInBytes: 95750,ProducerTimestamp: Tue Mar 27 10:25:12 AEDT 2018,ServerTimestamp: Tue Mar 27 10:25:06 AEDT 2018,FragmentLengthInMilliseconds: 700}, ... {FragmentNumber: 91343852333184738663131945375568733918304179512,FragmentSizeInBytes: 83715,ProducerTimestamp: Tue Mar 27 10:25:18 AEDT 2018,ServerTimestamp: Tue Mar 27 10:25:06 AEDT 2018,FragmentLengthInMilliseconds: 634}]

{StreamName: 25d30047-6cc0-4cd0-8dac-fc8ca3e4ca0c_live,MaxResults: 100,FragmentSelector: {FragmentSelectorType: SERVER_TIMESTAMP,TimestampRange: {StartTimestamp: Thu Jan 01 10:00:00 AEST 1970,EndTimestamp: Tue Mar 27 10:32:26 AEDT 2018}}}
requestId - f3b86261-314d-11e8-b48c-d5341dd3661a
null`

qqia, 2018-03-28:
@seanajar 

Hi 

Can you help us confirm if _NextToken_ field of _ListFragmentsResult_ has valid value when null _FragmentList_ is received with very early startTimeStamp set? 
If _NextToken_ is available, you can iteratively send ListFragments API with the previously received token and eventually the unexpired FragmentList will be returned. 

In the mean time, we have identified the root cause and will optimize our logic so that the first response returns valid FragmentList. 

Thanks for reporting the issue again.

seanajar, 2018-03-28:
@qqia 
Hi, the response does have a next token yes, I iterated through a few of them, all returning no fragments, but I stopped searching after a number of iterations due to timing limitations on my end, so perhaps these would have eventually yielded a result.

Thanks for your responses.

qqia, 2018-04-23:
@seanajar 

Thanks for following up. The fix hasn't been rolled out yet, but we do expect it is coming out soon. We will keep you posted.
Sorry about any inconvenience this issue is causing.

qqia, 2018-05-29:
@seanajar Hi, the fix has been deployed. Please let us know if you still see issues.

unicornss, 2018-07-13:
Closing this issue for now. Please reopen is further support is required. Thanks

closed 2018-07-13 by unicornss