awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #57: Producer SDK artefacts

pastawoua, 2018-03-22:
In order to improve our build times, we are looking at ways to cache the libproducer.so and libraries as an artefact so it doesn't have to build from scratch every time in our build environment.

The install-script is quite opinionated at the moment as it builds tests, sample apps, JNI app, Gstreamer, etc.

- What's the recommended way to build the bare minimum dependencies required by the libproducer?
- Any support for cross-compilation in the near future? We are currently manually building the libproducer.so on a arm64 machine and uploading artefacts on AWS.
- Do you have any plans to host the pre-built packages for x86 and other platforms (ie: arm64)?

MushMal, 2018-03-22:
@pastawoua thanks for the post. The install script was intended to download and build the "world" - it attempts to build the necessary open source components from the scratch to ensure the right versions of the components are used with the right configurations. If you see the SDK both the PIC and the C++/JNI - they are all cmake based. In fact, once you get the necessary open source components built/installed (either by building in the local downloads directory or installing the components on the build machine and pointing to the local directory) the building of the SDK is a simple cmake execution. You can build individual components of the SDK or all at once.

* Run install script once (with a successful completion). This will ensure you have the local folder
* Running install script again will just simply skip the steps as it will find the right components
* If the previous step is too arduous, you can set the env variables just like in the script and simply invoke the cmake to build the SDK.
* If you build the SDK into a dynamic library (like .so) and link it dynamically to your application then when modifying the code for the application you don't need to rebuild the SDK library.
* For now, we don't have an out-of-the-box support for cross-compilation. However, this is not very difficult to do if your toolchain supports a target platform other than the build machine ABI.
* We are working on trying to provide Docker images packaged for various target platforms.
* We do have plans to pre-built the artifacts for well-known platforms. This is not as simple unfortunately as aside from the platform ABI we need to deal with the software ABI too - for example, due to the open source components, we rely on the OS version/runtimes - this might not be a huge issue for some platforms but for example on ARM we can have different software platforms like Android vs. Raspberry PI, etc... So, in our case we need to be careful to get the entire ABI together - something like Android arm v7a.

Hope this sheds some light.

pastawoua, 2018-03-22:
@MushMal thanks for the reply. To clarify, installing the SDK and building locally is quite easy, just time consuming and not efficient especially when running in the cloud. What we are trying to achieve is exactly what you mentioned as being "not as simple", and wanted to check if you had a solution rather than us going down that path.

FYI, I've setup a Dockerfile that automatically builds libproducer for ubuntu:16.04, https://github.com/Nirovision/kinesis-video-producer/blob/master/x86_64/Dockerfile

Also pushed a build to Docker hub: https://hub.docker.com/r/nirovision/kinesis-video-producer/

Issue is doing this for aarch64 and armv7, so having cross-compilation would be greatly appreciated.

As you can see, I had to patch the file to reduce the log output as most cloud CI would stop if the log file exceeds 4mb, would be great to have an install-script that is less verbose and more modular in choosing what to build.

pastawoua, 2018-03-23:
@turb0kat the 4mb is more of a limitation on Travis where I was trying to automatically build the libproducer SDK, looks like Docker Hub doesn't have that limitation so it's ok now, the build is just under their 2 hours limit though.

MushMal, 2018-03-23:
@pastawoua , just curious, would it help in your case to have a per-platform Docker container which has pre-installed dependencies and binaries in case you need to re-build?

pastawoua, 2018-03-23:
@MushMal yes, that's pretty much what I've started to do:
https://hub.docker.com/r/nirovision/kinesis-video-producer/tags/

Issue is for other architectures (ie: aarch64, armv7) as it needs to be either cross-compiled, emulated or ran on the correct architecture.

MushMal, 2018-03-23:
Indeed, the emulation will take a lot longer. The alternative is to have a whole or selective cross-compilation on one box for different hardware and software ABIs. Let us take a stab at getting the right compiler toolchains, etc.. to get this cross-compiled. will update asap

unicornss, 2018-08-08:
Closing this issue for now.  Please reopen if required

closed 2018-08-08 by unicornss