awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #58: No error in logs. But still not streaming

sponaka, 2018-03-28:
Hi Team,
I dont see any error in my logs . But still streaming is not happening 

Pi model : Raspberry Pi Zero
RAM : 512 MB
Changes made in SDK: 
          1.memory set to 256 mb
          2. resolution from 1280, 720 to 640 ,480

logs 👍 

0:00:00.001184017  3428   0xef4000 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.003723054  3428   0xef4000 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib
0:00:00.006525094  3428   0xef4000 INFO                GST_INIT gst.c:528:init_pre: Linux raspberrypi 4.14.30+ #1102 Mon Mar 26 16:20:05 BST 2018 armv6l
0:00:00.014301207  3428   0xef4000 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.024896359  3428   0xef4000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.029339423  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.033008476  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.034505498  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.038086550  3428   0xef4000 INFO            GST_REGISTRY gstregistry.c:1721:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin
0:00:00.212382063  3428   0xef4000 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin in 0.171924 seconds
0:00:00.213636081  3428   0xef4000 INFO            GST_REGISTRY gstregistry.c:1577:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin
0:00:00.232105347  3428   0xef4000 INFO            GST_REGISTRY gstregistry.c:1679:scan_and_update_registry: Registry cache has not changed
0:00:00.233576368  3428   0xef4000 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.235419395  3428   0xef4000 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.237628426  3428   0xef4000 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
0:00:00.239768457  3428   0xef4000 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 268435456, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522257520 Expiration: 0
INFO - New credentials expiration is 1522259920
DEBUG - createDeviceHandler invoked

createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream pi-stream

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522257520 Expiration: 0
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.522168699674E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:806520990827:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674","StreamName":"pi-stream","Version":"DaTeCAShkkC2DIRHJvum"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674

describeStreamResultEvent(): Describe stream result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522257526 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1522257529 Expiration: 1522259920
INFO - New credentials expiration is 1522259929

getStreamingTokenResultEvent(): Get streaming token result event.DEBUG - Kinesis Video stream pi-stream is Ready.
DEBUG - Stream is ready
0:00:09.714561575  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:09.717199612  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "source_filter"
0:00:09.725139724  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0xfa40e0> adding pad 'sink'
0:00:09.727664760  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0xfa40e0> adding pad 'src'
0:00:09.731124808  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "encoder_filter"
0:00:09.735577871  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0xfa4310> adding pad 'sink'
0:00:09.739541927  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0xfa4310> adding pad 'src'
0:00:09.747113034  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/gstreamer-1.0/libgstapp.so" loaded
0:00:09.748844058  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "appsink" named "appsink"
0:00:09.754832143  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSink@0xfa8998> adding pad 'sink'
0:00:09.757267177  3428   0xef4000 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "h264parse"!
[... 1 lines cut ...]
0:00:09.789867638  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/gstreamer-1.0/libgstomx.so" loaded
0:00:09.791426660  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "omxh264enc" named "encoder"
0:00:09.796002724  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0xfb4248> adding pad 'sink'
0:00:09.797669748  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0xfb4248> adding pad 'src'
0:00:09.812663959  3428   0xef4000 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:09.814301983  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "v4l2src" named "source"
0:00:09.834318265  3428   0xef4000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0xfbc0a0> adding pad 'src'
0:00:09.836164291  3428   0xef4000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "pipeline" named "test-pipeline"
Not all elements could be created.

freeKinesisVideoClient(): Freeing Kinesis Video Client
commonHeapDebugCheckAllocator(): Heap is initialized
commonHeapDebugCheckAllocator(): Heap limit: 				268435456
commonHeapDebugCheckAllocator(): Heap size: 				0
[... cut ...]
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0xa619d008
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:	0xa619d008		size:	268435431
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Freeing Kinesis Video Stream pi-stream

freeKinesisVideoStream(): Freeing Kinesis Video stream.INFO - Curl shutdown.

MushMal, 2018-03-28:
@sponaka thanks for opening this issue.

It seems that your gstreamer pipeline failed to create one of the elements - from the logs: Not all elements could be created.

Could I ask you to do the following:

1) Run the following troubleshooting steps: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#raspberry-pi-failure-to-load-the-camera-device
2) After the above didn't work, please run the same scenario with the GStreamer debug logs enabled. You can run this by calling `export GST_DEBUG=4` before executing the application command line.

sponaka, 2018-03-29:
@MushMal Thanks for swift reply

i checked my camera and it shows as below : 

pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp $ ls /dev/video*
/dev/video0

Logs with GST_DEBUG: 
0:00:00.001253314  1455   0x3ae000 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.003909862  1455   0x3ae000 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib
0:00:00.004784384  1455   0x3ae000 INFO                GST_INIT gst.c:528:init_pre: Linux raspberrypi 4.14.30+ #1102 Mon Mar 26 16:20:05 BST 2018 armv6l
0:00:00.009920576  1455   0x3ae000 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.020675696  1455   0x3ae000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.025557028  1455   0x3ae000 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.029996601  1455   0x3ae000 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.031918550  1455   0x3ae000 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.034027397  1455   0x3ae000 INFO            GST_REGISTRY gstregistry.c:1721:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin
0:00:00.212289942  1455   0x3ae000 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin in 0.174653 seconds
0:00:00.213603224  1455   0x3ae000 INFO            GST_REGISTRY gstregistry.c:1577:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.armv6l.bin
0:00:00.231116649  1455   0x3ae000 INFO            GST_REGISTRY gstregistry.c:1679:scan_and_update_registry: Registry cache has not changed
0:00:00.232662804  1455   0x3ae000 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.234132001  1455   0x3ae000 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.236947461  1455   0x3ae000 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
0:00:00.240075751  1455   0x3ae000 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 268435456, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
heapInitialize(): Heap is initialized OKINFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522323828 Expiration: 0
INFO - New credentials expiration is 1522326228
DEBUG - createDeviceHandler invoked

createDeviceResultEvent(): Create device result event.DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream pi-stream

createKinesisVideoStream(): Creating Kinesis Video Stream.DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522323828 Expiration: 0
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.522168699674E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:806520990827:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674","StreamName":"pi-stream","Version":"DaTeCAShkkC2DIRHJvum"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:806520990827:stream/pi-stream/1522168699674

describeStreamResultEvent(): Describe stream result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522323831 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint with result Timeout was reached: Resolving timed out after 5517 milliseconds
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJTEUZ342UJZS3GNA/20180329/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;x-amz-date, Signature=c65a0103ea685fb2cf7dd9f3789a31845e769b3e927aa963124fc31798c2b32d
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180329T114351Z
DEBUG - getStreamingEndpoint response: 

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1522323837 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com

getStreamingEndpointResultEvent(): Get streaming endpoint result event.DEBUG - streamingTokenHandler invoked
[... cut ...]
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0xa6151008
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:	0xa6151008		size:	268435431
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Freeing Kinesis Video Stream pi-stream

freeKinesisVideoStream(): Freeing Kinesis Video stream.INFO - Curl shutdown.

sponaka, 2018-03-29:
Attaching screenshot file
[image]

MushMal, 2018-04-01:
@sponaka thanks for the logs. The first log had a timeout trying to get the data endpoint. However, on retry it recovered. I am not yet seeing anything obvious from the logs. The sample application attempts to load the hw encoder (vtenc_h264_hw) first and if it fails, it falls back to OMX hardware encoder with a consecutive fallback to a software encoder. This is done due to shared codebase of the sample application that works on Mac OS too. From the logs I see that it fails to load vtenc which is expected but then it loads the omx encoder OK.

It's not immediately clear as to what's going on - I wonder if I could ask you to run the same but with the verbose logs - just run `export GST_DEBUG=5` - this will provide more log spew.

Also, have you run the following steps?

* Run firmware update `sudo rpi-update` and restart
* `sudo modprobe bcm2835-v4l2` to check and load the camera driver

I apologize for the back-and-forth

chehefen, 2018-04-10:
Hi, I see from the log that it cannot find the factory for the "h264parse" element. The element belongs to the gst-plugins-bad library. Can you double check that you are loading the library correctly?

sponaka, 2018-04-11:
@chehefen @MushMal 
Thanks for the support !
Running the install script (./install-script)from **kinesis-video-native-build** folder has solved the issue
Pretty strange !

Regards,
Siva

closed 2018-04-11 by unicornss