awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #61: liblog4cplus-1.2.so.5 problem, no solution works for me

davidgatti, 2018-04-03:
Hello, 

Im trying to run the demo app, that you have here, and to do so I did the following:

```
sudo apt-get install cmake
sudo apt-get install libtool
sudo apt-get install libtool-bin
sudo apt-get install automake
sudo apt-get install g++
sudo apt-get install curl
sudo apt-get install pkg-config
sudo apt-get install flex
sudo apt-get install openjdk-8-jdk
export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/
Chmod +x ./install-script
./install-script
```

In addition to that I had to do `sudo apt install liblog4cplus-dev` to remove the missing lib error, then run `rm -rf ./kinesis-video-native-build/CMakeCache.txt ./kinesis-video-native-build/CMakeFiles` and build once more. 

But still, now when I run `kinesis_video_gstreamer_sample_rtsp_app` I get `liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory`. If I google search I get back here :). I did try to do all the provided solution and still nothing. Meaning I did try './start'

I'm working on a fresh installation of Debian 9 on a EC2 instance, here are the details `Linux 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u2 (2018-02-21) x86_64 GNU/Linux`.

Any idea what I could try to make this demo app work?

Tannz0rz, 2018-04-03:
Try `export LD_LIBRARY_PATH=path/to/liblog4cplus` before running the demo.

davidgatti, 2018-04-03:
Despite me doing:

`export LD_LIBRARY_PATH=/home/admin/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus-1.2.so.5.1.4`

I still get `/start: error while loading shared libraries: liblog4cplus-1.2.so.5: cannot open shared object file: No such file or directory`

davidgatti, 2018-04-03:
Hmm I did this:

`export LD_LIBRARY_PATH=/home/admin/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/log4cplus-1.2.0/.libs/`

And it worked, but now I get a new error:

`./start: error while loading shared libraries: libgtest.so: cannot open shared object file: No such file or directory`

I did then install `sudo apt install libgtest-dev` but that didn't fix it. Any idea what I could do next?

Tannz0rz, 2018-04-03:
Just try `export LD_LIBRARY_PATH=/home/admin/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib`

davidgatti, 2018-04-04:
Hmm, I think I'll end up revamping the README ;) we need to close off the last mile of mysteries. And I'm happy to report that @Tannz0rz solution worked, thank you very much for your time.

Now I have a new problem :D but I'll open a new Issue for that.

closed 2018-04-04 by davidgatti