awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #62: Install Issue on Raspberry Pi 3 B+

herosjourney, 2018-04-04:
Raspberry Pi Kernel 4.14.32-v7+
Rasbian GNU/Linux 9 

I receive the following error on running the following script (./kinesis-video-native-build/install-script) in the following directory (amazon-kinesis-video-streams-producer-sdk-cpp). I've provided some extra parts of the install/setup before the error language:

ranlib libssl.a || echo Never mind.
make -f ./Makefile.shared -e \
	PLATFORM=linux-armv4 \
	PERL="/usr/bin/perl" SRCDIR='.' DSTDIR="." \
	INSTALLTOP='/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/' LIBDIR='lib' \
	LIBDEPS=' '""' -ldl ' \
	LIBNAME=crypto LIBVERSION=1.1 \
	LIBCOMPATVERSIONS=';' \
	CC='gcc' CFLAGS='-DDSO_DLFCN -DHAVE_DLFCN_H -DNDEBUG -DOPENSSL_THREADS -DOPENSSL_NO_STATIC_ENGINE -DOPENSSL_PIC -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DBSAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/\"" -DENGINESDIR="\"/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/engines-1.1\"" -Wall -O3 -pthread  -march=armv7-a -Wa,--noexecstack -fPIC -DOPENSSL_USE_NODELETE' \
	LDFLAGS='' \
	SHARED_LDFLAGS='-Wl,-znodelete ' SHLIB_EXT=.so.1.1 \
	RC='windres' SHARED_RCFLAGS='' \
	link_shlib.linux-shared
make[2]: Entering directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/openssl-1.1.0f'
LD_LIBRARY_PATH=: gcc -DDSO_DLFCN -DHAVE_DLFCN_H -DNDEBUG -DOPENSSL_THREADS -DOPENSSL_NO_STATIC_ENGINE -DOPENSSL_PIC -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DBSAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/" -DENGINESDIR="/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib/engines-1.1" -Wall -O3 -pthread -march=armv7-a -Wa,--noexecstack -fPIC -DOPENSSL_USE_NODELETE -Wl,-znodelete -shared -Wl,-Bsymbolic -Wl,-soname=libcrypto.so.1.1 -o ./libcrypto.so.1.1 -Wl,--whole-archive,--version-script=crypto.map ./libcrypto.a -Wl,--no-whole-archive -ldl
./libcrypto.a: member ./libcrypto.a(cms_pwri.o) in archive is not an object
collect2: error: ld returned 1 exit status
Makefile.shared:197: recipe for target 'link_shlib.linux-shared' failed
make[2]: *** [link_shlib.linux-shared] Error 1
make[2]: Leaving directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/openssl-1.1.0f'
Makefile:639: recipe for target 'libcrypto.so' failed
make[1]: *** [libcrypto.so] Error 2
make[1]: Leaving directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/openssl-1.1.0f'
Makefile:128: recipe for target 'all' failed
make: *** [all] Error 2

Any input on how to troubleshoot this error?

Thanks,
Logan

unicornss, 2018-04-04:
Can you pl. try the steps in
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#building-from-source

> Important :  Change current working directory to the kinesis-video-native-build directory first. Then run the ./install-script from that directory.

Thanks

herosjourney, 2018-04-05:
I followed the steps in this link.

I have the following build prerequisites installed:
-autoconf 2.69
-cmake 3.7.2
-flex 2.6.1
-bison 3.0.4
-automake 1.15
-libtool 2.4.6
-java 1.8.0_65

I changed the working directory to kinesis-video-native-build.

I then ran the ./install-script from that working directory. 

These are the last few lines from the terminal install:

make[2]: Entering directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/openssl-1.1.0f'
LD_LIBRARY_PATH=.: gcc -DDSO_DLFCN -DHAVE_DLFCN_H -DNDEBUG -DOPENSSL_THREADS -DOPENSSL_NO_STATIC_ENGINE -DOPENSSL_PIC -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DBSAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/" -DENGINESDIR="/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/engines-1.1" -Wall -O3 -pthread -march=armv7-a -Wa,--noexecstack -fPIC -DOPENSSL_USE_NODELETE -Wl,-znodelete -shared -Wl,-Bsymbolic -Wl,-soname=libssl.so.1.1 -o ./libssl.so.1.1 -Wl,--whole-archive,--version-script=ssl.map ./libssl.a -Wl,--no-whole-archive -L. -lcrypto -ldl
./libssl.a: member ./libssl.a(d1_lib.o) in archive is not an object
collect2: error: ld returned 1 exit status
Makefile.shared:197: recipe for target 'link_shlib.linux-shared' failed
make[2]: *** [link_shlib.linux-shared] Error 1
make[2]: Leaving directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/openssl-1.1.0f'
Makefile:5660: recipe for target 'libssl.so' failed
make[1]: *** [libssl.so] Error 2
make[1]: Leaving directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/openssl-1.1.0f'
Makefile:128: recipe for target 'all' failed
make: *** [all] Error 2

MushMal, 2018-04-05:
@herosjourney thanks for contacting.

From the shell logs it seems that you are either missing a static library which should have been compiled or the library is not properly packaged for the linker to load.

I am not sure I understand the scenario too. Have you initially tried running the install-script from outside of the kinesis-video-native-build directory and then the second time tried from within? 

This will likely not work as the paths might be different.

Also, can I ask for some information about your compiler toolchain? Just thinking that it might be old or the tools might be for non-ARM platform?

bdhandap, 2018-04-05:
@herosjourney As you have tried to install the library from a different folder the paths have been messed up in the downloads folder. Can you please try removing the entire folder `<path>/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads` and try again launching the `./install-script.sh` from the kinesis-video-native-build folder?

herosjourney, 2018-04-06:
@bdhandap The build was successful after I removed the folder and launched the install script from the kinesis-video-native-build folder.

However, now when I attempt to provide the Kinesis Video Stream name I receive a different error.

I provide the AWS Access Key ID and Secret Access Key via the terminal. Then I attempt ./kinesis_vide-streamer_sample_app <MyStreamName here> (This is all listed in Step 41 of the Raspberry Pi guide here: https://s3-us-west-2.amazonaws.com/kinesis-video-streaming/kinesis-video-raspberry-pi-getting-started.pdf)  

I receive the following error: error while loading shared libraries: liblog4cplus-1.2.so.5: cannont open shared object file: No such file or directory. 

I assume this means that this shared library doesn't exist. However, I'm not sure how the library doesn't exist if the build was successful.

unicornss, 2018-04-06:
Hi @herosjourney  Please refer [readme](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md) and let us know if that helps.

> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md#library-not-found-error-when-running-the-demo-application

herosjourney, 2018-04-07:
@unicornss Yes, changing this did solve the issue.

Now, I'm getting a segmentation error:

INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
aivHeapInit(): Failed to allocate heap with limit size 536870912
heapInitialize(): Failed to initialize native heap.
heapRelease(): Freeing native heap.
Segmentation fault

I've read some forums on segmentation faults for Raspberry Pi's, but it's not clear what the recommended solution is.

chehefen, 2018-04-07:
can you try change this line `device_info.storageInfo.storageSize = 512 * 1024 * 1024;`  to something lower like 256 or even 64?

herosjourney, 2018-04-08:
@chehefen Ok. I changed the setting to 64 * 1024 * 1024. 

I then went back and got the same error:

INFO - Initializing curl.

createKinesisVideoClient(): Creating Kinesis Video Client
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
heapInitialize(): Creating AIV heap.
aivHeapInit(): Failed to allocate heap with limit size 536870912
heapInitialize(): Failed to initialize native heap.
heapRelease(): Freeing native heap.
Segmentation fault

I doubled checked that I had changed the setting. I opened the file in nano and confirmed  I had changed it to 64 * 1024 * 1024.

unicornss, 2018-04-09:
Now that you have changed the size of the memory to be 64*1024*1024, could you pl. try running the demo after compiling the application again by running `./install_script` as @chehefen  has mentioned?

briancarbs, 2018-04-11:
Thanks for pointing out this one! I preemptively changed the setting to 64 * 1024 * 1024 ahead of compiling on the raspi3 B+. Worked like a charm! @chehefen Let me know if you want me to dump a tar of the compiled binary somewhere.

unicornss, 2018-04-11:
Glad to hear that you were able to run streaming successfully with Pi B+.

Closing the issue for now.  Pl. reopen if needed. Thanks

zhiyua-git, 2018-04-14:
Hi @herosjourney,

Please use 
$ export AWS_ACCESS_KEY_ID={AccessKeyId}
$ export AWS_SECRET_ACCESS_KEY={SecretAccessKey}

instead of AWS_ACCESS_KEY. Let me know if this helps.

herosjourney, 2018-04-14:
@zhiyua-git Yes, I did add my actual Access Key and Secret Access Key. I just left those out of this thread b/c I didn't want to actually put them on here.

zhiyua-git, 2018-04-14:
@herosjourney Could you confirm you were using AWS_ACCESS_KEY_ID instead of AWS_ACCESS_KEY? Thanks.

herosjourney, 2018-04-14:
@zhiyua-git Ok. I figured out my issue. I was entering each line item separately (AWS_ACCESS_KEY_ID=<your aws_access_key_id>
AWS_SECRET_ACCESS_KEY=<your aws_secret_access_key>
./kinesis_video_gstreamer_sample_app ), instead of entering them all together. 

It now works intermittently. The stream works for about 30 seconds. I'll show the error logs in a separate message below.

zhiyua-git, 2018-04-15:
@herosjourney From the log you provided, your network bandwidth is not high enough to handle the stream to upload, so it was causing buffer pressure. Your allocated buffer was almost full (65808871 = 62Mb) and failed allocation of temporary buffer for new frames caused fatal failure. Can you tell us your bandwidth status and send us a full log file that we can understand and help you more? You can send the information to kinesis-video-support@amazon.com.

You can also refer to [Error Code Reference](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html) to better understand these errors and deal with them.

In your case, Error code 0x5200002e is out of memory error as your buffer was almost full. Error code 0x32000002 indicates your camera was producing b frames which we don't support. Can you double-check the setting of your camera? 

Thank you very much.

herosjourney, 2018-04-23:
@zhiyua-git Do you happen to have a link on how to modify the gstreamer config to turn off b frames? I've looked around for any documentation, but can't find any.

Also, is b frames active by default on gstreamer? 

As for the bandwidth, my upstream is about 5Mbps. 

Also, I can send the var/log/syslog and var/log/messages to the email address above.

unicornss, 2018-04-24:
Hi @herosjourney 

>> Current view duration: 866770000
>> Overall view duration: 880000000 

Log shows that the network connection is  _not_ able to send streams fast enough than what is being generated from the device.In good network  connection you will see  

> Current view duration : 0

The network latency issue needs to be resolved first to send the video frames successfully. 

Encoder settings are already configured in the sample application; in case you would like to change, please refer  to the [Sample application](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp)

```
 /* configure encoder */
    if (!data.h264_stream_supported){
        if (vtenc) {
            g_object_set(G_OBJECT (data.encoder), "allow-frame-reordering", FALSE, "realtime", TRUE, "max-keyframe-interval",
                              45, "bitrate", bitrateInKBPS, NULL);
        } else if (isOnRpi) {
            g_object_set(G_OBJECT (data.encoder), "control-rate", 1, "target-bitrate", bitrateInKBPS*10000,
                "periodicity-idr", 45, "inline-header", FALSE, NULL);
        } else {
            g_object_set(G_OBJECT (data.encoder), "bframes", 0, "key-int-max", 45, "bitrate", bitrateInKBPS, NULL);
        }
}

```

herosjourney, 2018-04-24:
Got it.

I'll work on this on my corporate network instead of the business class
cable modem service that I'm testing this on. And then, I'll be able to see
if I get any different results.

unicornss, 2018-04-25:
@herosjourney 

 Could you provide details on the camera (usb or [camera module](https://www.raspberrypi.org/products/camera-module-v2/) ) and what video resolutions you are using for streaming?  and short description of how are you planning to use Kinesis Video Streams is also helpful for us to support. Thanks.

herosjourney, 2018-04-25:
@unicornss 

The camera is the Raspberry Pi camera module v2. 

The resolution is set to the default (1920X1080)

I'm using Kinesis Video Streams to ingest video from the Raspberry Pi in order to do object and potentially facial recognition with the video. After the video is ingested into Kinesis Video Streams, my intention is give Rekognition access to the stream and build a facial database to recognize and then notify via SMS on the presence of certain faces within the video stream.

unicornss, 2018-04-27:
@herosjourney 

Glad to hear that you now have a stable streaming at **1920X1080** resolution. 

The sample application by default uses settings for highest resolution that camera your can support and best quality. That's the reason you are seeing higher bit-rate from your video stream.

You certainly can reduce the bps for your video streaming, and there are many options.

One of the immediate option is to reduce the resolution (1280x 720 or 640x480) which should reduce the bps of your video stream significantly.  

[Usage](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md) : `AWS_ACCESS_KEY_ID=<SAMPLEKEY> AWS_SECRET_ACCESS_KEY=<SAMPLESECRET> ./kinesis_video_gstreamer_sample_app -w 640 -h 480 <your-stream-name> `

Let us know if that works for your use case. 

Thanks.

herosjourney, 2018-04-27:
@grishmahabib Did you re-run the install script after changing the setting that @chehefen recommended? 

That worked for me.

grishmahabib, 2018-04-27:
@herosjourney 
You solved this problem by  entering each line item separately (AWS_ACCESS_KEY_ID=
AWS_SECRET_ACCESS_KEY=
./kinesis_video_gstreamer_sample_app ), instead of entering them all together.

But when I do this, I get 
**bash: export: `./kinesis_video_gstreamer_sample_app': not a valid identifier**

How do I solve this now?

herosjourney, 2018-04-27:
@grishmahabib 

Here are the steps I followed to resolve the fragmentation error:

1) Changed the setting in Line 97 here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L97 
2) Re ran the install script
3) Enter the AWS_ACCESS_KEY_ID, AWS_SECRET_ACCESS_KEY and ./kinesis_video_gstreamer_sample_app (enter name of your Kinesis Video stream here) 

I'm wondering if you didn't enter your stream name in the last part of #3 above. Also, you have to enter the access key, secret access key and the path for the sample app as one line, not separate lines.

MushMal, 2018-04-27:
@herosjourney thanks a lot for the info. 
@grishmahabib, could you please try the following from Readme.md: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#certificate-store-integration

It seems like your system doesn't have the CA set

herosjourney, 2018-04-27:
@mushmal is correct. You’ll need to install that pem file. I did that as well. I just forgot to mention it.

abhitechie, 2018-05-19:
@herosjourney @grishmahabib 
I checked following steps already  based on your suggestions above : 
- CA set is already installed at /etc/ssl/cert.pem
- storage setting is changed to 512 mb. 
- Executed AWS_ACCESS_KEY_ID=<> AWS_SECRET_ACCESS_KEY=<> ./kinesis_video_gstreamer_sample_app <rtsp_url> <stream-name> a single line. 
Despite all above, I am still getting following error logs : 

INFO - Initializing curl.
INFO -
createKinesisVideoClient(): Creating Kinesis Video Client
INFO -
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
INFO -
heapInitialize(): Creating AIV heap.
INFO -
heapInitialize(): Heap is initialized OK
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747532 Expiration: 0
INFO - New credentials expiration is 1526749932
DEBUG - createDeviceHandler invoked
INFO -
createDeviceResultEvent(): Create device result event.
DEBUG - clientReadyHandler invoked
DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream scorer-test-stream
INFO -
createKinesisVideoStream(): Creating Kinesis Video Stream.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747532 Expiration: 0
DEBUG - Awaiting for the stream to become ready...
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5513 milliseconds
WARN - HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2IKHNMIVNLEBTGA/20180519/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=6a272e335600aa4479f43f978d442b33332adb2319fdeea48bfee5fa4cf1ec9e
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180519T163212Z
DEBUG - describeStream response:
INFO - Describe stream did not find the stream scorer-test-stream in Kinesis Video (stream will be created)
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747538 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5516 milliseconds
WARN - HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2IKHNMIVNLEBTGA/20180519/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=d35d228d180c902159f0d86362c0d63510f36a2d6ecd2aecd631b1659079654d
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180519T163218Z
DEBUG - describeStream response:
INFO - Describe stream did not find the stream scorer-test-stream in Kinesis Video (stream will be created)
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747544 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5515 milliseconds
WARN - HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2IKHNMIVNLEBTGA/20180519/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=aad4f84c6753adcfa2b0d4eeb1fa72465fa827d2ad15d6835cf34501dbbc3cfb
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180519T163224Z
DEBUG - describeStream response:
INFO - Describe stream did not find the stream scorer-test-stream in Kinesis Video (stream will be created)
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747550 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5514 milliseconds
WARN - HTTP Error 0: Response:
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAJ2IKHNMIVNLEBTGA/20180519/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=22ee046f0b89982f08a1c7c6996305cc211f6385786eaef1a68dba4cda662ba7
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180519T163230Z
DEBUG - describeStream response:
INFO - Describe stream did not find the stream scorer-test-stream in Kinesis Video (stream will be created)
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1526747557 Expiration: 0
ERROR - Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.
Aborted

Can you please suggest what could be the possible troubleshooting steps  for above ?
For your information, If I executed "aws kinesisvideo describestream <stream-name>, output is fine as expected. So I assume credential details are good to go. 

Any leads, please ?

abhitechie, 2018-05-22:
Yes, I did. That log was little confusing to me as well. But then I verified through executing 'aws kinesisvideo describe-stream' cli call, it was able to successfully report desired output. 

Then, to debug the issue further, I changed timeout values in "kinesis-video-producer/src/KinesisVideoProducer.h", and after I was able to avoid timeout (but not always) for describe-stream call  but code fails further with timeout on putMedia call. 
So, Now I am wondering if it is a bandwidth issue causing timeout? 

------- Following are the lines of I changed to increase timeout kinesis-video-producer/src/KinesisVideoProducer.h
/**
 * Client ready timeout duration.
 **/
//#define CLIENT_READY_TIMEOUT_DURATION_IN_SECONDS 15
#define CLIENT_READY_TIMEOUT_DURATION_IN_SECONDS 60

/**
 * Stream ready timeout duration.
 **/
//#define STREAM_READY_TIMEOUT_DURATION_IN_SECONDS 30
#define STREAM_READY_TIMEOUT_DURATION_IN_SECONDS 120
/**

MushMal, 2018-05-22:
@abhitechie it seems that your backend calls are constantly timing out or you have no network.

* Please ensure adequate network bandwidth is allotted and you have adequate latency. As your describe stream calls are failing/timing out, I tend to believe that this is not a bandwidth or latency issues but more of a network topology/physical network layer issues. Please ensure you have adequate networking on the device.
* Some proxies might be blocking the network ports. Please ensure your ports are not blocked.
* Try a closer region to you. You can configure the region by setting AWS_DEFAULT_REGION env variable.

abhitechie, 2018-05-25:
Hi @MushMal, Thanks for your response. However, as I mentioned in my above comment, I was able to avoid timeout for describe-stream by changing CLIENT_READY_TIMEOUT_DURATION_IN_SECONDS from 15 to 60 and STREAM_READY_TIMEOUT_DURATION_IN_SECONDS from 30 to 120, but in this case program fails at later stage say "putMedia" call. Also, when I executed describe-stream command through aws cli, it was successful. So, I don't think any network is blocked completely. Thats where my suspect is on slow network. I wanted to understand what is the minimum bandwidth expected by program and what are typical ways to reduce bandwidth requirements e.g. changing resolution, increasing some other timeout or any other configuration ?

MushMal, 2018-05-25:
@abhitechie it does seem like you have major latency issues. The bandwidth requirement for the control plane APIs (describe, create, etc...) is very minimal - the exchange of the information is very small. However, the bandwidth for PutMedia call entirely depends on the density of your stream.

My suggestion would be understand why there is such a high latency for the calls.

As to how to lower the bandwidth you can try to lower the FPS, resolution and increase the compression ratio of your encoder (by dropping the quality).

abhitechie, 2018-05-27:
@MushMal Thanks for your comment. I am not network expert but tried following 

pi@RP3-2835-a32082-a08a8bc9:~/Work $ sudo traceroute -T kinesisvideo.us-west-2.amazonaws.com
traceroute to kinesisvideo.us-west-2.amazonaws.com (34.214.37.45), 30 hops max, 60 byte packets
 1  my.totolink.net (192.168.1.1)  1.862 ms  2.963 ms  2.969 ms
 2  abts-north-static-068.127.176.122.airtelbroadband.in (122.176.127.68)  7.941 ms  7.993 ms  8.096 ms
 3  125.19.32.53 (125.19.32.53)  7.700 ms 125.19.65.189 (125.19.65.189)  7.754 ms 125.19.32.37 (125.19.32.37)  7.264 ms
 4  182.79.217.217 (182.79.217.217)  242.655 ms 182.79.222.29 (182.79.222.29)  261.061 ms 182.79.190.70 (182.79.190.70)  248.633 ms
 5  paix01-sfo4.amazon.com (198.32.176.36)  272.850 ms 182.79.191.37 (182.79.191.37)  266.672 ms paix01-sfo4.amazon.com (198.32.176.36)  260.058 ms
 6  54.240.243.8 (54.240.243.8)  256.991 ms 182.79.175.214 (182.79.175.214)  53.436 ms 54.240.243.28 (54.240.243.28)  274.595 ms
 7  54.240.243.101 (54.240.243.101)  271.950 ms 182.79.201.83 (182.79.201.83)  57.745 ms 54.240.243.37 (54.240.243.37)  253.479 ms
 8  182.79.234.219 (182.79.234.219)  241.467 ms * *
 9  paix01-sfo4.amazon.com (198.32.176.36)  273.115 ms 54.239.43.55 (54.239.43.55)  281.351 ms paix01-sfo4.amazon.com (198.32.176.36)  261.836 ms
10  54.240.243.108 (54.240.243.108)  282.641 ms 54.240.243.128 (54.240.243.128)  261.390 ms 54.240.243.68 (54.240.243.68)  279.372 ms
11  54.240.243.155 (54.240.243.155)  253.611 ms 54.240.243.35 (54.240.243.35)  265.268 ms 54.240.243.101 (54.240.243.101)  264.609 ms
12  * * 52.93.12.183 (52.93.12.183)  275.590 ms
13  54.239.43.55 (54.239.43.55)  287.317 ms * *
14  52.93.12.252 (52.93.12.252)  286.412 ms * *
15  52.93.240.69 (52.93.240.69)  286.063 ms * 52.93.12.46 (52.93.12.46)  320.082 ms
16  52.93.12.215 (52.93.12.215)  291.948 ms 52.93.13.40 (52.93.13.40)  310.025 ms 52.93.12.222 (52.93.12.222)  307.537 ms
17  * * *
18  52.93.12.59 (52.93.12.59)  283.951 ms 52.93.12.173 (52.93.12.173)  281.704 ms *
19  * * *
20  52.93.15.217 (52.93.15.217)  286.225 ms * *
21  * * *
22  * * *
23  * * ec2-34-214-37-45.us-west-2.compute.amazonaws.com (34.214.37.45)  294.282 ms
24  * * *
25  * * *
26  * * ec2-34-214-37-45.us-west-2.compute.amazonaws.com (34.214.37.45)  298.917 ms
27  * * *
28  * * *
29  * * *
30  * * *

From above, I can see that latency comes from remote servers. At least at 23rd hope, my request has reached amazon network, but despite traceroute could not finish in 30 hops. I even tried by increasing max ttl to 60, but still traceroute could not finish. I am little bit stuck here, as from above it looks like latency is at remote site. Are there other better ways to diagnose network latency for amazon requests?

MushMal, 2018-05-27:
It's hard to tell what's going on. From the logs I can see that the device is not able to stream at all as the Producer SDK doesn't move past 'describe' state:

ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Timeout was reached: Resolving timed out after 5522 milliseconds
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=Changed/20180522/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=fadd6ed469f96bc4b51cad0a53241b30eefe15f6f3a677f302d05692a9caef06
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180522T174613Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream scorer-test-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.

It simply times out.

Would you be able to run some network diagnostics on the device itself first? Could you use the device with another AP/network entirely to eliminate the network issue?

abhitechie, 2018-05-28:
Hi @MushMal , Thanks for your comments. Finally, I was able to solve the problem. 
Actually my raspi had eth0 and wlan0, two interfaces both were up. But eth0 did not have route to external network while wlan0 has it.  When both interfaces are up, sample application somehow chooses eth0 only and fails to connect. While I was expecting that it will automatically select an interface which has internet connectivity but its not the case. When I turned eth0 off, it started communicating through wlan0 and everything was fine then. 

Thanks for your support during the course. Your suggestions/comments were quite helpful.

MushMal, 2018-05-28:
Sounds really cool. Indeed, the linux kernel will prioritize the wired connection by default so it was falling back on eth0 which didn't have the logical connection. We will take this scenario into account for documentation and/or diagnostics for the future tools.

Could I ask you to mark this as resolved please?

unicornss, 2018-05-29:
Closing the issue as it has been resolved; please reopen if support is required. 
Thanks.

closed 2018-05-29 by unicornss