awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #63: How to get stream info from FragmentAckReceivedFunc callback

ameyat2812, 2018-04-05:
I have multiple KinesisVideoStreams created and linked to KinesisVideoProducer instance. 

I have extended StreamCallbackProvider to override getFragmentAckReceivedCallback which provides my own version of onFragmentAck function.

I am getting the acknowledgements just right however I am not able to which stream they belong to (As I am uploaded multiple streams).

Is there any way I can get access to stream_info to identify which stream this ack is for.

I am using 1.2.1 SDK version BTW

MushMal, 2018-04-05:
@ameyat2812 thanks for reaching out.

Can I ask for more information about what you are attempting to do to help us understand it better. What's your end-to-end scenario?

This is a great questions and I would like to take time to explain the patterns better. As you have seen, you can create a single KVS producer object (the Client) and have multiple stream objects hanging form it (the Stream). This is a log more preferred pattern than having multiple single client single stream pairs. The reason is that multiple streams share the content store in a single client object.

Then, the question is how to distinguish between the streams in a callback? The C++ SDK acts as a client layer for PIC (Platform Independent Codebase). PIC communicates with the "outside" world through events and callbacks. Each stream in PIC is identified via a STREAM_HANDLE opaque handle. In order not to create a yet-another translation layer (as we already have to store the application callbacks and provide C++ SDK callbacks to PIC) in C++ layer, it was decided to expose the same callbacks from PIC through C++ layer. Thus, the identity of the KVS stream is 1:1 match to the STREAM_HANDLE handle.

What the application can do is to create a stream and note it's STREAM_HANDLE by calling getStreanHandle. Then, in a callback, you will be returned the STREAM_HANDLE corresponding to the stream the callback is called for.

Please let us know if this is helpful and if we can help more.

ameyat2812, 2018-04-07:
HI @MushMal. Thanks for your prompt reply. I am now mapping stream_handle to kinesisvideostream context and I am able to filter acknowledgements based on stream_handle now. 

If I am right the timestamp field in PFragmentAck gives out relative offset from the producer timestamp of first frame submitted. Correct me if I am wrong! and is there a way to get an absolute timestamp instead. 

P.S. The end to end scenario - I am creating multiple streams for varied purposes and they are all attached to KinesisVideoProducer instance. I am maintaining a metadata DB for some of the streams and am tagging the metadata by framgment_id and producer timestamp.

MushMal, 2018-04-07:
@ameyat2812 it's is a cool idea.

Regarding the ACK times: Our underlying streaming format is MKV. It has frames and clusters which are comprised from a group of frames. Frame timecodes are all relative to the beginning of the cluster they are in. However, the cluster timecodes (which FragmentAck.timestamp reflects) can be relative from the beginning of the "presentation" or absolute. This is controlled by StreamDefinition.absolute_fragment_times property when creating/starting stream. 

This said, the "absolute" time will be defined by whatever the system is configured with as the C++ implementation will utilize std::chrono::system_clock::now().

These are great questions - please keep them coming and I am looking forward to learn more about your product and how it's used.

unicornss, 2018-04-13:
Closing this issue for now as the question has been clarified.  Please reopen if required.
Thanks

closed 2018-04-13 by unicornss