awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #64: Failed to submit frame to Kinesis Video client. status: 0x30000005

davidgatti, 2018-04-05:
As shown bellow 

```
DEBUG - Wrote 1756 bytes to Kinesis Video. Upload stream handle: 1

on_new_sample(): kinesis_video_flags=0 pts=4289336818 dts=4289336818
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42893368 presentation timestamp: 42893368

on_new_sample(): kinesis_video_flags=0 pts=4256005091 dts=4256005091
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42560050 presentation timestamp: 42560050

on_new_sample(): kinesis_video_flags=0 pts=4322673351 dts=4322673351
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 43226733 presentation timestamp: 43226733

on_new_sample(): kinesis_video_flags=0 pts=4489341597 dts=4489341597
DEBUG - Wrote 1180 bytes to Kinesis Video. Upload stream handle: 1

on_new_sample(): kinesis_video_flags=0 pts=4422676498 dts=4422676498
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44226764 presentation timestamp: 44226764

on_new_sample(): kinesis_video_flags=0 pts=4389344719 dts=4389344719
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 43893447 presentation timestamp: 43893447

on_new_sample(): kinesis_video_flags=0 pts=4456012928 dts=4456012928
ERROR - Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44560129 presentation timestamp: 44560129
```

Some frames don't get saved to KVS, is this because of how the movie was encoded? Because I have other samples where I don't have any issues. 

I'm asking to be sure where to look, and also to better understand your producer app.

unicornss, 2018-04-05:
Pl. refer the details on the status code here: [Status Codes Reference for Producer SDK integration with Media pipeline (camera and other video stream sources)](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html)

MushMal, 2018-04-05:
@davidgatti , as @unicornss mentioned, Status Codes Reference doc is a great source of information when it comes to issues like these. 

Looking at your case, please ensure you have a stable encoder. We could try to help a little more if you describe what your scenario is, what's your setup - media pipeline (GStreamer? what kind of elements, etc...). What's your end-to-end scenario?

davidgatti, 2018-04-06:
OO, cool that page is on point, thanks for the tip @unicornss. @MushMal regarding your proposition. Right now I'm doing R&D for a client and see how to take advantage of KVS. This is my test setup:

# VLC Server - no transcoding

```
vlc -loop -vvv bunny.mp4 --sout '#rtp{sdp=rtsp://:8554/}'
```

Then I paste the RTSP URI, to the C++ app, and of it goes to KVS.

# VLC Server - with transcoding

```
vlc -loop -vvv blender.mov --sout '#transcode{vcodec=h264,scale=Auto,acodec=mpga,ab=128,channels=2,samplerate=44100}:rtp{sdp=rtsp://:8554/}'
```

Then I paste the RTSP URI, to the C++ app, and of it goes to KVS, while some time missing some frames.

# VLC Server - with transcoding and image resize

```
vlc -loop -vvv blender.mov --sout '#transcode{vcodec=h264,scale=Auto,width=640,acodec=mpga,ab=128,channels=2,samplerate=44100}:rtp{sdp=rtsp://:8554/}'
```

Then I paste the RTSP URI, to the C++ app, and of it goes to KVS, while some time missing some frames.

# My additional questions

This are some questions that I have after reading the error page.

> The encoder produces a burst of encoded frames, and their time stamps are smaller than the intra-frame durations. 

Is this sentence trying to imply that the VLC in this case is sending frame faster then the playback itself?

> The stream is configured to use SDK time stamps, and the frames are sent faster than the frame durations. To help with some "jitter" in the encoder, specify a smaller frame duration in the StreamInfo.StreamCaps structure. For example, if the stream is 25FPS, each frame's duration is 40 ms. However, to handle the encoder jitter, we recommend that you use half of that frame duration (20 ms). Some streams require more precise control over the timing for error detection.

This "jitter" situation is bit strange, let me explain: from the AWS materials what I got is that KVS will set its own time stamp to the frames, this way "I" don't have to worry about that. So the data is saved correctly. 

Based on the above quote it seams that there is more granular control, and either I set a time stamp or you do. Is this the correct assumption?

# Why should KVS care about the speed?

Why should KVS care if the frame are coming faster or slower? Can't KVS just buffer them, and save them in a FIFO manner?

MushMal, 2018-04-06:
@davidgatti thanks for the details. 

Some encoders/muxers might delay the frames based on the complexity of the encoding resulting in smaller intra-frame delays. In your case its possible that the VLC does transcoding and produces burst in frames. KVS stream can be configured to use the provided frame timestamps or timestamp it directly as the frames are pushed into the KVS. StreamDefinition.frame_timecodes is set to "true" by default which means that the KVS will use the frame timestamp. If we drop a frame with the "overlap" error this means that the encoder upstream has produced timestamps which overlap with the previous frames timestamp + the previous frames duration. If the "jitter" is small, you can attempt to shorten the frame duration (I haven't experimented with the VLS server with transcoding). 
Some encoders do produce bursts of frames with no timestamps so the KVS based application would use StreamDefinition.frame_timecodes = false (so the KVS will timestamp on arrival of the frame) and have a temporal buffer of a few frames deep that would accumulate the frames and release to KVS in a delayed order. Others, would use very small frame durations to handle the jitter. The default in the sample application is 20 ms. (create_kinesis_video_frame())

In either case, KVS will utilize the internal buffer.

We are working on adding some heuristics to the sample application that would select the frame durations based on the FPS and more resilient to the jitter.

davidgatti, 2018-04-09:
OK, interesting, thanks for the extra information - i'll take the time to process it :)

closed 2018-04-09 by davidgatti

sailesharya, 2019-02-28:
> @davidgatti , as @unicornss mentioned, Status Codes Reference doc is a great source of information when it comes to issues like these.
>
> Looking at your case, please ensure you have a stable encoder. [...]

> @davidgatti , as @unicornss mentioned, Status Codes Reference doc is a great source of information when it comes to issues like these.
>
> Looking at your case, please ensure you have a stable encoder. [...]

I am trying with rtsp rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov and still getting 0x30000005 issue
any idea?

guillermoAlv, 2019-03-21:
> I am trying with rtsp rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov and still getting 0x30000005 issue
> any idea?

have you make it work? i'm having the same error in a android emulator

davidgatti, 2019-03-23:
@guillermoAlv I didn't do Android development for over 10 years now, but I'm 99.99% sure Google didn't improve it much ;) and way back when it was super slow, and hardware acceleration was not a thing Google was aware of ;) I'm sure they didn't improve much with it.

Meaning you'll never have a good stream - but if they maybe updated the emulator (haha), search for an option for hardware acceleration (good luck), and that should improve things (lol).

Sorry for the sarcasm, I had to do it :D