awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #66: Minimum hardware spec

davidgatti, 2018-04-06:
Hello, 

Based on this issue #26 what would be a good memory allocation per stream? And do you have a way to com up with a way to calculate that? by taking in consideration resolution? kbps? and other factors?

davidgatti, 2018-04-09:
Right but in that issue your are not saying anything about how to come up with a good number, you are just writing that you can set the amount fo memory used by yourself. 

Now now that I red the issue one more time, I have the following questions:

1. is `storageSize` used for buffering the video when there is not connectivity to KVS?
1. if the previous point turns out to be true, then i can set `storageSize` even to 1MB, meaning this setting is not used by the app itself to work?

MushMal, 2018-04-09:
@davidgatti thanks for you question. BTW, we will be updating our documentation to detail all this information and making the sample applications be smarter about allocation sizes.

1) DeviceInfo.StorageInfo.storageSize represents the overall memory footprint the SDK is allowed to pre-allocate/reserve for the storage. This will act as storage to accumulate the frame data from all streams until it's no longer needed. This can happen in two cases a) the stream Persisted ACK comes in or b) The frame rolls off the temporal window and no longer needed. In case of b) if the frame has not been sent to the cloud then a DroppedFrame callback will be called for the application act upon (alarm perhaps, etc..). All streams share this content store but have their unique content view which governs the temporal allocation, etc.. As I mentioned in the post #26 the allocator will attempt to "reserve" - meaning it will simply block virtual contiguous memory on some platforms and not to back it up with physical pages which allows those pages to be used by the OS. The memory usage will be increased as the pressure arises to store more data but in a general case of adequate bandwidth the memory will not be utilized. Some applications can configure their store and the view to be able to absorb multi-minute or even hour temporary storage.
2) The minimum allocation size for the content store is 10 MB - not sure whether you need smaller storage size.

davidgatti, 2018-04-10:
1. now this is more clear 
2. it was just an extreme example ;)

closed 2018-04-10 by davidgatti