awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #68: Won't Build with ./install-script or ./min-install-script - Mac OS 10.13.4

jacksonrdlc, 2018-04-09:
Really need to get this to work. Any help would be amazing. I have installed the recommended project dependencies but I feel like the issues are that the project is running on older packages and maybe the solution is incompatible with newer packages.

When I run `./install-script`,  the script abruptly stops here:
```
checking for bison... /usr/bin/bison
checking bison version 2.3 >= 2.4... configure: error: no
```

When I run `./min-install-script`,  the script blows up like this:
```
[ 97%] Building CXX object CMakeFiles/kinesis_video_gstreamer_sample_app.dir/Users/jack.rudelic/src/hackathon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp.o
/Users/jack.rudelic/src/hackathon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp:1:10: fatal error: 
      'gst/gst.h' file not found
#include <gst/gst.h>
         ^~~~~~~~~~~
1 error generated.
make[3]: *** [CMakeFiles/kinesis_video_gstreamer_sample_app.dir/Users/jack.rudelic/src/hackathon/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp.o] Error 1
make[2]: *** [CMakeFiles/kinesis_video_gstreamer_sample_app.dir/all] Error 2
make[1]: *** [CMakeFiles/kinesis_video_gstreamer_sample_app.dir/rule] Error 2
make: *** [kinesis_video_gstreamer_sample_app] Error 2
```

I have rigorously researched and attempted to address both errors. I have uninstalled and reinstalled the project and have uninstalled and reinstalled bison and gstreamer which i thought maybe had bad initial builds.

If you have seen similar issues or have an idea on a solution. Please help.

unicornss, 2018-04-09:
@jacksonrdlc 
Could you please tell us how did you install the dependencies? using 
`brew install bison`, 
`brew install gstreamer`,
 `brew install pkg-config`?  

If that is the case, before running `./install-script`  - you can try exporting the PATH of bison as below (change the bison directory as per you installed version of bison).
` export PATH=/usr/local/Cellar/bison/3.0.4_1/bin:$PATH`

This should get the binaries built.

Also note that depending upon your High Sierra updates there is an open issue as noted in  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/60  when running the demo.  We don't have an update for this issue at the moment for High Sierra with command line tools update.

unicornss, 2018-04-26:
@jacksonrdlc Please try the latest  SDK release that should fix the issue https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/60 .

prvcy, 2018-05-02:
Hi,

I am also getting this error when trying to build on high sierra 10.13.4 (I am using the latest SDK version as well 1.4.0):

```
[ 38%] Linking CXX shared library libproducer.dylib
[ 38%] Built target producer
Scanning dependencies of target kinesis_video_gstreamer_sample_rtsp_app
[ 39%] Building CXX object CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/Users/lab/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp.o
/Users/lab/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp:1:10: fatal error:
      'gst/gst.h' file not found
#include <gst/gst.h>
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/Users/lab/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp.o] Error 1
make[1]: *** [CMakeFiles/kinesis_video_gstreamer_sample_rtsp_app.dir/all] Error 2
make: *** [all] Error 2

```

nsyll, 2019-11-06:
On SDK Release 2.0.1 (25th Jul 2019) and OSx version 10.15
I had the same error 
checking bison version 2.3 >= 2.4... configure: error: no

I had install bison 3.4.2 by brew
so I run  > export PATH=/usr/local/Cellar/bison/3.0.4_1/bin:$PATH
and it works fine