awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #69: g_object_set: assertion 'G_IS_OBJECT (object)' failed

davidgatti, 2018-04-10:
And the saga continues ;) 

I have my EC2 instance where I'm able to stream a RTSP stream from another EC2 instance, and all is good. Now that I know that this works, I decided to go one step further, and actually stream the video coming out from the office camera. The camera spits out RTSP, and I'm able to connect to the stream with VLC, so that works for sure. 

On the Debian installation that I have in the office, i copied from the EC2 server where I compiled the project the following files:

- **file**: kinesis_video_gstreamer_sample_rtsp_app
- **file**: libproducer.so
- **folder**: kinesis-video-native-build/downloads/local/lib

I did put eventually the `libproducer.so` file in the `lib` folder, and set the env variable: `export LD_LIBRARY_PATH=/home/company/Documents/kvs_app/libs`.

Debian info:

- **EC2**: Linux ip-172-31-70-223 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u3 (2018-03-02) x86_64 GNU/Linux
- **Office (VirtualBox)**: Linux debian9 4.9.0-4-amd64 #1 SMP Debian 4.9.51-1 (2017-09-28) x86_64 GNU/Linux

My distribution is slightly behind, but both are **Debian 9**.

When I start the demo app I get the following error: 

```
(kinesis_video_gstreamer_sample_rtsp_app:6470): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Not all elements could be created:
	rtspsrc (gst-plugins-good)
	rtph264depay (gst-plugins-good)
	appsink (gst-plugins-base)

freeKinesisVideoClient(): Freeing Kinesis Video Client
commonHeapDebugCheckAllocator(): Heap is initialized
commonHeapDebugCheckAllocator(): Heap limit: 				536870912
commonHeapDebugCheckAllocator(): Heap size: 				0
commonHeapDebugCheckAllocator(): Number of allocations: 				0
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 				(nil)
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x7fdb21ca2010
aivHeapDebugCheckAllocator(): *******************************************
aivHeapDebugCheckAllocator(): Block:	0x7fdb21ca2010		size:	536870879
aivHeapDebugCheckAllocator(): *******************************************
heapRelease(): Freeing native heap.INFO - Freeing Kinesis Video Stream staging

freeKinesisVideoStream(): Freeing Kinesis Video stream.INFO - Curl shutdown.
```

Any idea how to solve this issue?

unicornss, 2018-04-10:
Hi @davidgatti 

Do you have the libraries (which includes the dependent gstreamer plugin elements libgstbase-1.0 libgstrtsp-1.0 ) also synced to local folder? 

- folder: kinesis-video-native-build/downloads/local/lib

If you have synced the folder from remote into local then you can export the library path as 
```
export LD_LIBRARY_PATH=/home/company/Documents/kvs_app/libs:<absolute path to the libraries copied into local that have been synced from EC2 kinesis-video-native-build/downloads/local/lib >:$LD_LIBRARY_PATH
```
Pl. use the start up script in docker scripts directory for reference:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/4758725db2ae15f795ca533dcc3d3462d8f5e84a/docker_native_scripts/start_rtsp_in_docker.sh#L8

Let us know if you have questions.
Thanks

davidgatti, 2018-04-11:
Hmm I'm not sure we are on the same page. I did mention in my original post that I did export the path to the `libs` folder. And I did it the same way that I did it on the EC2 instance.

Just to add some extra information if I do `ldd kinesis_video_gstreamer_sample_rtsp_app` on my Office machine, I get this:

```
	linux-vdso.so.1 (0x00007ffe18b22000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fddcb433000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fddcb22b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fddcb00e000)
	libproducer.so => /home/COMPANY_NAME/Documents/kvs_app/libs/libproducer.so (0x00007fddcacd0000)
	libgstreamer-1.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libgstreamer-1.0.so.0 (0x00007fddca973000)
	libgstapp-1.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libgstapp-1.0.so.0 (0x00007fddca763000)
	libgobject-2.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libgobject-2.0.so.0 (0x00007fddca511000)
	libglib-2.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libglib-2.0.so.0 (0x00007fddca1d2000)
	liblog4cplus-1.2.so.5 => /home/COMPANY_NAME/Documents/kvs_app/libs/liblog4cplus-1.2.so.5 (0x00007fddc9f57000)
	libcrypto.so.1.1 => /home/COMPANY_NAME/Documents/kvs_app/libs/libcrypto.so.1.1 (0x00007fddc9acd000)
	libssl.so.1.1 => /home/COMPANY_NAME/Documents/kvs_app/libs/libssl.so.1.1 (0x00007fddc985e000)
	libgtest.so => /home/COMPANY_NAME/Documents/kvs_app/libs/libgtest.so (0x00007fddc95a3000)
	libgtest_main.so => /home/COMPANY_NAME/Documents/kvs_app/libs/libgtest_main.so (0x00007fddc93a1000)
	libcurl.so.4 => /home/COMPANY_NAME/Documents/kvs_app/libs/libcurl.so.4 (0x00007fddc9134000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fddc8db2000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fddc8aae000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fddc8897000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fddc84f8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fddcb861000)
	libffi.so.6 => /home/COMPANY_NAME/Documents/kvs_app/libs/libffi.so.6 (0x00007fddc82ed000)
	libiconv.so.2 => /home/COMPANY_NAME/Documents/kvs_app/libs/libiconv.so.2 (0x00007fddc8008000)
	libgmodule-2.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libgmodule-2.0.so.0 (0x00007fddc7e04000)
	libgstbase-1.0.so.0 => /home/COMPANY_NAME/Documents/kvs_app/libs/libgstbase-1.0.so.0 (0x00007fddc7b8e000)
```

Stuff seams to be linked to the right path, I'm also pasting the structure of the `libs` folder. I did use `SCP` to copy the whole folder, so it be strange if something was missing. But in any case, this is what I have.

```
libs
├── charset.alias
├── engines-1.1
│   ├── afalg.so
│   ├── capi.so
│   └── padlock.so
├── gettext
│   ├── cldr-plurals
│   ├── hostname
│   ├── project-id
│   ├── urlget
│   └── user-email
├── gio
│   └── modules
├── glib-2.0
│   └── include
│       └── glibconfig.h
├── gstreamer-1.0
│   ├── libgstaccurip.la
│   ├── libgstaccurip.so
│   ├── libgstadder.la
│   ├── libgstadder.so
│   ├── libgstadpcmdec.la
│   ├── libgstadpcmdec.so
│   ├── libgstadpcmenc.la
│   ├── libgstadpcmenc.so
│   ├── libgstaiff.la
│   ├── libgstaiff.so
│   ├── libgstalaw.la
│   ├── libgstalaw.so
│   ├── libgstalphacolor.la
│   ├── libgstalphacolor.so
│   ├── libgstalpha.la
│   ├── libgstalpha.so
│   ├── libgstapetag.la
│   ├── libgstapetag.so
[... 528 lines cut ...]
│   ├── gstreamer-rtp-1.0.pc
│   ├── gstreamer-rtsp-1.0.pc
│   ├── gstreamer-sdp-1.0.pc
│   ├── gstreamer-tag-1.0.pc
│   ├── gstreamer-video-1.0.pc
│   ├── gthread-2.0.pc
│   ├── libcrypto.pc
│   ├── libcurl.pc
│   ├── libffi.pc
│   ├── libpcrecpp.pc
│   ├── libpcre.pc
│   ├── libpcreposix.pc
│   ├── libssl.pc
│   ├── log4cplus.pc
│   ├── mount.pc
│   ├── openssl.pc
│   ├── x264.pc
│   └── zlib.pc
├── preloadable_libiconv.so
└── preloadable_libintl.so

10 directories, 573 files
```

I'm having a hard time figuring out what is missing since it seams that everything is in the right place.

unicornss, 2018-04-11:
@davidgatti Thanks for the details.  Yes, your export of the environment variable is matching to the local libs folder which contains all the libraries. 

The error shows that it cannot create the plugins:

(kinesis_video_gstreamer_sample_rtsp_app:6470): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Not all elements could be created:
	rtspsrc (gst-plugins-good)
	rtph264depay (gst-plugins-good)
	appsink (gst-plugins-base)

Related code reference: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_rtsp_app.cpp#L372 
```
   if (!data.pipeline || !data.source || !data.depay  || !data.appsink) {
        g_printerr("Not all elements could be created:\n");
        if (!data.pipeline) g_printerr("\tCore pipeline\n");
        if (!data.source) g_printerr("\trtspsrc (gst-plugins-good)\n");
        if (!data.depay) g_printerr("\trtph264depay (gst-plugins-good)\n");
        if (!data.appsink) g_printerr("\tappsink (gst-plugins-base)\n");
        return 1;
}
```
We will see if we can recreate to find what else could be missing since you have the gstreamer libraries synced from remote to local libs folder.  

In the meantime, could you please check if exporting the ```GST_PLUGIN_PATH=/home/company/Documents/kvs_app/libs``` works for your local environment? 

Reference:
* [GstRegistry]( https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gstreamer/html/GstRegistry.html) 
* [Commandline Pluginpath](https://gstreamer.freedesktop.org/documentation/plugin-development/basics/testapp.html)

davidgatti, 2018-04-12:
@unicornss your tip to create a GST_PLUGIN_PATH variable actually worked, and I'm able to stream to KVS. And just so you know, when I do this the app starts with the following warnings

```
(kinesis_video_gstreamer_sample_rtsp_app:3794): GStreamer-WARNING **: External plugin loader failed. This most likely means that the plugin loader helper binary was not found or could not be run. You might need to set the GST_PLUGIN_SCANNER environment variable if your setup is unusual. This should normally not be required though.

(kinesis_video_gstreamer_sample_rtsp_app:3794): GStreamer-WARNING **: Failed to load plugin '/home/COMPANY/Documents/kvs_app/libs/libfl.so': /home/COMPANY/Documents/kvs_app/libs/libfl.so: undefined symbol: yylex
INFO - Initializing curl.
```

Dose this means that I should also point GST_PLUGIN_SCANNER to the libs folder? 

And one last thing for your information, while streaming (actively sending frames to KVS) I get the following warning

```
(kinesis_video_gstreamer_sample_rtsp_app:3794): GLib-GObject-WARNING **: gsignal.c:3375: invalid param spec type 'GParamBoxed' for value type 'GParam'
```

Are this warnings ok, or should be fixed?

unicornss, 2018-04-12:
Hi @davidgatti 
As you are able to stream successfully from local machine you may not need these helper binaries. But if you want, you can copy those binaries

-  gst-completion-helper 
- gst-plugin-scanner     <-- the one you are seeing the warning
- gst-ptp-helper

 from the EC2 build machine folder (`<your sdk download directory>/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/libexec/gstreamer-1.0/
`) to local (say `/opt/gstpluginhelperbinaries/`) and set your PATH variable to include this as below:

`export PATH=/opt/gstpluginhelperbinaries:$PATH`

We have not seen these warning messages before, and if this is happening only in the local it could be due to the glib install version is slightly different.   Please check installed versions in your system and setting the same glib version might help to clear the warning.  Thanks.

davidgatti, 2018-04-13:
aaa that is where they were hiding :D and regarding your hope about different version I did run `ldd --version` on both machines and...

**EC2 Instance**

```
ldd (Debian GLIBC 2.24-11+deb9u1) 2.24
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

**VirtualBox**

```
ldd (Debian GLIBC 2.24-11+deb9u1) 2.24
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

davidgatti, 2018-04-16:
Yep this is it. I think truly putting everything needed in the lib folder should make it even more stream line in the future.

closed 2018-04-16 by davidgatti