awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #70: ./kinesis_video_gstreamer_sample_app : No such file or directory found

HarmanakSingh, 2018-04-12:
Hi!

I have been trying to setup a kinesis video stream using my raspberry pi with a usb camera attached to it.

I have installed all the dependencies and the sdk on the Pi and while I try to stream to Kinesis, it gives me the above mentioned error. 

Please help me out on this.

Thanks

davidgatti, 2018-04-12:
Did you run `./install-script` inside the `kinesis-video-native-build directory`? You have to compile the project to have the executable for your platform.

MushMal, 2018-04-12:
@HarmanakSingh as @davidgatti suggests please run the install-script from within the kinesis-video-native-build directory. Also, could I ask you to clean the repository first? We have seen times when the intermediate libraries get linked with an incorrect path.

HarmanakSingh, 2018-04-13:
Ok. I am running the ./install-script from the native folder. I had previously run it inside the amazon-kinesis-sdk-cpp folder. 

@MushMal Could you please elaborate on what you mean by cleaning the repository?

HarmanakSingh, 2018-04-13:
OK. When i try to run it on the Pi, The Pi is freezing midway through the build. The only thing i can do from there is to unplug it and then re run the command but the build is not successful.

Any idea why the Pi might freeze midway?

davidgatti, 2018-04-16:
Not enough storage? The install scripts downloads a lot of information over 1GB, just an idea.

HarmanakSingh, 2018-04-16:
I dont think so. I am using a 64gb card. All other peripherals are also the RP originals. I am really annoyed with RPi freezing in the mid of a download on any script. Is this an issue with the B+ models?

davidgatti, 2018-04-16:
Sadly I won't be able to help you after this pint, the only thing that you could do is to paste the last 50 lines of the log, so others can take a look at what happened, and maybe find the reason.

closed 2018-04-18 by HarmanakSingh