awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #71: 267.6 MB of Libs, quite the requirement :)

davidgatti, 2018-04-12:
Thanks to you all I managed to make the demo app work. But now I have the following questions. The `libs` folder takes 267.6 MB which is quite a lot. Of course I understand that you need SSL, CURL, GStream and other stuff. 

And from what I read here, your approach seams to... lets put everything that we need with the right version in the libs folder so we minimize any drama with different version of different libs on different systems. 

Which makes sense. So the question is... if I'd like to use this project on a device with limited storage, could I decrees the size of the libs folder? Do we really need all of this? 

And also if I were to use your SDK and write my own app, would I end up in the same situation, needing 100 of MB of libraries?

I'm trying to get a frame of reference how to think about this repo, project, demo app end everything (no 42 is not the answer that I'm looking for 😇).

bdhandap, 2018-04-12:
@davidgatti We are working on the optimizing the downloads folder as once you build and install the the dependencies, all you need is the binaries. The dependencies source can be removed. I'll get back to you once we run the tests with optimized binaries in our lab.

davidgatti, 2018-04-13:
Hmm the 267.6 MB that I refer to is the lib folder, and nothing else. Meaning I'm taking in consideration only:

- .a
- .so
- .la

Are we talking about the same?

[image]

bdhandap, 2018-04-13:
Yes. One part of the optimization effort is to optimize the binaries as well by running `make install-strip` rather than `make install`. But not all the dependencies support `install-strip`option. I'll update this thread once I have something solid.

davidgatti, 2018-04-16:
Cool. In the mean time could you share a tip on more or less what could be easily removed, and was added for super edge cases? I'm planing over the week to start removing libraries and run the app and see at what stage it will start complaining, but if you could push me in the right direction, it be nice, but no rush :)

bdhandap, 2018-05-09:
Also please note that the demo app uses GStreamer pipeline so when you run the `install-script.sh`, it builds the entire GStreamer framework along with its plugins. If you want to integrate only with producer sdk and use your own media pipeline then you can exclude the GStreamer part in the build script which will reduce the size considerably.

bdhandap, 2018-05-10:
One more thing to add : If you look at the `downloads/local/` folder, there is a `share` folder inside which is not required by the SDK/GStreamer to run the demo app. Removing that folder will reduce the size to 76MB. Yet the GStreamer framework and plugin components which are not related to the SDK take half of the remaining space.  

     $:~/workspace/amazon-kinesis-video-streams-producer- sdk-cpp/kinesis-video-native-build/downloads$ du -sh ./local
     76M	./local
     $:~/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads$ du -s ./local/lib/* | grep gst | cut -f 1 | awk '{s+=$1} END {print s}'
     33252
     $:~/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads$ du -s ./local/include/* | grep gst | cut -f 1 | awk '{s+=$1} END {print s}'
     2740
     $:~/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads$ du -s ./local/bin/* | grep gst | cut -f 1 | awk '{s+=$1} END {print s}'
     216
     $:~/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads$ du -s ./local/libexec/* | grep gst | cut -f 1 | awk '{s+=$1} END {print s}'
     48

davidgatti, 2018-05-16:
Got it, thank you very much :)

closed 2018-05-16 by davidgatti