awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #72: How to go about buffering when no Internet connectivity - no RAM allowed ;)

davidgatti, 2018-04-17:
Greetings masters of the stream! 😇

You are probably getting a good sense that I'm building something serious with KVS. I believe that I figure out almost everything–but one last peace of the puzzle. 

How to buffer the video to disk if either:

- there is no internet access
- KVS for whatever reason is down, or not accessible  

I don't want to lose any frames when one of the situation mentioned above happens. Because of this here is my question

> can the producer send frames faster the the real playback if the time stamp is way behind. The goal would be to catch up to real time as fast as possible.

I also have the following idea if the above answer is yes. I was thinking to:

- use ffmpeg to get a stream from the camera and save it to disk
- have another ffmpeg process that would read the file created from the first ffmpeg, and create a RTSP stream for the KVS producer 

Would this be a way to achieve buffering on disk?

# Do you have a time line?

Last thing, in one of the issues–you mentioned that you plan to have an option to choose between buffering in RAM or Disk. But I'd like to know at least if this is something that you will be releasing this year. I don't need to have the exact month or day. But I'd need to know if it is worth waiting, or should I come up with something myself.

MushMal, 2018-04-17:
Lol - that's a great job title I think. I should talk to my bosses about it:)

@davidgatti - please do share your ideas - I am sure others might benefit too on the techniques or learnings in general. We are also very interested in your end-to-end, what kind of cool stuff you are building.

KVS producer can and will stream faster-than-realtime if the bandwidth allows. That's the primary reason why we recommend having as large of a storage (and buffer duration for that matter) as your device allows. We are working on providing some heuristics/automation/samples in the sample application on how to automatically detect the available RAM and configure KVS producer with the optimal content store size.

Regarding the disk storage. You can potentially have two paths currently:

1) As you've mentioned, create a separate media pipeline that will mux the stream into a disk in an MKV form and then use PutMedia Service API to push the resulting file in - subject to adherence to KVS-specifics for the MKV.
2) Implement your own content store in PIC. The content store is an interface with the default implementation mapping to a low-fragmentation fast heap that is optimal for the streaming scenario and buffer utilization patterns for KVS. There is a separate implementation (for debug purposes and systems which don't handle virtual memory mapping well) which relies on processes default heap implementation (read malloc). You could implement disk-based storage there.

This said, we do know the urgency and the importance for implementing disk-backed content store. We do have plans of providing a hybrid store which will utilize the available RAM with a spill-over to disk. 

Unfortunately, the implementation timeline is a subject of multiple external factors and it would be unwise for me to provide any guidance on this. I would just state that we know the customers are asking for it and we will do everything to get this out as soon as possible.

davidgatti, 2018-04-20:
@MushMal you if your business card dose not say that already, you should definitely change that ;) 

Regarding your answers:

1. I think I understand what your saying
2. you lost me :D 

I'll do some experiments, and see what happens. In the mean time do you recommend some articles or even books on streaming video, that could help me and others better understand this complex topic?

MushMal, 2018-04-20:
@davidgatti the KVS producer has a single Content Store which is shared among the streams for simply storing data. The implementation is in <cpp sdk root>/kinesis-video-pic/src/heap/ directory.
The PIC is plain ANSI C so it doesn't have OO like C++. We implement a fast polymorphism which allows to provide different implementations behind the same interface. The default is currently using our own heap implementation. What I was stating is that you could potentially provide a file based implementation (unless you wait for it and we will provide one *soon*).

RE: Books - KVS spans two worlds really - the video encoding/processing and streaming. Traditionally, these have been far apart and over the years they have started coming together - mostly on the consuming side but less on the producing side. KVS attempts to do exactly that.

I will update the thread if I have anything.

davidgatti, 2018-04-23:
OO, ok. now all is clear. Thank you :)

unicornss, 2018-04-26:
As we  have clarifications for the questions, closing this post for now.  Please reopen if required. 
Thanks.

closed 2018-04-26 by unicornss

ashutoshIITK, 2019-09-11:
@MushMal 
Hi MushMal, do you have any update on this? I am also planning to build something like that.

MushMal, 2019-09-11:
@ashutoshIITK apologies for this taking longer than anticipated. We still don't have an ETA for disk/flash based content store.

schmidty1913, 2024-06-08:
@MushMal 
Sorry, this is an old thread, but is there a solution to this problem yet?