awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #73: curl perform failed for <url> with SSL_write() returned SYSCALL, errno = 32

ameyat2812, 2018-04-19:
I am uploading an rtsp stream to Kinesis Video Stream with a logic similar to rtsp streaming sample app.

Sometimes is the network has some glitches and would get the SSL_write() error as stated above from Response.cpp and after that, the stream handle stops uploading the video. I have an instance of KinesisVideoProducer and have instances of KinesisVideoStreams attached to it. After the error; not one the stream which threw this error gets into a bad state but also the other streams stop uploading.

I have already implemented streamerrorhandler callbacks, but none of them report any error. Which makes the application think that the VideoStream is just fine and it keep pushing data to it.

Here are my questions
1. What is the best way to get a status update on any curl/ssl related failure.
2. Is there any way to reconnect the existing KinesisVideoStream. e.g. can successive stop + start API calls reset CURL/SSL handles?
3. What all types of errors handled by KVP ?

ameyat2812, 2018-04-19:
@MushMal any thoughts?

MushMal, 2018-04-19:
@ameyat2812 this sounds quite odd. Can you please clarify your scenario and most importantly, attempt to reproduce it using one stream and capture the logs?

The curl/ssl failures will be translated to PIC error codes. Some will be re-triable. If the retry fails a number of times (in a progressive back-off fashion) then the code will call the stream error callback with the appropriate status code.

Let's look at a hypothetical scenario of turning off networking on an active stream. In this case, an error will be triggered from the networking stack. The error will be re-triable and the stream state machinery will more to the appropriate state - likely to DescribeStream state. The state machine will attempt to call DescribeStream a number of times - the last will be called around 2 seconds later. The SDK will then issue an error in the handler.

The application can differentiate errors and act appropriately. In case of network connection being re-established, the application can call resetConnection API to re-try streaming.

blechatellier, 2018-04-23:
@MushMal we are experiencing a bunch of dropped frames we can't recover from resulting in curl errors, any idea what is going on please? Our upload connection is pretty solid 100mb so unlikely to be a bandwidth issue.

```
23.04.18 17:24:55 (+1000)  main  WARN [23/04/18 07:24:55.031][kvs_client][kvs_client.cc:180] Reporting dropped frame. Frame timecode 1524468213188000
```

```
24.04.18 09:44:22 (+1000)  main  ERROR[23/04/18 23:44:22.993][com.amazonaws.kinesis.video][Response.cpp:191] curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Failed sending data to the peer: SSL_write() returned SYSCALL, errno = 104
24.04.18 09:44:22 (+1000)  main  WARN [23/04/18 23:44:22.994][com.amazonaws.kinesis.video][Response.cpp:211] HTTP Error 0: Response:
24.04.18 09:48:12 (+1000)  main  ERROR[23/04/18 23:48:12.043][com.amazonaws.kinesis.video][Response.cpp:191] curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 10 bytes/sec transferred the last 10 seconds
24.04.18 09:48:12 (+1000)  main  WARN [23/04/18 23:48:12.043][com.amazonaws.kinesis.video][Response.cpp:211] HTTP Error 0: Response:
```

MushMal, 2018-04-24:
@blechatellier - you seem to have two issues: One is caused by SSL_write(), errno = 104 and the other is related to the Operation too slow error. 

The first one indicates ECONNRESET. The second one means that the connection is below 10 bytes per second. Both indicate TCP connection problems.

While your network might have 100mbps, its possible that there is a permission or port issue. Also, it's possible that there are internal firewalls preventing 443 port (standard SSL).

MushMal, 2018-04-24:
KVS producer performs streaming token rotation when the token expires. Even if the token doesn't technically expire (when you provided access key and secret key) we will cap it to 40 minutes. I wonder if the errors correspond to the new session. 

Still, the question stands - why the error.

* Please run the exact same scenario and try to capture the logs preceding the error - few 10s of seconds is good.
* Ensure your network topology is OK - no firewall quirks. Also, some proxies block the HTTP 1.1 response until the request is done. In our case, we need the response to be coming in as we are sending the HTTP body (request) which is the payload. This should still work technically if the connection is maintained.
* Could you try to run the sample/demo application on your laptop and see whether it works? Does it work on a non-corp network?

MushMal, 2018-04-24:
One more thing... the backend load balancer can at times drop the connection (for load shedding, etc...). Getting both of the behaviors is normal in those cases. However, load balancer induced connection reset is a rare event and the SDK should be able to re-start the connection automatically.

ameyat2812, 2018-04-24:
@MushMal To start with, can we have connection reset logic for all the streamErrors callbacks? 

There is no one to one mapping between the STATUS that is sends to StreamErrorReportFunc callback and ErrorCodes listed in here https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html.

MushMal, 2018-04-24:
@ameyat2812 you are touching a few points if I understand you correctly. Please correct me if I am wrong.

* The PIC as well as the C++ code (and Java) will utilize STATUS codes defined in the error codes that you listed. The result value that's specified in the operation result events that go to PIC are in parallel and some of them do get mapped to errors (as results could be errors).
* We will be publishing a doc describing error recovery and token rotation logic and details. However, in the interim, if the networking error is not and internal error (which it shouldn't be in your case anyway) or an authentication issue then the logic would retry up-to max retry count which is defined as 5 with progressive time back-off. After this, the error callback is called.
* We will be updating the sample applications with better error and events handling that will demonstrate how the application might behave on network pressures for example.

In this particular case I am just not sure whether this is a network topology issue or something else. This is why I am asking whether you folks can run the sample or your application from a non-proxy network.

ameyat2812, 2018-04-26:
@MushMal I am now using STATUS codes mentioned however it is still not very clear how to handle stream errors just right. Let me step back and explain the way we are using SDK in the first place and then I will state some of the issues that we are facing right now.

Top level overview of the use case - We are creating two streams per camera one for request based upload and the other uploads video based on some smarts. Both the streams have RTSP input which is something similar to RTSP sample app.

- I create the KVProducer instance one for each camera and then create two KVStream instances right after that. The stream start is called when the first h264 frame is received along with CPD just like the sample app.

- For both the pipelines the streams are never stopped, and new frames are fed to KVStream (putframe) from key frames whenever upload is required.

Feel free to point out if there is any obvious fault in this logic.

Questions regarding stream error handling.

1. I recon SDK has some sort of retry logic to handle errors, does it mean streamError callback is called 
only for non recoverable errors for which retry logic has failed.

2. Is reset connection expected to work always (cause we are facing failure to reset connection on INVALID_MKV_DATA) and should we stop the stream before reseting connection or that is handled by SDK.

3. Can stop and start on the stream be an alternative to reset connection. If not what is the typical use of stop/start.

INVALID_MKV_DATA issue that we are facing.

We are getting bunch of dropped frame errors followed by INVALID_MKV_DATA errors on the smart stream sometimes, I recon there is a connection between size of buffer and incomplete fragments being sent and INVALID_MKV_DATA. (correct me if I am wrong). 

1. What is the formula to arrive at a buffer size so that these errors are avoided?
2. Does max stream latency play any role in this? we are setting it to 60Secs right now.
3. Has device_info.storageInfo.storageSize got to do anything with this? We are setting it to 128Mb

MushMal, 2018-04-26:
The logic looks right - one producer client object and multiple streams (the unit tests in the C++ SDK allow for creating multiple streams - like your case).

SDK has rather complex error recovery/re-streaming from checkpoints. Only when it gives up then it calls the error handler. Also, if there are errors returned from the backend, those will be returned to the callback - even though, the SDK will be retrying. So, it's important to separate those. For example, INVALID_MKV_DATA will be returned to the application through the callback as it's been returned from the backend as an ERROR ACK and the SDK will be backing-off and retrying from the last checkpointed frame (with fixing-up that's required for the stream start).

Stream.resetConnection "simulates" an error which allows the SDK to kick-off the recovery/re-stream logic. The net result is that when your stream is "stuck" on a bad connection or your buffer fills up because of a stale connection you could "pulse" the stream without losing any data.

You can stop the stream but you can't simply start it again. You need to tear down the stream and re-create. This is why we introduced a simple resetConnection to simply pulse the network connection whereas the re-creation of the stream will also mean data loss. 

There are situations, however, when the client application might want to stop/free/re-create the stream - those are reserved for example when you have a long buffer (hours perhaps) that you just want to drop all together without re-streaming.

INVALID_MKV_DATA can be caused by a few external factors (very unlikely) but the likely cause is a latency or storage out of memory caused dropped frame. As per KVS documents, I recommend using as much latency/buffer/storage as possible.

The default storage size should be calculated based on the following formula:

Size = NumberOfStreams * AverageFrameSize * FramesPerSecond * BufferDurationInSeconds * DefragmentationFactor

Where the DefragmentationFactor should be 1.2 (as 20%). For example, if a device has Audio and Video streams. Audio has 512 samples per second, average sample 100 bytes. Video has 25 frames per second, average 10,000 bytes and each has 3 minutes of buffer duration.

Size = (512 * 100 * (3 * 60) + 25 * 10000 * (3 * 60)) * 1.2 = (9216000 + 45000000) * 1.2 = 65059200 = ~ 66MB.

Max stream latency will be used for the stream latency pressure notification function. When your buffered content gets over this duration then the callback will be called.

DeviceInfo.storageInfo.storageSize controls how much content storage to use - the formula above controls the behavior.

Please do not hesitate to ask more questions. Also, we are in a process of publishing docs that go over the details.

MushMal, 2018-04-26:
1) Yes, by introducing a connection termination it will kick off the re-streaming sequence which will, in turn, create a new connection
2) If I understand it correctly you are trying to handle two cameras two streams each on a device - similar to an NVR. In that case, you don't need to create two producer clients - create just one so the storage can be shared which is optimal. I would assume you would want to differentiate the streams by adding to some sort of a map of stream to camera or simply encode the camera name in the stream name. In this case, you really create one producer which will handle all cameras/streams. This will not work if each camera will require it's own different security context - different access key/secret key.

ameyat2812, 2018-04-26:
We need to have different key/secret key for every camera ATM. So I guess we might need to have one producer clients per camera.

I have increased the buffer size but seems to just push the problem a bit further.  As our app keeps uploading to the same stream on and off I suspect if over the period of time we are left with some dormant fragments in the buffer.

What are typical cases where the buffer gets filled by the way?

I would like to refer to DefaultDeviceInfoProvider.cc where it reads
// We put any overflow storage in /tmp as it is in memory.
    string dir = "/tmp";

Does it mean device_info_.storageInfo.storageSize is allocated on RAM and if it overflows then new frames are written on the disk in /tmp/<some_file>??

MushMal, 2018-04-26:
@ameyat2812 interesting case. It's possible to handle multiple security tokens for different streams but it will mean overriding the DefaultCallbackProvider functionality. Perhaps, in this case it's better to isolate the cameras in their own producer clients. 

When you say you've increased the buffer size I assume you've increased the storage size as if you increase the buffer size (as in buffer duration in the stream definition) and you are low in memory and your connection is not fast then you will fill the buffer and then overflow the content store and the latest frame will be dropped instead of the earliest.

The root directory in the device info is not currently used. As you've noticed, this will bee needed for the hybrid RAM/file backed content store which we haven't implemented yet.

One thing you could do is to capture the logs leading to the issue so we can see the metrics reported in the logs which will show the snapshot of what the stream buffer looks like.

ameyat2812, 2018-04-26:
I meant storage size. I still kept buffer size as 120sec as it was and boosted storage size to 512MB.

Will capture and attach logs when the failure occurs. But is slow network the only reason for storage getting full. Could memory leaks/ incomplete  fragments waiting in storage cause buffer to overflow.

PS. When the stram gets stuck due to buffer being full. The other stream in KV producer client works just fine

MushMal, 2018-04-26:
The producer SDK utilizes it's own memory manager to handle the storage for multiple reasons:

* Less de-fragmentation
* Optimized for the streaming scenario
* Having hard boundaries on the allocations
* Portable
* Could be replaced by any implementation
* Fully tested 

This is for the storage itself. The SDK does have a few minor allocations too - for the content view. The good thing is that the entire codebase for PIC is allocating using MEMALLOC* macros. If you look at the unit tests, the client unit tests override the default allocator with a custom one that does accounting. So, after all unit tests are run for the client object the memory is checked for any leaks.

I am not sure what you mean by the "incomplete" fragment. One of the things the SDK allows is to abstract the actual packaging mechanism. So, the application just puts the frames and doesn't worry about the fragmentation. 

I would really like to understand the details when you say "when the stream gets stuck due to buffer being full". The streams are independent from each other - each has it's own content view (buffering) but the actual frames are stored in the common storage. If you mean latency pressure or dropping the frames due to the pressure then it's caused by that particular stream not making progress or there is a bandwidth issue which makes the stream fall behind. 

Having the logs can help to understand what's going on.

ameyat2812, 2018-05-07:
Hey @MushMal  Since the 512Mb bump to my device storage, I am able to see this issue with following DEBUG logs.
[com.amazonaws.kinesis.video.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/1978723/com.amazonaws.kinesis.video.log)

Please let me know what is causing it to drop frames

MushMal, 2018-05-07:
There a re a lot of latency pressure notifications followed by dropped frame notifications:

DEBUG[06/05/18 23:59:51.109][com.amazonaws.kinesis.video][KinesisVideoStream.cpp:63] Kinesis Video client and stream metrics
	>> Overall storage size: 536870912
	>> Available storage size: 531666755
	>> Allocated storage size: 5204157
	>> Total view allocation size: 115344
	>> Total streams frame rate: 49
	>> Total streams transfer rate: 4246857
	>> Current view duration: 804960000
	>> Overall view duration: 804960000
	>> Current view size: 4931710
	>> Overall view size: 4931710
	>> Current frame rate: 24.8811
	>> Current transfer rate: 52553 
WARN [06/05/18 23:59:51.128][com.amazonaws.kinesis.video][KinesisVideoProducer.cpp:354] 
viewItemRemoved(): Reporting a dropped frame/fragment. 
DEBUG[06/05/18 23:59:51.128][com.amazonaws.kinesis.video][DefaultCallbackProvider.cpp:826] droppedFrameReportHandler invoked 
WARN [06/05/18 23:59:51.142][com.amazonaws.kinesis.video][KinesisVideoProducer.cpp:354] 
viewItemRemoved(): Reporting a dropped frame/fragment. 
DEBUG[06/05/18 23:59:51.142][com.amazonaws.kinesis.video][DefaultCallbackProvider.cpp:826] droppedFrameReportHandler invoked 
DEBUG[06/05/18 23:59:51.142][com.amazonaws.kinesis.video][DefaultCallbackProvider.cpp:809] streamLatencyPressureHandler invoked 
WARN [06/05/18 23:59:51.168][com.amazonaws.kinesis.video][KinesisVideoProducer.cpp:354] 
viewItemRemoved(): Reporting a dropped frame/fragment. 

You seem to have two streams - each around 25 fps so overall observed FPS is 50. The output I brought in here indicates that your bandwidth is oversaturated and can't keep up with the encoder pace. The observed combined upload bandwidth is ~34Mbps.

The suggestion here would be to
1) Ensure you don't have an intermittent networking issue where it can slow down and accumulate the buffer but then can't stream fast enough to catch up
2) Ensure your encoders are tuned to produce lesser density streams by
    a) Reducing the resolution
    b) Reducing the frame rate
    c) Decreasing the video quality thus increasing the compression rate
3) Try on another network??

Please let us know how this helps/

ameyat2812, 2018-05-08:
@MushMal I have 2 streams per Camera and the MAX latency for both the streams in set to 60Sec at the moment.

Let me also specify our encoder settings for both the streams just to put things into perspective.

1. H.264-1280x720p-1Mbps-25FPS-I frame interval every 1 sec 
2. H.264- 640x360 -512Kbps-25FPS- I frame interval every 1 sec

Questions:

1. What is observed combined upload bandwidth? I reckon going by the above encoder setting it can never really cross 2Mbps even if both the streams are uploading.

2. Our office network has an upload BW of 400Mbps, considering that the network might have gone bad for more than Max latency time (60 Secs) which would have triggered Frame drops, shouldn't it catchup to realtime rate the moment network recovers.

3. Which frames are dropped on latency pressure? The new frames being submitted now or the old frames residing in storage?

MushMal, 2018-05-08:
@ameyat2812 based on your encoder settings it should only be around 1.5 Mbps but what I can see from the logs that you have variable rates up-to 34Mbps. The observed rate is the actual rate at which the stream occurs. For example, you might have a dedicated line with 400Mbps but the actual throughput through the gateways and switches to the AWS data center could be less (are you attempting to stream to a closest region)? One thing you can do is to measure the data rate (when putting frame into the SDK) from the encoder and ensure that it really produces the advertised bitrate and not something much higher.

The design of the KVS is such that it can absorb the duration worth of data and attempt to catch up when the bandwidth allows. The buffer is FIFO so in your case the frame are dropped due to the buffer pressure which means that the eldest frames are being dropped from the buffer.

The entire log already starts from dropped frames. The SDK is designed to retry and progressively back-off on failures. For example, if you have an intermittent network outage, the streaming will be terminated with an error. The stream state machine will likely loop back to "Describe" state where it will attempt to call DescribeStream API. It will attempt to call it up-to 5 times, with increasing timeouts. If after the last attempt we still get an error, the error will be returned to the caller application - in your case it will be returned as a result of putFrame(). 

Could I ask you to double-check the region and the encoder settings to ensure you are not attempting to stream a high-density streams (you can log/output the throughput just before you put the frame into the SDK so you know the frequency and the actual frame sizes). Next, please try to capture the logs from the beginning so we can see what's happening right from the beginning. Next, add some logic to fail over on error. For example, on hard errors you can stop, free and re-create the stream.

We are going to publish an update to the sample applications with some more illustrative example of error handling in different scenarios. Perhaps that might help too to get your application to be able to properly respond to the error conditions.

unicornss, 2018-05-21:
Closing the issue for now as it has been answered. 
Please reopen if needed.  
Thanks.

closed 2018-05-21 by unicornss