awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #77: Log4cplus url change?

paulhilsen, 2018-05-07:
In the install-script, the log4cplus url on line 128 appeared to not work for me. Even trying to hit the url in my computer's browser gave me a "ERR_ADDRESS_UNREACHABLE"

For reference line 128: curl -L "https://versaweb.dl.sourceforge.net/project/log4cplus/log4cplus-stable/1.2.0/log4cplus-1.2.0.tar.xz" -o "log4cplus-1.2.0.tar.xz"

I ended up manually downloading log4cplus by using:
wget https://sourceforge.net/projects/log4cplus/files/log4cplus-stable/1.2.0/log4cplus-1.2.0.tar.xz

This new url appears to have worked for me, though I wouldn't mind having the install-script updated if the current url in is indeed outdated. Thanks!

For the record this is on a Raspberry Pi Zero.

unicornss, 2018-05-07:
Thanks @paulhilsen for notifying us.  
The _url not reachable_ issue has been noticed late last week and we are in the process of updating the script with the location you have mentioned. 
The other option is to use the Github release location for [log4cplus](https://github.com/log4cplus/log4cplus/releases/download/REL_1_2_0/log4cplus-1.2.0.tar.gz) .

closed 2018-05-07 by unicornss

paulhilsen, 2018-05-07:
Excellent. Thank you for the response. Is there somewhere on git where you are publishing  known issues that are being worked on? Looking to not annoy you in the future if I find something and it's already known.

Paul