awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #80: What is the best way to reset the KinesisVideoStream which is added to KinesisVideoProducer instance

ameyat2812, 2018-05-22:
Our setup has one KinesisVideoProducer instance created per camera which has number of KinesisVideoStreams attached to it for different purposes.

I am trying to reset the streams in case of FATAL errors. Refer to this issue[https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/73](url).

I have tried two approaches

1. Stop and Start the stream

- The stream never gets ready even after successful Start.

2. freeStream and Create a new KinesisVideoStream instance

<ProducerInstance>->freeStream(<StreamName>);
<ProducerInstance>->createStreamSync(move<New Stream Definition >);

- This logic generates Unable to create Kinesis Video stream. <StreamName> Error status: 0x5200004b error with Failed to create Kinesis Video Stream.

I reckon this error is STATUS_DUPLICATE_STREAM_NAME, however I think it should still work as we have already called freeStream on the stream and that should clear all the context for the stream.

ameyat2812, 2018-05-22:
@MushMal Any suggestions?

MushMal, 2018-05-22:
@ameyat2812 thanks for the question.  Here are some of my thoughts...

* You can't re-start a stopped stream. Stopping stream will issue an async command to deplete the buffer and stop. You will get a streamStopped notification when and if successfully streamed and finished the last bits.
* In case of an error, it's preferred method is to "pulse" the stream. We have a resetConnection() API on the stream which would be useful. This allows a retrying without dropping the buffer.
* In case if the reset doesn't "help" then you could try to stop, await for a certain period of time for the streamStopped notification and if it times out of successfully completes then free the stream and re-create.
* You can certainly call freeStream() which will drop the existing buffer (might loose frames) and re-create.
* You might still be having a pointer to the stream object and that's why you can't re-create. This actually is an issue and we perhaps should handle this better in the freeStream API. For now, you can call freeStream() and then delete the pointer to the stream itself or call ->free() on the stream itself before attempting to create the stream again.

We are in a process of pushing out some sample code that can demonstrate the different stages of attempting to restart the streaming. I can update you when we publish the sample.

ameyat2812, 2018-05-24:
@MushMal I have handled critical stream errors by stopping and recreating the stream. 

1. I now trigger stop on stream when I get streamError callback and resetConnection() API fails.
2. Then I wait for stream to be closed.
3. I Free the stream on closed
4. Create a new stream and start pushing on success.

The above stated logic woks however, I think it would be nice to have an Thread safe API which can be called on critical errors where data loss is expected on the stream and which leaves stream in ready state always.

MushMal, 2018-05-24:
@ameyat2812 - thanks for the feedback. We agree - as with the create API there should be a corresponding sync api for stop. We are also going to make the stream teardown more streamlined. Stay tuned.

AndrewChubatiuk, 2018-06-07:
@ameyat2812 
Can you please share your code? Have problems with stream recreation.

ameyat2812, 2018-06-08:
@AndrewChubatiuk I can't share the code however let me try and explain how I am doing it.

1. I implemented streamErrorReportHandler callback and where I try and reset stream in question using KinesisVideoStream::resetConnection API. If that fails (go to step 2)

2. Call KinesisVideoStream::stop on the stream in question. And wait for streamClosedHandler callback

3. On streamClosedHandler I firstly free stream context 
kinesis_video_producer_->freeStream(std::move(<KVS stream shared pointer>));

Then create a new stream and start it
kvs_stream_vid_ = KinesisVideoStreamCreate("name", "video/h264", "kinesis_video","V_MPEG4/ISO/AVC");
kvs_stream_vid_->start(stream_cpd_string);

I am also using conditional_variables to protect frame producer thread to call putFrame API when the video stream is being restarted.

I hope this helps. l would be happy to help if you can specify the exact issue that you are facing.

AndrewChubatiuk, 2018-06-08:
@ameyat2812 
Thank you very much! It was really helpful.

unicornss, 2018-06-28:
Closing this issue for now. Please reopen or create a new thread if required. Thanks

closed 2018-06-28 by unicornss