awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #82: No Frame Acknowledgement received if video frames pushed to KVS as fast as possible

ameyat2812, 2018-06-05:
Hi @MushMal ,

I am working on logic to upload video to one of the KinesisVideoStream from the files stored on HDD. Unlike live contents coming from camera, these frames would have all the data available and can generate frames for an hour long video data in milliseconds.

I am using following logic

while(1) {
   // Get frame from HDD then create kinesis video frame
   Frame frame;
   bool put_frame_act_ret = kvs_stream_act_vid_->putFrame(frame);
    if (put_frame_act_ret != true) {
        LOG("Unable to put frame to kinesis");
    }

    if(TimeOfFrame >= stopTime)
         break;
}

I am not getting any error back from putFrame() however I don't receive any acknowledgement either. But if I slow it down by putting some sleeps but still keeping it faster than real time (5x to be precise), I start getting fragment ack again.

Can you please explain this behaviour?

MushMal, 2018-06-05:
@ameyat2812 - this is what's called an "offline" scenario. Your code is hitting a throttling limits - in this case you need to produce frames at lower fps. We are working on supporting the "offline" case - no ETA yet but we understand how important these scenarios are (especially for the migration).

ameyat2812, 2018-06-05:
Can I expect to get a specific error code being returned by putFrame API if I am hitting throttling limits. Cause at the moment, there is now way for me to know if I have pushed frames faster than allowed FPS and have slow down a bit. This sleeping between successive frame feeds is very lame logic.

MushMal, 2018-06-05:
It's a good topic for a conversation. The issue is that any errors will also terminate the stream. We are still thinking about the best way to handle this as we are not officially supporting this scenario. One of the ways would be to indicate the pressures via ACKs.

ameyat2812, 2018-06-05:
As offline mode is not supported and there is no ETA on it as well. I guess I am left with only option to fake real-time upload even for historical footage. I would still like to speed things up a bit, can you comment on maximum allowed FPS per stream?

I do have a slightly better implementation in mind,

I am thinking of pushing frames as fast as I can for a fixed length of video (10Sec) and monitor fragment acks for the pushed frames in another thread. Once the ack for the last uploaded I frame is received, I would start pumping next chunk out. Do you think this makes any sense?

MushMal, 2018-06-05:
I think it's a good workaround for now. An alternative (thinking out lout) would be to have a larger buffer and enable fast push of frames until you get enough buffer fullness and/or you get stale connection notification.

unicornss, 2018-06-14:
Closing the issue for now as the question has been answered. Please reopen if required.

closed 2018-06-14 by unicornss