awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #83: SSL CA cert issue while running kinesis_video_gstreamer_sample_app

anmol1ratn, 2018-06-11:
I cloned the amazon-kinesis-video-streams-producer-sdk-cpp, following all the necessary steps including setting the environment variables like:
export AWS_ACCESS_KEY_ID={MY_ACCESS_KEY_ID}
export AWS_SECRET_ACCESS_KEY={MY_SECRET_ACCESS_KEY}, and 
export LD_LIBRARY_PATH=/home/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH

Then I ran the command from inside the native build folder: 
anmol@anmol-pc:~/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ ./kinesis_video_gstreamer_sample_app my-stream

I got the following error:

INFO - Initializing curl.
INFO - 
createKinesisVideoClient(): Creating Kinesis Video Client
INFO - 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
INFO - 
heapInitialize(): Creating AIV heap.
INFO - 
heapInitialize(): Heap is initialized OK
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739902 Expiration: 0
INFO - New credentials expiration is 1528742302
DEBUG - createDeviceHandler invoked
INFO - 
createDeviceResultEvent(): Create device result event.
DEBUG - clientReadyHandler invoked
DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream my-stream
INFO - 
createKinesisVideoStream(): Creating Kinesis Video Stream.
DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739902 Expiration: 0
**ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none**
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=JKIAS3I4Q7TUE73AC3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=26209161a20fc1f34f333b05985d1955a9c7143f805d3333cabbe10f03a6c609
    content-type: application/json 
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175822Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739903 Expiration: 0
**ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none**
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI3G4Q73GYS3ZJ3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=32374ce9d3ec4a910af7d8579d658785c89e52bf1a32123bfad32ebf6990e0ed
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175823Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739904 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI3G4Q73GYS3ZJ3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=c5e3e59913b329436e6fd13194efe4d468a9709c5d4fe58f3595da25abaf7ef3
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175824Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739905 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI3G4Q73GYS3ZJ3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=ce48d49684df87131b661071d2211a4c6ca658fd183daf71c9b2066b822d1002
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175825Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739907 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI3G4Q73GYS3ZJ3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=10f7877daec0a42342abdd6475f514bc5ff631b7ec4b2c4e0f2e1a6b49ecf60a
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175827Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528739910 Expiration: 0
ERROR - curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
  CAfile: /etc/ssl/cert.pem
  CApath: none
WARN - HTTP Error 0: Response: 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI3G4Q73GYS3ZJ3NA/20180611/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;x-amz-date, Signature=4d7b7c049e6faa8e3e0d32414fb9bbc1c89cb3c37ef2b40377d2a25d8dc8fee6
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    X-Amz-Date: 20180611T175830Z
DEBUG - describeStream response: 
INFO - Describe stream did not find the stream my-stream in Kinesis Video (stream will be created)
INFO - 
describeStreamResultEvent(): Describe stream result event.
ERROR - Submitting event result for stream: 94708690864654 failed with: 1375731729
ERROR - Reporting stream error. Errored timecode: 0 Status: 1375731729
ERROR - Failed to create Kinesis Video Stream - timed out.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to create Kinesis Video Stream - timed out.

Please help me with this issue? It's pretty hard to push live feed from the cameras to the Kinesis Video Stream.

PS: A video stream named "my-stream" had already been created in the aws console.

chehefen, 2018-06-11:
Hi, can you check if your user has access to the /etc/ssl/ directory? If so, can you try the following steps?
- download the PEM file from https://www.amazontrust.com/repository/SFSRootCAG2.pem
to /etc/ssl/cert.pem. Append to the end of the file if it exists.
- rm /home/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl*
- rm /home/Project/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/bin/curl*
- run install-script again.

MushMal, 2018-06-11:
Thanks @chehefen . @anmol1ratn , there is some information about this and other issues listed in Readme.md

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#certificate-store-integration

closed 2018-06-15 by anmol1ratn