awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #84: on_new_sample() error

julianoteixeirafox, 2018-06-11:
I've tried to stream an RTSP video from my camera without success. I also tried other sample RTSP urls and I get the same error. The url I tried is rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov and this is the output:

```INFO - Initializing curl.

createKinesisVideoClient(): EnterINFO -
createKinesisVideoClient(): Creating Kinesis Video Client
INFO -
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
INFO -
heapInitialize(): Creating AIV heap.
INFO -
heapInitialize(): Heap is initialized OK
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528748797 Expiration: 0
INFO - New credentials expiration is 1528751197
DEBUG - createDeviceHandler invoked
INFO -
createDeviceResultEvent(): Create device result event.
DEBUG - clientReadyHandler invoked
DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream front-door-camera
INFO -
createKinesisVideoStream(): Creating Kinesis Video Stream.
DEBUG - describeStreamHandler invoked
DEBUG - Awaiting for the stream to become ready...
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528748797 Expiration: 0
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.528591028827E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:857745839647:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827","StreamName":"front-door-camera","Version":"UGTgyEZaXYlqCGQvRdeU"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528748798 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
INFO -
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1528748798 Expiration: 1528751197
INFO - New credentials expiration is 1528751198
INFO -
getStreamingTokenResultEvent(): Get streaming token result event.
DEBUG - streamReadyHandler invoked
DEBUG - Kinesis Video stream front-door-camera is Ready.
DEBUG - Stream is ready
DEBUG - Pipeline OK
DEBUG - rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov
DEBUG - appsink configured
DEBUG - Pipeline built, elements *NOT* yet linked
DEBUG - Pipeline playing
New RTSP source found: recv_rtp_src_0_150577551_96
New RTSP source found: recv_rtp_src_1_68827483_97
Source link FAILED
Source linked.
INFO -
kinesisVideoStreamFormatChanged(): Stream format changed.
ERROR -
on_new_sample(): kinesis_video_flags=1 pts=605968510 dts=605968510

DEBUG - putStreamHandler invoked
DEBUG - Created a new PutMedia request with stream upload handle: 0
INFO -
putStreamResultEvent(): Put stream result event.
INFO - Creating new connection for Kinesis Video stream: front-door-camera
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528748799 Expiration: 0
INFO -
getKinesisVideoMetrics(): Get the memory metrics size.
INFO -
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00007f94da8ee66e.
DEBUG - Kinesis Video client and stream metrics
	>> Overall storage size: 536870912
	>> Available storage size: 536869760
	>> Allocated storage size: 1152
	>> Total view allocation size: 144072
	>> Total streams frame rate: 30
	>> Total streams transfer rate: 4194304
	>> Current view duration: 200000
	>> Overall view duration: 200000
	>> Current view size: 1112
	>> Overall view size: 1112
	>> Current frame rate: 30
	>> Current transfer rate: 4194304
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=647968510 dts=647968510

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=688968510 dts=688968510

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=730968510 dts=730968510

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=772968510 dts=772968510

DEBUG - Curl post header write function returned:HTTP/1.1 100 Continue

DEBUG - Curl post header write function returned:

DEBUG - Wrote 1340 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=813968510 dts=813968510

DEBUG - Wrote 156 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=855968510 dts=855968510

DEBUG - Wrote 163 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=897968510 dts=897968510

DEBUG - Wrote 157 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=938968510 dts=938968510

DEBUG - Wrote 162 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=980968510 dts=980968510

DEBUG - Curl post header write function returned:HTTP/1.1 200 OK

DEBUG - Curl post header write function returned:x-amzn-RequestId: beeef72d-6db5-11e8-957d-dd8a783f5ab7

DEBUG - Curl post header write function returned:Content-Type: application/json

DEBUG - Curl post header write function returned:Transfer-Encoding: chunked

DEBUG - Curl post header write function returned:Date: Mon, 11 Jun 2018 20:26:39 GMT

DEBUG - Curl post header write function returned:

DEBUG - Wrote 42 bytes to Kinesis Video. Upload stream handle: 0
INFO - Curl post body write function for stream: front-door-camera and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":0,"FragmentNumber":"91343852333181774400852596062858897863980421376"}

DEBUG - fragmentAckReceivedHandler invoked
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1022968510 dts=1022968510

DEBUG - Wrote 99 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1063968510 dts=1063968510

DEBUG - Wrote 139 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1105968510 dts=1105968510

DEBUG - Wrote 172 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1147968510 dts=1147968510

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1188968510 dts=1188968510

DEBUG - Wrote 174 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - Wrote 74 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1230968510 dts=1230968510

DEBUG - Wrote 165 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1272968510 dts=1272968510

DEBUG - Wrote 180 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1313968510 dts=1313968510

DEBUG - Wrote 185 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1355968510 dts=1355968510

DEBUG - Wrote 211 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=1397968510 dts=1397968510

DEBUG - Wrote 190 bytes to Kinesis Video. Upload stream handle: 0```

chehefen, 2018-06-11:
Hi, my apologies. You can get rid of that by commenting out the line 212 in the kinesis_video_gstreamer_sample_rtsp_app.cpp.

chehefen, 2018-06-11:
Hi, just wanna confim if you were able to stream successfully. Thanks

julianoteixeirafox, 2018-06-12:
Thank you for the quick reply but the issue is not the log message. The problem is that the content is not showing in KVS.

MushMal, 2018-06-12:
@julianoteixeirafox could you please provide more information. Assuming you are trying to view the content in the KVS console window. What does it show? Could you also include more of the log spanning a few tens of seconds to ensure you are getting the ACKs back from the service so we can see whether the ingestion works just fine.

One important note: The KVS console playback will only "work" if the retention period of the stream is not set to 0.

unicornss, 2018-06-13:
@julianoteixeirafox  
[image]

Just checked the console view for the test stream  `rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov` and it renders the video without any issues. 

Could you attach your screenshot? 
Thanks

julianoteixeirafox, 2018-06-13:
[image]

julianoteixeirafox, 2018-06-13:
And these is the logs I get:
INFO - Initializing curl.

createKinesisVideoClient(): EnterINFO -
createKinesisVideoClient(): Creating Kinesis Video Client
INFO -
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
INFO -
heapInitialize(): Creating AIV heap.
INFO -
heapInitialize(): Heap is initialized OK
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528932363 Expiration: 0
INFO - New credentials expiration is 1528934763
DEBUG - createDeviceHandler invoked
INFO -
createDeviceResultEvent(): Create device result event.
DEBUG - clientReadyHandler invoked
DEBUG - Awaiting for the producer to become ready...
DEBUG - Kinesis Video producer is Ready.
DEBUG - Client is ready
INFO - Creating Kinesis Video Stream front-door-camera
INFO -
createKinesisVideoStream(): Creating Kinesis Video Stream.
DEBUG - describeStreamHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528932363 Expiration: 0
DEBUG - Awaiting for the stream to become ready...
DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.528591028827E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:857745839647:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827","StreamName":"front-door-camera","Version":"UGTgyEZaXYlqCGQvRdeU"}}
INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827
INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:857745839647:stream/front-door-camera/1528591028827
INFO -
describeStreamResultEvent(): Describe stream result event.
DEBUG - streamingEndpointHandler invoked
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528932363 Expiration: 0
DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
INFO -
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
DEBUG - streamingTokenHandler invoked
INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1528932363 Expiration: 1528934763
INFO - New credentials expiration is 1528934763
INFO -
getStreamingTokenResultEvent(): Get streaming token result event.
DEBUG - streamReadyHandler invoked
DEBUG - Kinesis Video stream front-door-camera is Ready.
DEBUG - Stream is ready
DEBUG - Pipeline OK
DEBUG - rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov
DEBUG - appsink configured
DEBUG - Pipeline built, elements *NOT* yet linked
DEBUG - Pipeline playing
New RTSP source found: recv_rtp_src_0_291024170_96
Source link FAILED
New RTSP source found: recv_rtp_src_1_1977569481_97
Source linked.
INFO -
kinesisVideoStreamFormatChanged(): Stream format changed.
ERROR -
on_new_sample(): kinesis_video_flags=1 pts=625242015 dts=625242015

DEBUG - putStreamHandler invoked
DEBUG - Created a new PutMedia request with stream upload handle: 0
INFO -
putStreamResultEvent(): Put stream result event.
INFO - Creating new connection for Kinesis Video stream: front-door-camera
INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1528932364 Expiration: 0
INFO -
getKinesisVideoMetrics(): Get the memory metrics size.
INFO -
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00007fee65031a6e.
DEBUG - Kinesis Video client and stream metrics
[... 585 lines cut ...]
DEBUG - Wrote 492 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6042936182 dts=6042936182

DEBUG - Wrote 439 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6083957097 dts=6083957097

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6125977844 dts=6125977844

DEBUG - Wrote 453 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - Wrote 417 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6167998425 dts=6167998425

DEBUG - Wrote 498 bytes to Kinesis Video. Upload stream handle: 0
ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6209018842 dts=6209018842

ERROR -
on_new_sample(): kinesis_video_flags=0 pts=6251034170 dts=6251034170

DEBUG - Wrote 508 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - Wrote 473 bytes to Kinesis Video. Upload stream handle: 0

MushMal, 2018-06-13:
Looking at the logs I can confirm that you have no issues streaming data up - you have adequate bandwidth and you are getting ACKs back OK.

We need to understand why you are unable to see the stream. Could you please try server-side timestamps from the console - just under the video screen?

julianoteixeirafox, 2018-06-13:
[image]

Same problem

julianoteixeirafox, 2018-06-14:
[image]

MushMal, 2018-06-14:
Seems like the screen shot is the same - at least the button hasn't changed to Producer time stamps.

At any rate, could you please confirm that you are using the right account/stream that was used in the producer? Also, what's the browser you are using? Assuming you have already ensured you have the adequate bandwidth/latency from your browser.

julianoteixeirafox, 2018-06-14:
[image]

julianoteixeirafox, 2018-06-14:
Yes. It has to be same account because I only have one kinesis stream in my aws console. I'm using chrome.

zhiyua-git, 2018-06-14:
@julianoteixeirafox Can you change the region to us-west-2 in your aws console? It should be on the right top corner. Thanks.

julianoteixeirafox, 2018-06-14:
That did it!

MushMal, 2018-06-14:
Lol! Could you please go ahead and close the issue.

roccomuso, 2018-06-18:
Switching to `us-east-2` did it also for me, but why? what's the reason?

zhiyua-git, 2018-06-18:
@roccomuso I suspect you are talking about us-west-2. By default the producer will send stream data into us-west-2 region and it will create a stream for you if the stream didn't exist. If you want to use Kinesis Video Streams in other regions, please follow [this instruction](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#setting-credentials-in-environment-variables).

optionally, set AWS_DEFAULT_REGION environment variable for the region other than us-west-2.

sailesharya, 2019-03-13:
> Hi, my apologies. You can get rid of that by commenting out the line 212 in the kinesis_video_gstreamer_sample_rtsp_app.cpp.

no file as such, we have amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp file.

can you let me know which line to remove to get rid of 

ERROR -
on_new_sample(): kinesis_video_flags=0 error

Raghava248, 2019-05-04:
Hi Iam trying to live stream using gstreamer plugin in amazon kinesis
rohan@rohan-HP-15-Notebook-PC:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format =I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-recording=FALSE realtime= TRUE ma-keyframe-interval=45 bitrate=500 ! h264parse ! video/-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name= "Aerologiks2" storage-size= 1024 aws-region = us-east-1 log-config=kvslog
WARNING: erroneous pipeline: no element "vtenc_h264_hw"

But Iam stuck here please help me in doing this
Iam using ubuntu 16.04
I perfectly installed ./install-script

MushMal, 2019-05-05:
Addressed in another issue: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/138#issuecomment-489386942